MichaIng/DietPi #4745: ReadyMedia (miniDLNA) cannot start

AnzoP, 2021-09-17:
- DietPi version | v7.5.2 (MichaIng/master)
- Distro version | 11.0
- Kernel versions | Linux 5.10.60-v8+ #1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Official RPi4 Power Supply (5.1V 3A)
- SDcard used | No SDCard, Samsung M.2 SATA 250GB
- Software title | ReadyMedia (miniDLNA)
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
1. Install ReadyMedia on first boot

#### Expected behaviour
Access ReadyMedia on browser, http://<IP Address>:8200

#### Actual behaviour
Service stopped, checking the status, '/var/log/minidlna/minidlna.log' is not available.

#### Extra details
Created '/var/log/minidlna/minidlna.log' with proper ownership to run the service.
Also tried it in a VMware image with the same outcome.

~# sudo service minidlna status
Loaded: loaded (/etc/systemd/system/minidlna.service; disabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Fri 2021-09-17 14:18:45 PST; 5h 14min ago
Process: 4167 ExecStart=/usr/sbin/minidlnad -S -R -f /etc/minidlna.conf (code=exited, status=255/EXCEPTION)
Main PID: 4167 (code=exited, status=255/EXCEPTION)
CPU: 132ms

Sep 17 14:18:45 systemd[1]: Started MiniDLNA (DietPi).
Sep 17 14:18:45 minidlnad[4167]: minidlna.c:1028: fatal: Failed to open log file '/var/log/minidlna/minidlna.log': No such file or directory
Sep 17 14:18:45 systemd[1]: minidlna.service: Main process exited, code=exited, status=255/EXCEPTION
Sep 17 14:18:45 systemd[1]: minidlna.service: Failed with result 'exit-code'.

ravenclaw900, 2021-09-17:
I had the same issue, could you try creating the log directory:
```
mkdir /var/log/minidlna
```
If that doesn't work, also create the file:
```
touch /var/log/minidlna/minidlna.log
```

Joulinar, 2021-09-17:
I guess it's an issue of the apt package themselves. Usually it should create `/var/log/minidlna` by it's own.

not sure if we should create a workaround on our script or report it the the package maintainer. @MichaIng any thoughts?

ravenclaw900, 2021-09-17:
I got the problem after upgrading to Bullseye, so I think that it's related to that.

MichaIng, 2021-09-17:
We actively remove the log directory as it was not used when running minidlnad in foreground mode: https://github.com/MichaIng/DietPi/commit/35c56eb15f1eeb70eba3698edf53c4cb8758b748

But this seems to be not true anymore on Bullseye (with latest MiniDLNA version) and I couldn't find a way to disable the log file and have it logging to STDOUT instead. What a nasty step backwards...

Works still pretty well on Stretch and Buster with logs to STDOUT.

MichaIng, 2021-09-17:
Found it, its a Debian maintainer patch in the latest version: https://metadata.ftp-master.debian.org/changelogs//main/m/minidlna/minidlna_1.3.0+dfsg-2_changelog
> * Add 09-do-not-disable-logs-with-systemd.patch.

Reasons can be discussed, but the problem is that there is no way to have logs at STDOUT now, while it is trivial to send outputs on STDOUT to a log file, if this is wanted.

I'll see if we can revert this patch.

MichaIng, 2021-09-17:
Fixed with: https://github.com/MichaIng/DietPi/commit/ed4c3116f4e44045ad35e2143da7175bfae6a02a
However, we should ask the Debian maintainers to revert this change and instead implement a file logging variable in /etc/default/minidlna to redirect the systemd units output, or the other way round ask upstream to implement Debian's patch + the ability to use STDOUT as special value for "log_dir".

closed 2021-09-17 by MichaIng