MichaIng/DietPi #4748: Dietpi on Pine64 fails to install

Pittconnect, 2021-09-18:
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Sat Sep 18 21:23:20 BST 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | Linux DietPi 5.10.4-sunxi64 #20.11.7 SMP Tue Jan 5 23:28:09 CET 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...I first got an System WARNING when I try to use pip3 install pyserial details can be found here 
2. `WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behavior with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv`
2. ...SO I created a new user group and rebooted the and my MATE Desktop quite blank screen So I created a new image of Dietpi and now I will not complete the installation Once I log in as root it will not allow me to install python3 and pip3
#### Expected behavior:

- I should be able to save the old image configuration and just import that image configureation with all the installed software. Now I am trying to install all the packaged. 
#### Actual behaviour:

- The development package is NOT listed in the software I search and install python3 and pip3 but no development libraries like gcc etc. 
#### Extra details:
More details can be found here Mpoint is me. 
- ...
#### Additional logs:  CAN be found here https://pastebin.pl/view/1f1313d1
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.

Joulinar, 2021-09-19:
The warning created by `pip` could be ignored usually. `pip` is complaining about using a super user. But it should be ok.

To try to fix your issue, you could run following command as requested by the error message

```
dpkg --configure -a
```

MichaIng, 2021-09-19:
pip aims to apply common best practice for development and when installing modules in user mode. When running as root user, modules are installed to /usr/local/, so are available system wide and override those in /lib and /bin. In theory it is indeed possible that native Debian APT packages require a specific Python module version, shipped by `python3-<module>` APT package while not working with the most recent module version shipped by pip by default. But it is a rare case in my experience. We install Python modules in system mode for many dietpi-software options and at least never got a report about a failing Debian package due to this. It was common practice before, now this changes towards local/user mode, like Rust/cargo and newest Go do it by default to avoid conflicts in the first place, with the downside that all dependencies need to be installed for every project separately.

MichaIng, 2021-10-20:
I'll mark this as closed now due to outstanding reply. Not sure why `dpkg` reported an interruption (like when you hit CTRL+C) on every upgrade attempt, probably an issue with the SSH client, not sure.

> I should be able to save the old image configuration and just import that image configureation with all the installed software. Now I am trying to install all the packaged.

This is btw not really possible. You can of course backup the whole system or clone the system drive and restore it, but that will restore the whole system, including possible issues you had. Common approach is to backup userdata and selected system/software configurations (`/mnt/dietpi_userdata`, `/boot/dietpi.txt`, selected files from `root`, `/home`, `/etc` and `/var` in case.), place those into a fresh system and then re-install the software titles/packages.

> The development package is NOT listed in the software I search and install python3 and pip3 but no development libraries like gcc etc.

The common C compiler package and headers is called "Build-Essential", or the APT package `build-essential`. Generally we do not provide every single APT package as an individual software option in `dietpi-software`. It is expected that especially for own compiling (that is not done as part of a `dietpi-software` install) you need to install the related compiler and in case development libraries and headers via `apt`, like you would do on every other Debian based distribution.