MichaIng/DietPi #4749: RPI Zero W with RealVNC not starting missing libbcm_host.so

Axellarator, 2021-09-18:
### Creating a bug report/issue

Sep 18 23:21:27 ZeroPi systemd[1]: Started VNC Server in Service Mode daemon.
Sep 18 23:21:27 ZeroPi vncserver-x11-serviced[2353]: /usr/bin/vncserver-x11-core: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
Sep 18 23:21:27 ZeroPi vncserver-x11-serviced[2352]: vncserver-x11-serviced: Server exited with status 127, restarting in 5s

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`- G_DIETPI_VERSION_CORE=7
``` 
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
``` 
bullseye 1
``` 
- Kernel version | `uname -a`
``` 
Linux ZeroPi 5.10.60+ #1449 Wed Aug 25 14:59:20 BST 2021 armv6l GNU/Linux
``` 
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
``` 
RPi Zero W (armv6l)
 ``` 
- Power supply used | (EG: 5V 1A RAVpower)
``` 
5V 3A Original Raspi
``` 
- SDcard used | (EG: SanDisk ultra)
``` 
SanDisk 64 
``` 
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
``` 
RealVNC Server
``` 
- Was the software title installed freshly or updated/migrated?
``` 
fresh install, then did a delete and install via DPI Software and later reinstall via cli
```   
- Can this issue be replicated on a fresh installation of DietPi?
``` 
yes, always the same message as above
```  

- Bug report ID | `echo $G_HW_UUID`
``` 
5e35d679-c0d2-4f29-b948-0b99a800d7d3
``` 
#### Steps to reproduce

``` 
1. ...Install RealVNC, tested under 7.5.2 and 7.6.2, same outcome
``` 
#### Expected behaviour

- ...

#### Actual behaviour

``` 
- root@ZeroPi:~# systemctl status vncserver-x11-serviced
● vncserver-x11-serviced.service - VNC Server in Service Mode daemon
     Loaded: loaded (/lib/systemd/system/vncserver-x11-serviced.service; disabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-09-18 23:21:27 CEST; 14min ago
   Main PID: 2352 (vncserver-x11-s)
      Tasks: 1 (limit: 877)
        CPU: 264ms
     CGroup: /system.slice/vncserver-x11-serviced.service
             └─2352 /usr/bin/vncserver-x11-serviced -fg

Sep 18 23:22:48 ZeroPi vncserver-x11-serviced[2413]: /usr/bin/vncserver-x11-core: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
Sep 18 23:22:48 ZeroPi vncserver-x11-serviced[2352]: vncserver-x11-serviced: Server exited with status 127, restarting in 5s
Sep 18 23:22:53 ZeroPi vncserver-x11-serviced[2419]: /usr/bin/vncserver-x11-core: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
Sep 18 23:22:53 ZeroPi vncserver-x11-serviced[2352]: vncserver-x11-serviced: Server exited with status 127, restarting in 5s
Sep 18 23:22:58 ZeroPi vncserver-x11-serviced[2420]: /usr/bin/vncserver-x11-core: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
Sep 18 23:22:58 ZeroPi vncserver-x11-serviced[2352]: vncserver-x11-serviced: Server exited with status 127, restarting in 5s
Sep 18 23:23:03 ZeroPi vncserver-x11-serviced[2426]: /usr/bin/vncserver-x11-core: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
Sep 18 23:23:03 ZeroPi vncserver-x11-serviced[2352]: vncserver-x11-serviced: Server exited with status 127, restarting in 5s
Sep 18 23:23:08 ZeroPi vncserver-x11-serviced[2427]: /usr/bin/vncserver-x11-core: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory
Sep 18 23:23:08 ZeroPi vncserver-x11-serviced[2352]: vncserver-x11-serviced: Server exited with status 127, restarting in 3600s
``` 

#### Extra details

``` 
- ...Dietpi-Services displays VNC-Server as active
```

Joulinar, 2021-09-18:
Hi,

This should be similar to #4729 

Pls could you run `apt update && apt upgrade` to update all packages. Try to restart service afterwards.

MichaIng, 2021-09-19:
RealVNC likely requires a rebuild as well, or alternatively the workaround. I'll also ping the RPi devs about it.

Axellarator, 2021-09-19:
+ I did the update/upgrade procedure, no changes in behaviour.
+ Did the stuff suggested in  __#4729__ ´´´ __ln -s libbcm_host.so.0 libbcm_host.so__ ´´´ outcome see below. Still no access via RealVNC viewer. Win10 shows only that the address could not be contacted. Tablet says, the listening port for a connection could not be contacted. Everything runs in a 192.168.178.0  environment and all other PI are connected.

## Output from Dietpi-Services

+ vncserver.service - Manage VNC Server (DietPi)      
```                              
  Loaded: loaded (/etc/systemd/system/vncserver.service; enabled; vendor preset: enabled)
  Active: active (exited) since Sun 2021-09-19 11:31:37 CEST; 8min ago  
  Main PID: 2427 (code=exited, status=0/SUCCESS)                                 
  Tasks: 0 (limit: 877)                                                                
  CPU: 0                                                                                   
  CGroup: /system.slice/vncserver.service                                                     
                                    
 Sep 19 11:31:37 ZeroPi systemd[1]: Started Manage VNC Server (DietPi).                         
 Sep 19 11:31:37 ZeroPi systemd[2427]: pam_unix(login:session): session opened for user  root(uid=0) by (uid=0)

```
## Identical output systemctl status vncserver

* vncserver.service - Manage VNC Server (DietPi)
``` 
    Loaded: loaded (/etc/systemd/system/vncserver.service; enabled; vendor preset: enabled)
     Active: active (exited) since Sun 2021-09-19 11:31:37 CEST; 24min ago
   Main PID: 2427 (code=exited, status=0/SUCCESS)
      Tasks: 0 (limit: 877)
        CPU: 0
     CGroup: /system.slice/vncserver.service

Sep 19 11:31:37 ZeroPi systemd[1]: Started Manage VNC Server (DietPi).
Sep 19 11:31:37 ZeroPi systemd[2427]: pam_unix(login:session): session opened for user root(uid=0) by (uid=0)

```

Joulinar, 2021-09-19:
Can you check what port VNC server is LISTEN on

```
ss -tulpn | grep LISTEN
```

Axellarator, 2021-09-19:
Did that
```
* cd /usr/lib/arm-linux-gnueabihf
  > ln -s libmmal.so.0 libmmal.so
  > ln -s libmmal_core.so.0 libmmal_core.so
  > ln -s libmmal_components.so.0 libmmal_components.so
  > ln -s libcontainers.so.0 libcontainers.so

* systemctl restart vncserver
* systemctl status vncserver-x11-serviced

  > Sep 19 13:58:31 ZeroPi systemd[1]: Started VNC Server in Service Mode daemon.
  > Sep 19 13:58:31 ZeroPi vncserver-x11[5032]: ServerManager: Server started
  > Sep 19 13:58:32 ZeroPi vncserver-x11[5032]: ConsoleDisplay: Found running X
  > server (pid=540, binary=/usr/lib/xorg/Xorg)

 vncserver-x11-serviced.service - VNC Server in Service Mode daemon
     Loaded: loaded (/lib/systemd/system/vncserver-x11-serviced.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2021-09-19 13:58:31 CEST; 5s ago
   Main PID: 5031 (vncserver-x11-s)
      Tasks: 6 (limit: 877)
        CPU: 2.303s
     CGroup: /system.slice/vncserver-x11-serviced.service
             ├─5031 /usr/bin/vncserver-x11-serviced -fg
             ├─5032 /usr/bin/vncserver-x11-core -service
             ├─5043 /usr/bin/vncagent service 14
             ├─5046 /usr/bin/vncserverui service 17
             └─5054 /usr/bin/vncserverui -statusicon 5
```

## ss -tulpn | grep LISTEN
```
tcp   LISTEN 0      5            0.0.0.0:5900      0.0.0.0:*    users:(("vncserver-x11-c",pid=5032,fd=11))
tcp   LISTEN 0      5            0.0.0.0:5999      0.0.0.0:*    users:(("vncserver-virtu",pid=308,fd=8))
tcp   LISTEN 0      1024         0.0.0.0:80        0.0.0.0:*    users:(("lighttpd",pid=747,fd=4))   
tcp   LISTEN 0      128          0.0.0.0:22        0.0.0.0:*    users:(("sshd",pid=430,fd=3))       
tcp   LISTEN 0      5               [::]:5900         [::]:*    users:(("vncserver-x11-c",pid=5032,fd=10))
tcp   LISTEN 0      5               [::]:5999         [::]:*    users:(("vncserver-virtu",pid=308,fd=7))
tcp   LISTEN 0      1024            [::]:80           [::]:*    users:(("lighttpd",pid=747,fd=5))   
tcp   LISTEN 0      128             [::]:22           [::]:*    users:(("sshd",pid=430,fd=4))  
```

Axellarator, 2021-09-19:
## With 192.168.178.202:5900 RealVNC runs now
* Shall I reinstall Realvnc with Dietpi-Launcher for testing ?

## Output vncser-x11.log
```
<14> 2021-09-19T11:58:31.923Z ZeroPi vncserver-x11[5032]: SFeatureRaspberryPi: Raspberry Pi edition on Raspberry Pi Zero W Rev 1.1
<14> 2021-09-19T11:58:31.958Z ZeroPi vncserver-x11[5032]: GoogleAnalytics: X.509 Error: Certificate expired
<14> 2021-09-19T11:58:32.077Z ZeroPi vncserver-x11[5032]: TcpListenerManager: listening on [::]::5900
<14> 2021-09-19T11:58:32.078Z ZeroPi vncserver-x11[5032]: TcpListenerManager: listening on 0.0.0.0::5900
<14> 2021-09-19T11:58:32.088Z ZeroPi vncserver-x11[5032]: SModuleRsaKey: Using new key: 31-d5-c1-18-b7-7a-5c-24 (catchphrase "Sample command collect. Chario>
<14> 2021-09-19T11:58:32.144Z ZeroPi vncserver-x11[5032]: getPrimaryAddress: (running /etc/vnc/get_primary_ip4) command<pid=5035> failed
<14> 2021-09-19T11:58:32.147Z ZeroPi vncserver-x11[5032]: SModuleCtrlCommsListen: listening on /tmp/.vnc-vncservice/vncserver-x11.CtrlComms
<13> 2021-09-19T11:58:32.152Z ZeroPi vncserver-x11[5032]: ConsoleDisplay: Found running X server (pid=540, binary=/usr/lib/xorg/Xorg)
<14> 2021-09-19T11:58:32.153Z ZeroPi vncserver-x11[5032]: AgentMgrService: consoleDisplayChanged, running 1
<14> 2021-09-19T11:58:32.153Z ZeroPi vncserver-x11[5032]: AgentMgrLauncher: Restarting agent in 0ms
<14> 2021-09-19T11:58:32.257Z ZeroPi vncserver-x11[5032]: SDesktopAgent: agent connected
<14> 2021-09-19T11:58:32.258Z ZeroPi vncserver-x11[5032]: UiProcess: Restart in 500ms
<14> 2021-09-19T11:58:32.508Z ZeroPi vncserver-x11[5032]: UiProcess: Restart in 500ms
<14> 2021-09-19T11:58:32.509Z ZeroPi vncserver-x11[5032]: Agent: Randr: XRandr extension present - version 1.5
<14> 2021-09-19T11:58:32.509Z ZeroPi vncserver-x11[5032]: Agent: SDesktopX11: XFixes extension present - version 5.0
<14> 2021-09-19T11:58:32.510Z ZeroPi vncserver-x11[5032]: Agent: SDesktopX11: XDamage extension present - version 1.1
<14> 2021-09-19T11:58:32.510Z ZeroPi vncserver-x11[5032]: Agent: TXClipboardMgr: XFixes extension present - version 5.0
<14> 2021-09-19T11:58:32.510Z ZeroPi vncserver-x11[5032]: Agent: DesktopUidMonitor: Starting desktop uid monitor
<14> 2021-09-19T11:58:32.510Z ZeroPi vncserver-x11[5032]: Agent: DesktopUidMonitor: Desktop user is 0
<14> 2021-09-19T11:58:33.026Z ZeroPi vncserver-x11[5032]: UiProcess: UI process started
<14> 2021-09-19T11:58:33.027Z ZeroPi vncserver-x11[5032]: SModulePrint: set printer (none) as default
<14> 2021-09-19T11:58:33.175Z ZeroPi vncserver-x11[5032]: getPrimaryAddress: (running /etc/vnc/get_primary_ip4) command<pid=5047> failed
<14> 2021-09-19T12:28:33.169Z ZeroPi vncserver-x11[5032]: getPrimaryAddress: (running /etc/vnc/get_primary_ip4) command<pid=6017> failed
<14> 2021-09-19T12:32:50.740Z ZeroPi vncserver-x11[5032]: getPrimaryAddress: (running /etc/vnc/get_primary_ip4) command<pid=6155> failed

Joulinar, 2021-09-19:
Do you have a shared desktop active? Means autologin into a desktop? At least VNC is running on port 5900.

Axellarator, 2021-09-19:
Yes, LXDE is accessible, thanks for your help

Joulinar, 2021-09-19:
Does it mean VNC is working again?

Axellarator, 2021-09-19:
Yes, VNC is working and LXDE, too. It was running, but no right mouseclicks available and software not or with huge delay starting. As of now everything is fine.

Joulinar, 2021-09-19:
OK can we close the issue if it is working fine now?

Axellarator, 2021-09-19:
Yes, close it

Axellarator, 2021-09-20:
Did a DietPI Install on a Raspberry B from 2012. RealVNC died as expected. Did the ```ln -s´`` as above, RealVNC still dead. Did the following steps and RealVNC start working. The workaround is now complete.

```
# apt reinstall realvnc-vnc-server
# systemctl start vncserver-x11-serviced.service
# systemctl enable vncserver-x11-serviced.service
> Created symlink /etc/systemd/system/multi-user.target.wants/vncserver-x11-serviced.service → /lib/systemd/system/vncserver-x11-serviced.service.
# systemctl start vncserver-virtuald.service
# systemctl enable vncserver-virtuald.service
> Created symlink /etc/systemd/system/multi-user.target.wants/vncserver-virtuald.service → /lib/systemd/system/vncserver-virtuald.service.
# ss -tulpn | grep LISTEN
>>  tcp   LISTEN 0      5        0.0.0.0:5900   0.0.0.0:*    users:(("vncserver-x11-c",pid=2485,fd=11))
>>  tcp   LISTEN 0      5        0.0.0.0:5999   0.0.0.0:*    users:(("vncserver-virtu",pid=2522,fd=8))
>>  tcp   LISTEN 0      5        [::]:5900         [::]:*           users:(("vncserver-x11-c",pid=2485,fd=10))
>>  tcp   LISTEN 0      5        [::]:5999         [::]:*           users:(("vncserver-virtu",pid=2522,fd=7))
```

closed 2021-09-20 by Axellarator

MichaIng, 2021-09-21:
@Axellarator 
`vncserver-virtuald.service` spawns a virtual desktop automatically once a VNC client tries to connect, which is a nice feature, but it is only available with a RealVNC premium account, as it has no effect. Since you obviously connect to a local/shared desktop, you won't need this service anyway.

`vncserver-x11-serviced.service` is RealVNC's own shared desktop VNC service. If you installed RealVNC via `dietpi-software`, then the same should work via our own `vncserver.service`. If this didn't work, you can check via:
```sh
journalctl -u vncserver
```

If the shared desktop is what you want, then whether using `vncserver-x11-serviced.service` or `vncserver.service` makes no difference, the letter allows to switch between virtual and shared desktop via `dietpi.txt` setting and aligns it with TigerVNC, that's all. But since both services conflict, you should disable the one you do not use.