MichaIng/DietPi #4750: DietPi update from 7.5.2 to 7.6.2

benoit2021, 2021-09-19:
#### Details:
- Date           | Sun Sep 19 18:29:38 CEST 2021
- Bug report     | 18c5a170-abd6-4840-8e07-8d451fc3e814
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -qq upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Packages were downgraded and -y was used without --allow-downgrades.
```

Joulinar, 2021-09-19:
Pls can you share full output of `apt update && apt upgrade`

benoit2021, 2021-09-19:
**Here is the output during diet-update:**
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Get:1 http://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Hit:2 https://deb.nodesource.com/node_12.x buster InRelease
Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Fetched 61.7 kB in 2s (34.9 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
E: Packages were downgraded and -y was used without --allow-downgrades.
[FAILED] DietPi-Update | APT upgrade

**And if I run apt update && apt upgrade manually:**
root@DietPi:~# apt update && apt upgrade
Atteint :1 http://deb.debian.org/debian buster-backports InRelease
Réception de :2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15,0 kB]                                                                     
Atteint :3 https://deb.nodesource.com/node_12.x buster InRelease                                                                                        
Atteint :4 https://archive.raspberrypi.org/debian buster InRelease                                          
Réception de :5 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13,0 MB]
13,0 Mo réceptionnés en 6s (2 141 ko/s)                                                                                                                 
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Tous les paquets sont à jour.
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Calcul de la mise à jour... Fait
Les paquets suivants seront mis à une VERSION INFÉRIEURE :
  cmake-data
0 mis à jour, 0 nouvellement installés, 1 remis à une version inférieure, 0 à enlever et 20 non mis à jour.
Il est nécessaire de prendre 1 628 ko dans les archives.
Après cette opération, 0 o d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] O
Réception de :1 https://archive.raspberrypi.org/debian buster/main armhf cmake-data all 3.16.3-3~bpo10+1 [1 628 kB]
1 628 ko réceptionnés en 0s (5 060 ko/s)
(Lecture de la base de données... 67316 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../cmake-data_3.16.3-3~bpo10+1_all.deb ...
Dépaquetage de cmake-data (3.16.3-3~bpo10+1) sur (3.16.3-3~bpo10+1) ...
Paramétrage de cmake-data (3.16.3-3~bpo10+1) ...
Traitement des actions différées (« triggers ») pour man-db (2.8.5-2) ...
root@DietPi:~#

Joulinar, 2021-09-19:
Basically same issue as #4358

Solution would be to purge `cmake` as follow

```
apt purge --autoremove cmake cmake-data
```

benoit2021, 2021-09-19:
Actually I just add "--allow-downgrades" option to the command line using the diepi-update interface, and the update finally succeeded.

Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://deb.debian.org/debian buster-backports InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 https://deb.nodesource.com/node_12.x buster InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
E: Packages were downgraded and -y was used without --allow-downgrades.
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
[ INFO ] DietPi-Update | Executing alternative command: apt-get -qq upgrade --allow-downgrades
(Reading database ... 67316 files and directories currently installed.)
Preparing to unpack .../cmake-data_3.16.3-3~bpo10+1_all.deb ...
Unpacking cmake-data (3.16.3-3~bpo10+1) over (3.16.3-3~bpo10+1) ...
Setting up cmake-data (3.16.3-3~bpo10+1) ...
Processing triggers for man-db (2.8.5-2) ...
[  OK  ] Alternative command execution | Completed

closed 2021-09-19 by benoit2021

Joulinar, 2021-09-19:
I guess issue still persists and you might run into it next time again. Can you check if `apt update && apt upgrade` is running without issues?

benoit2021, 2021-09-19:
Oups even if I am nor running DietPi v7.6.2, indeed each time I run `apt update && apt upgrade` I got:
root@DietPi:~# apt update && apt upgrade
Atteint :1 http://deb.debian.org/debian buster-backports InRelease
Atteint :2 https://deb.nodesource.com/node_12.x buster InRelease                                                                                                   
Atteint :3 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                                               
Atteint :4 https://archive.raspberrypi.org/debian buster InRelease                         
Lecture des listes de paquets... Fait                    
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Tous les paquets sont à jour.
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Calcul de la mise à jour... Fait
Les paquets suivants seront mis à une VERSION INFÉRIEURE :
  cmake-data
0 mis à jour, 0 nouvellement installés, 1 remis à une version inférieure, 0 à enlever et 20 non mis à jour.
Il est nécessaire de prendre 1 628 ko dans les archives.
Après cette opération, 0 o d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] O
Réception de :1 https://archive.raspberrypi.org/debian buster/main armhf cmake-data all 3.16.3-3~bpo10+1 [1 628 kB]
1 628 ko réceptionnés en 4s (366 ko/s)      
(Lecture de la base de données... 67316 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../cmake-data_3.16.3-3~bpo10+1_all.deb ...
Dépaquetage de cmake-data (3.16.3-3~bpo10+1) sur (3.16.3-3~bpo10+1) ...
Paramétrage de cmake-data (3.16.3-3~bpo10+1) ...
Traitement des actions différées (« triggers ») pour man-db (2.8.5-2) ...
root@DietPi:~# apt update && apt upgrade
Atteint :1 http://deb.debian.org/debian buster-backports InRelease
Atteint :2 https://deb.nodesource.com/node_12.x buster InRelease                                                                                                    
Atteint :3 http://raspbian.raspberrypi.org/raspbian buster InRelease                                                                                                
Atteint :4 https://archive.raspberrypi.org/debian buster InRelease                          
Lecture des listes de paquets... Fait                     
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Tous les paquets sont à jour.
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Calcul de la mise à jour... Fait
Les paquets suivants seront mis à une VERSION INFÉRIEURE :
  cmake-data
0 mis à jour, 0 nouvellement installés, 1 remis à une version inférieure, 0 à enlever et 20 non mis à jour.
Il est nécessaire de prendre 1 628 ko dans les archives.
Après cette opération, 0 o d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] O
Réception de :1 https://archive.raspberrypi.org/debian buster/main armhf cmake-data all 3.16.3-3~bpo10+1 [1 628 kB]
1 628 ko réceptionnés en 4s (455 ko/s)      
(Lecture de la base de données... 67316 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../cmake-data_3.16.3-3~bpo10+1_all.deb ...
Dépaquetage de cmake-data (3.16.3-3~bpo10+1) sur (3.16.3-3~bpo10+1) ...
Paramétrage de cmake-data (3.16.3-3~bpo10+1) ...
Traitement des actions différées (« triggers ») pour man-db (2.8.5-2) ...

Joulinar, 2021-09-19:
Yes the loop is caused by the Raspbian package `cmake-data_3.16.3-3~bpo10+1_all.deb`. Way out is to purge the package as stated above.

benoit2021, 2021-09-19:
You are absolutely right, I ran `apt purge --autoremove cmake cmake-data` and everything works fine now.
Thanks a lot !

Joulinar, 2021-09-19:
Yep, Not sure what you used `cmake` for. Probably for compiling some software. Keep that in mind if things might fail later on. Just in case 😉

benoit2021, 2021-09-19:
It was for compiling RPiPlay, I'll remember ;-) tthx again.

MichaIng, 2021-09-21:
For the record: Issue has been reported to RPi repo maintainers: #245