MichaIng/DietPi #4751: DietPi-Drive_Manager | Can't mount Samba share

ravenclaw900, 2021-09-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bullseye 0
```
- Kernel version | `uname -a`
```
Linux DietPi 5.10.60-v8+ #1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
RPi 4 Model B (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)
```
Canakit
```
- SDcard used | (EG: SanDisk ultra)
```
SanDisk Ultra 32GiB
```

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Samba Client / DietPi Drive Manger
- Was the software title installed freshly or updated/migrated?
Installed freshly
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce

1. Try mounting a Samba share in DietPi Drive Manger (server is on macOS).

#### Expected behaviour

- Share mounts

#### Actual behaviour

- Fails with:
```
│ mount error(95): Operation not supported                                     │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log  │
│ messages (dmesg)                                                             │
│ mount error(13): Permission denied                                           │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log  │
│ messages (dmesg)                                                             │
│ mount error(13): Permission denied                                           │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log  │
│ messages (dmesg)                                                             │
│ mount error(13): Permission denied                                           │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log  │
│ messages (dmesg)                                                             │
│ mount error(13): Permission denied                                           │
│ Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) and kernel log  |
│ messages (dmesg) 
```

#### Extra details

```
root@DietPi:~# dmesg -l err,crit,alert,emerg
[    1.921769] sd 0:0:0:0: [sda] No Caching mode page found
[    1.921779] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  790.106337] CIFS: VFS: \\192.168.1.160 Dialect not supported by server. Consider  specifying vers=1.0 or vers=2.0 on mount for accessing older servers
[  790.106413] CIFS: VFS: cifs_mount failed w/return code = -95
[  790.236957] CIFS: VFS: \\192.168.1.160 Send error in SessSetup = -13
[  790.237027] CIFS: VFS: cifs_mount failed w/return code = -13
[  790.284700] CIFS: VFS: \\192.168.1.160 Send error in SessSetup = -13
[  790.284758] CIFS: VFS: cifs_mount failed w/return code = -13
[  790.342661] CIFS: VFS: \\192.168.1.160 Send error in SessSetup = -13
[  790.342714] CIFS: VFS: cifs_mount failed w/return code = -13
[  790.357254] CIFS: VFS: Use of the less secure dialect vers=1.0 is not recommended unless required for access to very old servers
[  790.399315] CIFS: VFS: \\192.168.1.160 Send error in SessSetup = -13
[  790.399380] CIFS: VFS: cifs_mount failed w/return code = -13
```

Joulinar, 2021-09-19:
Best to my knowledge you need to set something specific in macOS to allow the mount. It was a misleading setting if I recall correctly. Something like allow Windows. At least we had such a case some weeks back.

ravenclaw900, 2021-09-19:
Indeed, after enabling that it works. For anyone with the same problem in the future, the setting is under `Options... > Windows File Sharing`.

Joulinar, 2021-09-19:
Ok going to close this issue for now.

closed 2021-09-19 by Joulinar

ravenclaw900, 2021-09-21:
Probably, I can't imagine that this is too uncommon.