MichaIng/DietPi #4752: DietPi-LetsEncrypt error when trying renew certificate ( sed -e ... )

billouetaudrey, 2021-09-19:
#### Details:
- Date           | Sun Sep 19 19:48:43 CEST 2021
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux raspberrypi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `sed -i s/'http://localhost/nextcloud'/'https://***.fr/nextcloud'/ /var/www/nextcloud/config/config.php`
- Exit code      | 1
- Software title | DietPi-LetsEncrypt
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
sed: -e expression #1, char 11: unknown option to `s'
```
Hi,

I have this bug when want renew my certificates, any help please ?

Joulinar, 2021-09-19:
Probably it's caused by this code change

https://github.com/MichaIng/DietPi/commit/9909a4f750c1ed50719d74b3dbc995fc4abee992

@MichaIng 
Can you have a look

MichaIng, 2021-09-19:
Yes bug our end. Please select "change command" and replace it with "true". I'll send a live patch.

billouetaudrey, 2021-09-19:
Work, thanks very much :)

Joulinar, 2021-09-19:
@billouetaudrey 
The command `true` is bypassing the issue. A fix will be provided via live patch later on.

billouetaudrey, 2021-09-19:
Yes it's what I understand lol 

Will try when fix will me provided

MichaIng, 2021-09-21:
Solved with: https://github.com/MichaIng/DietPi/commit/ae86046166df1f97d0c1ed404b5913e0196d1211
Changelog: https://github.com/MichaIng/DietPi/blob/dev/CHANGELOG.txt
Live patch: #4764