MichaIng/DietPi #4753: MotionEye problem on bullseye

cocoflan, 2021-09-19:


cocoflan, 2021-09-19:
Maybe use docker version of motioneye for the moment till it is fixed.

https://github.com/ccrisan/motioneye/wiki/Install-in-Docker

Tested and works on bullseye with docker on an x86 running bullseye 7.6.2, maybe a temporary solution.

Greetings

MichaIng, 2021-09-21:
The Docker container also ships deprecated Python 2. If we wanted to, we could install Python 2 `apt install python2` and get it running without a container, but I'm not keen to ship EOL software to our end users.

If the project doesn't make the step to Python 3, as sad it is, but then we won't support it anymore and need to look for an alternative project or fork. To me it currently looks like the maintainer doesn't have time to maintain the project. More and more Python 3 fixes are provided by volunteers (including me) to various Python 3 fixes, but nothing happens for quite a while. Also the (Python 2) development branch hasn't seen any commit since more than half a year. If I had time, I'd create a fork and ask all Python 3 branch contributors to switch their PRs over, as we did for HTPC Manager.

Of course everyone is free to install the Docker container or Python 2 manually, it's just not something that we can or want to provide as a supported solution our end.

MichaIng, 2022-03-20:
Done for next release, using the Python 3 compatible development branch of the project: https://github.com/motioneye-project/motioneye

Until there is a stable release available at PyPI, it is marked as "beta", but testing I did the last days was very promising.

closed 2022-03-26 by MichaIng