MichaIng/DietPi #4755: Dietpi-upgrade v7.5.2 - v7.6.2 Failed

Scoobdriver, 2021-09-20:
#### Details:
- Date           | Mon Sep 20 11:35:29 BST 2021
- Bug report     | f056260d-7dc4-42ca-955f-3ee409623dc8
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Debian mini.iso
- Hardware       | Virtual Machine (x86_64) (ID=20)
- Kernel version | `Linux unifi 4.19.0-17-amd64 #1 SMP Debian 4.19.194-3 (2021-07-18) x86_64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `apt-get -qq upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...run Dietpi-upgrade
2. ...
#### Expected behaviour:

- ...dietpi updates 
#### Actual behaviour:

- ... failing 
```
- ssh.service - OpenBSD Secure Shell server
   Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-09-20 11:35:29 BST; 5ms ago
     Docs: man:sshd(8)
           man:sshd_config(5)
  Process: 6714 ExecStartPre=/usr/sbin/sshd -t (code=exited, status=0/SUCCESS)
  Process: 6715 ExecStart=/usr/sbin/sshd -D $SSHD_OPTS (code=exited, status=255/EXCEPTION)
 Main PID: 6715 (code=exited, status=255/EXCEPTION)

Sep 20 11:35:29 unifi systemd[1]: Starting OpenBSD Secure Shell server...
Sep 20 11:35:29 unifi sshd[6715]: error: Bind to port 22 on 0.0.0.0 failed: Address already in use.
Sep 20 11:35:29 unifi sshd[6715]: error: Bind to port 22 on :: failed: Address already in use.
Sep 20 11:35:29 unifi sshd[6715]: fatal: Cannot bind any address.
Sep 20 11:35:29 unifi systemd[1]: ssh.service: Main process exited, code=exited, status=255/EXCEPTION
Sep 20 11:35:29 unifi systemd[1]: ssh.service: Failed with result 'exit-code'.
Sep 20 11:35:29 unifi systemd[1]: Failed to start OpenBSD Secure Shell server.
dpkg: error processing package openssh-server (--configure):
 installed openssh-server package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of ssh:
 ssh depends on openssh-server (>= 1:7.9p1-10+deb10u2); however:
  Package openssh-server is not configured yet.

dpkg: error processing package ssh (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 openssh-server
 ssh
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.
```
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
```
Setting up openssh-server (1:7.9p1-10+deb10u2) ...
rescue-ssh.target is a disabled or a static unit, not starting it.
Job for ssh.service failed because the control process exited with error code.
See "systemctl status ssh.service" and "journalctl -xe" for details.
invoke-rc.d: initscript ssh, action "start" failed.

MichaIng, 2021-09-20:
There seem to be another SSH server listening on port 22 already. Please check:
```sh
ss -tulpn
```

Scoobdriver, 2021-09-20:
> There seem to be another SSH server listening on port 22 already. Please check:
> 
> ```shell
> ss -tulpn
> ```

I guess this is the line we are interested in ? 
dropbear is running. I think this is the recommended ssh server 

tcp   LISTEN 0      128             [::]:22             [::]:*               users:(("dropbear",pid=513,fd=4))

Joulinar, 2021-09-20:
Did you installed `openssh-server` manually?

Scoobdriver, 2021-09-20:
> Did you installed `openssh-server` manually?

Not that I am aware of unless it was installed as part of the script I used to install my UniFi controller.

Scoobdriver, 2021-09-20:
OK I removed openssh-server with  
apt remove openssh-server

Thanks You
This now allows me to update. 
Should I mark this as closed ?

Joulinar, 2021-09-20:
I mark the issue as closed now. Feel free to reopen if needed.

closed 2021-09-20 by Joulinar