MichaIng/DietPi #4756: Filebrowser login issue

barukh27, 2021-09-20:
Hi there!

I've installed filebrowser after the las dietpi's update. Everything works fine but I can't log in. I've tried some credentials:

admin-admin
dietpi-dietpi
filebrowser-filebrowser
dietpi-(my frequent password)
dietpi-admin
dietpi-filebrowser

What should I try?

Thanks in advance.

MichaIng, 2021-09-20:
It should be `dietpi` and the global software password you at at first boot or changed manually via dietpi-software, and it defaults to `dietpi` as well if you didn't change it.

You can change the password via CLI, but we need to find the exact command. As a start:
```sh
/opt/filebrowser/filebrowser -h users
```

barukh27, 2021-09-20:
It worked now! 

Thanks!!!

closed 2021-09-20 by barukh27