MichaIng/DietPi #4757: Mosquitto fails to install from `dietpi-software`

dnknth, 2021-09-20:
#### Details:
- Date           | Mon Sep 20 20:50:44 CEST 2021
- Bug report     | 681a98cb-8801-4d70-8223-e07df3263f90
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -qq --allow-change-held-packages install mosquitto`
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

- Attempt to install `mosquitto` via `dietpi-software`

#### Expected behaviour:

- Installation should succeed

#### Actual behaviour:

- Installation fails due to missing dependency `libwebsocket18`

#### Additional logs:

```
The following packages have unmet dependencies:
 mosquitto : Depends: libwebsockets18 (>= 4.2.1) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Joulinar, 2021-09-20:
There seems to be an issue with the `libwebsocket18` package on the Mosquitto `apt` repository. Even that the package exist on the server https://repo.mosquitto.org/debian/pool/main/libw/libwebsockets/

It did not shown up as package to be installed

```
root@DietPi3:/tmp# apt show libwebsockets18
Package: libwebsockets18
State: not a real package (virtual)
N: Can't select candidate version from package libwebsockets18 as it has no candidate
N: Can't select versions from package 'libwebsockets18' as it is purely virtual
N: No packages found
root@DietPi3:/tmp#
```

A workaround would be to install the package manually before install Mosquitto again

```
cd /tmp/
wget  https://repo.mosquitto.org/debian/pool/main/libw/libwebsockets/libwebsockets18_4.2.1-0mosquitto1~bullseye1_armhf.deb
dpkg -i libwebsockets18_4.2.1-0mosquitto1~bullseye1_armhf.deb
rm libwebsockets18_4.2.1-0mosquitto1~bullseye1_armhf.deb
dietpi-software install 123
```

dnknth, 2021-09-20:
Wow! Thanks for the quick response.
I'll try it, but would like to leave the issue open until the workaround is not needed any more.
OK?

Joulinar, 2021-09-20:
well not sure if this is something we could influence as for me it seems a challenge of Mosquitto apt repository. I guess @MichaIng could qualify it way better than I 😄

dnknth, 2021-09-21:
I can confirm the workaround. Mosquito is up and running. Thanks for saving my day.