MichaIng/DietPi #4759: mjpg-streamer timeout fails to install (dietpi-software)

kb8080, 2021-09-21:
#### Details:
- Date           | Mon Sep 20 19:56:48 EDT 2021
- Bug report     | 5953697c-e7a1-461c-8089-56e13a618f45
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | `Linux util 4.14.241+ #1 SMP PREEMPT Wed Jul 28 16:55:16 UTC 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `curl -ILfvm 10 https://github.com/jacksonliam/mjpg-streamer/archive/master.tar.gz`
- Exit code      | 28
- Software title | DietPi-Software
#### Steps to reproduce:

1. Installed via dietpi-software
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
* Expire in 0 ms for 6 (transfer 0x516758)
* Expire in 10000 ms for 8 (transfer 0x516758)
* Expire in 1 ms for 1 (transfer 0x516758)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 0 ms for 1 (transfer 0x516758)
* Expire in 2 ms for 1 (transfer 0x516758)
[... 683 lines cut ...]
*  issuer: C=US; O=DigiCert, Inc.; CN=DigiCert High Assurance TLS Hybrid ECC SHA256 2020 CA1
*  SSL certificate verify ok.
* Operation timed out after 5179 milliseconds with 0 out of 0 bytes received
  0     0    0     0    0     0      0      0 --:--:--  0:00:10 --:--:--     0
* Closing connection 1
curl: (28) Operation timed out after 5179 milliseconds with 0 out of 0 bytes received
HTTP/2 302
server: GitHub.com
date: Mon, 20 Sep 2021 23:56:32 GMT
content-type: text/html; charset=utf-8
vary: X-PJAX, X-PJAX-Container, Accept-Encoding, Accept, X-Requested-With
permissions-policy: interest-cohort=()
location: https://codeload.github.com/jacksonliam/mjpg-streamer/tar.gz/master
cache-control: max-age=0, private
strict-transport-security: max-age=31536000; includeSubdomains; preload
x-frame-options: deny
x-content-type-options: nosniff
x-xss-protection: 0
referrer-policy: no-referrer-when-downgrade
expect-ct: max-age=2592000, report-uri="https://api.github.com/_private/browser/errors"
content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.githubassets.com; connect-src 'self' uploads.github.com www.githubstatus.com collector.githubapp.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events translator.github.com wss://alive.github.com *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations raw.githubusercontent.com github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com; frame-ancestors 'none'; frame-src render.githubusercontent.com viewscreen.githubusercontent.com; img-src 'self' data: github.githubassets.com identicons.github.com collector.githubapp.com github-cloud.s3.amazonaws.com secured-user-images.githubusercontent.com/ *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.githubassets.com github.com/socket-worker-0af8a29d.js gist.github.com/socket-worker-0af8a29d.js
content-length: 133
x-github-request-id: 5C1C:43CC:2D303B:4A2E4D:61491FBB

```

Joulinar, 2021-09-21:
Hi,

are you able to open `https://github.com/jacksonliam/mjpg-streamer/archive/master.tar.gz` locally on a desktop computer?

kb8080, 2021-09-21:
Hello, Yes I am able to download and extract.

Joulinar, 2021-09-21:
could you try following on command line.

`curl -ILfvm 10 https://github.com/jacksonliam/mjpg-streamer/archive/master.tar.gz`

kb8080, 2021-09-21:
I get the same output pretty much with the end result being "curl: (28) Operation timed out after 5166 milliseconds with 0 out of 0 bytes received"

kb8080, 2021-09-21:
Another update. I changed the curl timeout from 10 to 20 and it completes as expected. So I went back into the automated software installer and when it failed again I edited the command via the developer option to use the value of 20 instead of 10 for the curl command and it completes successfully.

Joulinar, 2021-09-21:
strange that that it takes that long for `curl` to complete the request. 🤔

Joulinar, 2021-09-24:
@kb8080 
I guess issue has been fixed and topic can be closed?

kb8080, 2021-09-24:
Yes I think it can be. Still not sure why this particular app had the
timeout issue while others are fine.

closed 2021-09-25 by Joulinar