MichaIng/DietPi #4760: Dietpi Update failed

tauchvogel, 2021-09-21:
#### Details:
- Date           | Tue Sep 21 09:19:07 CEST 2021
- Bug report     | c7517bf8-1236-42f0-999c-9d0c0a708625
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `curl -sSfLO https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches`
- Exit code      | 6
- Software title | DietPi-Update
#### Steps to reproduce:

#### Actual behaviour:

-  Dietpi told me to update to a newer Version. I tried to do it twice. But it stopped after a few seconds and told the error.
#### Additional Info:
-RPi 4 Model B (armv7l) | IP: 192.168.178.40 | Use up/down butt
│ APT update
│  - Command: apt-get --allow-releaseinfo-change-suite -q
│ update
│  - Exit code: 255
│  - DietPi version: v7.5.2 (MichaIng/master) | HW_MODEL: 4
│ | HW_ARCH: 2 | DISTRO: 6
│  - Image creator: DietPi Core Team
│  - Pre-image: from scratch
│  - Error log:
│ Err:1 http://raspbian.raspberrypi.org/raspbian bullseye
│ InRelease
│   Temporary failure resolving 'raspbian.raspberrypi.org'
│ Err:2 https://download.docker.com/linux/raspbian bullseye
│ InRelease
│   Temporary failure resolving 'download.docker.com'
│ Err:3 https://archive.raspberrypi.org/debian bullseye
│ InRelease
│   Temporary failure resolving 'archive.raspberrypi.org'
│ Reading package lists...
│ W: Failed to fetch
│ http://raspbian.raspberrypi.org/raspbian/dists/bullseye/In
│ Release  Temporary failure resolving
│ 'raspbian.raspberrypi.org'
curl: (6) Could not resolve host: raw.githubusercontent.com
```

Joulinar, 2021-09-21:
Hi,

this looks like a local issue with your DNS name resolution as you are not able to resolve global domains like `raw.githubusercontent.com`, `download.docker.com` and `raspbian.raspberrypi.org`

Pls check your local settings if DNS is working fine. Maybe choose a different DNS server.

tauchvogel, 2021-09-21:
Now, I choosed a different DNS Server via "dietpi-config 8 1" in the terminal. I have made the commands like in the dietpi posting on this page (https://dietpi.com/docs/software/dns_servers/) - section PiHole -> Configuration.

Now I've got these message :
`WARNING: Concurrent execution of DietPi-Update detected                                                              │ 
                                                     │                                                                                                                      │ 
                                                     │ Please check if one of the following applies:                                                                        │ 
                                                     │  - This script already runs on another terminal/SSH session.                                                         │ 
                                                     │  - Currently a cron or systemd background job executes the script.                                                   │ 
                                                     │  - You started this script from within another DietPi program, causing a loop.                                       │ 
                                                     │                                                                                                                      │ 
                                                     │ Please assure that the concurrent execution has finished, before retrying, otherwise cancel this instance.           │ 
                                                     │                                                                                                                      │ 
                                                     │ The following info might help:                                                                                       │ 
                                                     │   221 root     ?        /usr/bin/vmtouch -l /boot/dietpi /boot/dietpi.txt                                            │ 
                                                     │  2331 root     pts/1     |       \_ /bin/bash /boot/dietpi/dietpi-update                                             │ 
                                                     │  2363 root     pts/1     |           \_ /bin/bash /boot/dietpi/dietpi-update                                         │ 
                                                     │  2364 root     pts/1     |               \_ whiptail --title DietPi-Update --backtitle RPi 4 Model B (armv7l) | IP:  │ 
                                                     │ 192.168.178.40 | Use up/down buttons to scroll text --menu APT update                                                │ 
                                                     │  - Command: apt-get --allow-releaseinfo-change-suite -q update  - Exit code: 255  - DietPi version: v7.5.2           │ 
                                                     │ (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 2 | DISTRO: 6  - Image creator: DietPi Core Team                          │ 
                                                     │  - Pre-image: from scratch                                                                                           │ 
                                                     │  - Error log: Err:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease   Temporary failure resolving        │ 
                                                     │ 'raspbian.raspberrypi.org' Err:2 https://download.docker.com/linux/raspbian bullseye InRelease   Temporary failure   │ 
                                                     │ resolving 'download.docker.com' Err:3 https://archive.raspberrypi.org/debian bullseye InRelease   Temporary failure  │ 
                                                     │ resolving 'archive.raspberrypi.org' Reading package lists... W: Failed to fetch                                      │ 
                                                     │ http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  Temporary failure resolving                       │ 
                                                     │ 'raspbian.raspberrypi.org' W: Failed to fetch https://download.docker.com/linux/raspbian/dists/bullseye/InRelease    │ 
                                                     │ Temporary failure resolving 'download.docker.com' W: Failed to fetch                                                 │ 
                                                     │ https://archive.raspberrypi.org/debian/dists/bullseye/InRelease  Temporary failure resolving                         │ 
                                                     │ 'archive.raspberrypi.org' W: Some index files failed to download. They have been ignored, or old ones used instead.  │ 
                                                     │ --ok-button Ok --cancel-button Exit --default-item Retry --scrolltext 32 62 2 Retry : Re-run the last command that   │ 
                                                     │ failed DietPi-Config : Edit network, APT/NTP mirror settings etc Open subshell : Open a subshell to investigate or   │ 
                                                     │ solve the issue Send report : Uploads bugreport containing system info to DietPi  ●─ Devs only                       │ 
                                                     │ ──────────────────────────────────────● Change command : Adjust and rerun the command                                │ 
                                                     │  4945 root     pts/2             \_ /bin/bash /boot/dietpi/dietpi-update                         `

What can I do?

Joulinar, 2021-09-21:
you have another open SSH session running `dietpi-update`. Pls close the other session first.

MichaIng, 2021-10-20:
I'll mark this as closed due to outstanding reply. As Joulinar stated: There was an instance of dietpi-update running already, probably at a different SSH session (`pts/2` is an SSH pseudo terminal device).

Generally to get an overview over running processes: `htop`

> I have made the commands like in the dietpi posting on this page (https://dietpi.com/docs/software/dns_servers/) - section PiHole -> Configuration.

Pi-hole is generally NOT intended to solve or provide local DNS resolving and by default it is not configured to be used by the server itself. It is meant as ad blocker for the network clients only. If DNS resolving on the server fails, then selecting a different upstream (public or e.g. your router to benefit from a local cache) can be tried, and checking the network connectivity itself:
```sh
G_CHECK_CON
G_CHECK_DNS
```

closed 2021-10-20 by MichaIng