MichaIng/DietPi #4765: Odroid C4/HC4 image will boot into petitboot but not load system on SD card on HC4 SBC

wdirksen, 2021-09-22:
Diet PI will boot, but unable to load system files and is essentially DOA following a successful boot sequence

- Using current DietPi_OdroidC4-ARMv8-Buster.img on Hardkernel Odroid HC4 with factory power supply
- This Odroid HC4 is currently running well on a custom ArchLinux ARM aarch64 system using SD card so it works
- Copied DietPi image to SD card with Rufus on Windows 10 (works for all other DietPi devices I've tried)
- The issue was replicated on 2 different SD cards

Thanks, would like to switch over to DietPi

Wes

wdirksen, 2021-09-22:
Will try Armbian with PREP_SYSTEM_FOR_DIETPI.sh

MichaIng, 2021-09-22:
Many thanks for your report.

Hmm, what do you mean by "but unable to load system files and is essentially DOA", do you have any related screen output, errors, logs or so?

wdirksen, 2021-09-22:
It's hard to know what is going on. After petitboot countdown and the system should startup:
- HDMI goes black
- There is no IP on LAN so SSH not possible
- I took out SD to read log files on another machine but couldn't find it

Is there something you would want me to post that I can from the SD card?
Would it help to post the petitboot files from the ArchArm system that is working?

BTW: Same problem with Armbian HC4 image

MichaIng, 2021-09-22:
Did you add petitboot manually? As our image is shipped with U-Boot only.

As you say screen goes blank, can you try edit the `boot.ini` (on the FAT partition 1) and replace
```
setenv condev "console=ttyS0,115200n8"
```
with
```
setenv condev "console=ttyS0,115200n8 console=tty1"
```
I was always wondering whether the information is correct that the local console/screen is appended internally for boot/kernel message output.

Can you check whether the following files exist (on the ext4 partition 2) and in case paste their content here:
```
/var/tmp/dietpi/logs/fs_partition_resize.log
/var/tmp/dietpi/logs/dietpi-ramlog.log
/var/tmp/dietpi/logs/dietpi-firstboot.log
```

wdirksen, 2021-09-25:
Thanks for getting back and sorry for the delay.

So I'm using an untouched DietPi C4 image as downloaded from your site. It really is booting with Petitboot as it enters the clearly marked Petitboot menu where it sits for +/- 10 seconds before trying to startup. BTW, the Armbian HC4 image, which is based on bullseye, behaves identically to the DietPi image also boots (then fails) with Petitboot.

When I added `setenv condev "console=ttyS0,115200n8 console=tty1"` to boot.ini it did render an on-screen starting sequence instead of a pure black screen but doesn't ever get to see the DietPi rootfs stopping with the following error:

```
Gave up waiting for root file device.   Common problems:
 - Boot args (cat /proc/cmdline)
    - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/cmdline; ls /dev)
ALERT!   /dev/mmcblk0p2 does not exist.   Dropping to a shell!
(initramfs) [    51.51333000]  cectx  ff80023c.aocec: bad iniator with self 0x0
(initramfs) [    51.51333000]  cectx  ff80023c.aocec: bad iniator with self 0x0
(initramfs) [    51.51333000]  cectx  ff80023c.aocec: bad iniator with self 0x0
```
In both scenarios (with and without console=tty1 added) the /var/tmp/dietpi/log directory exists but is empty

MichaIng, 2021-09-25:
Ah, opposed to the C4, the HC4 has an onboard SPI module with bootloader (petitboot) on it already. There should be a switch to force booting from SD card instead, can you try that?

wdirksen, 2021-09-25:
There is indeed a momentary push type boot switch on the bottom. It is not a hardware toggle switch like on the Odroid N2 for SD <-> Emmc. Pushing the switch when the SBC has power applied does not allow it to remember a new setting. It must be pressed apparently every time at boot. This is quite silly as the boot switch is on the bottom and that is quite impractical if not impossible to do with 2x hard disks in the slots. See diagram below from Hard Kernel. So the Arch Arm image does cold boot correctly, even if it does not hot reboot correctly.

[image]

wdirksen, 2021-09-25:
> Good to know about kernel output on tty1, I'll add that as default to our Odroid boot configurations.

Just to be clear, In this problematic scenario there is 'something' on screen instead of blackness. It is not the typical systemd startup sequence however FYI

wdirksen, 2021-09-27:
FWIW, this is the boot.ini from the H4/HC4 image from ArchDroid's image generator which boots OK but doesn't reboot
https://github.com/archdroid-org/alarm-image-generator

```
ODROIDC4-UBOOT-CONFIG
# DO Not Edit this line: KERNEL: mainline

setenv board "odroidc4"

# Show logo as soon as possible
showlogo

# System Label
setenv bootlabel "ArchLinux Minimal"

# Default Console Device Setting
setenv condev "console=ttyAML0,115200n8 console=tty1"

# Video Output
## preferred_resolution
## set to your preferred resolution using the format:
##    {width}x{heigth}@{hertz}
##    example: 1920x1080@60
## or
##    Set to an empty string to let the kernel automatically
##    choose a resolution for you.
setenv preferred_resolution "1920x1080@60"

#setenv hpd "false"
# displayhpd=false
#setenv display_autodetect "false"
#setenv disablehpd "false"

## petitboot_edid
## if you experience issues of display not showing any
## image when using petitboot you can use this setting
## to overwrite the default display edid information.
##
## A valid value for this option can be:
##    800x600, 1024x768, 1280x720, 1280x1024, 1600x1200,
##    1680x1050, 1920x1080, 2560x1440, 2880x1800, 3840x2160
## or
##    Set to an empty string to use the display provided
##    edid information.
setenv petitboot_edid "1920x1080"

## Uncomment to force petitboot been always set as
## active to always force the edid overwrite.
#setenv petitboot_active "1"

if test "${petitboot_edid}" != "" && test "${petitboot_active}" = "1"; then
    setenv voutput "drm_kms_helper.edid_firmware=HDMI-A-1:edid/${petitboot_edid}.bin"
elif test "${preferred_resolution}" != ""; then
    setenv voutput "video=HDMI-A-1:${preferred_resolution}"
fi

# Boot Args
setenv bootargs "root=UUID=c507df78-c7cc-49a6-8329-8f7227bc0b3c rootwait rw"
setenv bootargs "${bootargs} mitigations=off ${condev}"
setenv bootargs "${bootargs} logo=osd0,loaded no_console_suspend"
setenv bootargs "${bootargs} fsck.repair=yes net.ifnames=0 cma=800M"
setenv bootargs "${bootargs} clk_ignore_unused ${voutput}"

# Set load addresses
setenv dtb_loadaddr "0x20000000"
setenv loadaddr "0x1080000"
setenv initrd_loadaddr "0x4080000"

# Load kernel, dtb and initrd
load mmc ${devno}:1 ${loadaddr} /Image
if test "${variant}" = "hc4"; then
    load mmc ${devno}:1 ${dtb_loadaddr} /dtbs/amlogic/meson64_odroidhc4.dtb
else
    load mmc ${devno}:1 ${dtb_loadaddr} /dtbs/amlogic/meson64_odroidc4.dtb
fi
load mmc ${devno}:1 ${initrd_loadaddr} /initramfs-linux.uimg

# boot
booti ${loadaddr} ${initrd_loadaddr} ${dtb_loadaddr}
```

MichaIng, 2021-11-11:
Is there any question left here? I.e. does it work when erasing the SPI flash?

wdirksen, 2021-11-12:
Hi Guys,

I've moved on to the Odroid provided version based on Ubuntu 20.04 which works with booting and restarting and is running as my file server. As far pressing the button is concerned, if this is a one time necessity to initialize the system the very first time and afterwards boots without pressing the button that would work I suppose. As I said above, you can't press the button when two hard drives are sitting in their slots.

I didn't mean to create a headache for you guys and because it doesn't startup on either of the Armbian HC4 images I thought it might not be your problem to fix when there was not more going on in this thread. I wanted to run DietPI and I thought it would be good for you as well as me.

FWIW, as I stated above, this [ArchArm](https://github.com/archdroid-org/alarm-image-generator) based image builder does install and boot fine. It doesn't restart however

MichaIng, 2021-11-12:
Okay. There is nothing we can do about this as simply the wrong bootloader is loaded. An installer image may work, as those are naturally build in a very compatible way without the need for updates/configurations etc, and it can then erase or overwrite the bootloader as part of the install process. I guess the SPI bootloader was installed by the Hardkernel provided image in the first place, so this of course works.

I'll mark the issue as closed for now. If you or someone else finds time to test pushing the button once and erasing the SPI with the steps mentioned above, we could add this to our documentation. The same will work on Odroid N2.

closed 2021-11-12 by MichaIng

wdirksen, 2021-11-13:
- I didn't see your boot advice was from September. I missed that somehow. Sorry
- I decided to try it on another SD card.  It worked!
- When I went to remove the mtd partitions, they did not exist in /dev. Not sure if the odroid Ubuntu installation removed them already or not

So yes, booted into DietPi by pressing button only once (without the drives) and reboots fine now without having to have the button pressed. Still does go through petitboot at boot time after the first time but finds DietPi

MichaIng, 2021-11-13:
Interesting, looks like petitboot got more compatible then, probably the Ubuntu installation did an update of it. As long as petitboot is loaded, obviously it is still stored on the SPI flash.

So there is no `ls -l /dev/mtd*` present on DietPi? Is is on Ubuntu or ArchArm? I found a related kernel driver now, on x86_64 at least, so probably this makes those devices appear:
```sh
modprobe mtdblock
```

wdirksen, 2021-11-15:
The HC4 has been on Ubuntu for a a few months. I followed your instructions...
```
sudo modprobe mtdblock
sudo ls -l /dev/mtd*
ls: cannot access '/dev/mtd*': No such file or directory
```

- So it looks like the SPI flash is empty.
- The difference between DietPI and Hardkernel's Ubuntu is different at initial boot. DietPi requires the button to be pressed the first time as you have correctly instructed. The Ubuntu image runs the 1st time without pressing the button but does not actually load image then. It runs an init script and moves files around then afterward reboots fine into the image on the SD card without ever having to press the button
- Once everything is up and going the boot behavior on DietPi and Hardkernel's Ubuntu image is similar where it boots from the SD card in petitboot after a timeout of +/- 10 seconds.

MichaIng, 2021-11-15:
> So it looks like the SPI flash is empty.

No this doesn't mean it is empty, but the devices are not available. Looks like there is some other driver required to make them visible. As long as petitboot is loaded, the SPI flash obviously is there are not empty as petitboot cannot come from anywhere else than from the SPI flash 😉. So petitboot is waiting for some other root partition and times out doing so, on both distros? At least then the correct bootloader is finally loaded. Still, would be good to find out how to erase the SPI to get rid of the unnecessary petitboot attempt and delay. I'll try to find further information about that.