MichaIng/DietPi #4766: Snakeoil Certificate

Gill-Bates, 2021-09-22:
I would love it, when Dietpi brings a Snakeoil Certificate by default:

```
apt install ssl-cert -y

# 10 years Expiration Days Sucks. Set it down to 90 Days. See here -> https://letsencrypt.org/2015/11/09/why-90-days.html
sed -i 's/opt_expiration_days="3650"/opt_expiration_days="90"/' /usr/sbin/make-ssl-cert

# Do the Magic
make-ssl-cert generate-default-snakeoil
sudo usermod --append --groups ssl-cert dietpi
ls -l /etc/ssl/certs/ssl-cert-snakeoil.pem /etc/ssl/private/ssl-cert-snakeoil.key

# Location
# SSL-Cert	/etc/ssl/certs/ssl-cert-snakeoil.pem
# SSL-Key	/etc/ssl/private/ssl-cert-snakeoil.key

# Adding Cert to the RootCA
cp /etc/ssl/certs/ssl-cert-snakeoil.pem /usr/local/share/ca-certificates/ssl-cert-snakeoil.crt
sudo update-ca-certificates

# Create daily Cron Script to check the Validation:
nano /etc/cron.daily/snakeoil
chmod +x /etc/cron.daily/snakeoil

if openssl x509 -checkend 86400 -noout -in /etc/ssl/certs/ssl-cert-snakeoil.pem
then
  echo "Certificate is valid"
else
  echo "Certificate has expired or will do so within 24 hours!"
  echo "Renew Certificate now ..."
  sudo make-ssl-cert generate-default-snakeoil --force-overwrite
fi
```

Joulinar, 2021-09-22:
What is the benefits of self-signed certificates? Nearly every web browser or app will complain on it. And there is no real need if you don't plan to make your system availability from external. Because what are you trying to secure inside your local network? Or are there family mates you don't trust?

Gill-Bates, 2021-09-22:
That's right. That's the stupid thing about self-signed certificates. The skin reason why I did it is because I want to use DNS-over-TLS for Adguard. And this is only possible if you have a valid certificate.

Joulinar, 2021-09-22:
Not fully correct as there are others option. You could have used our `unbound` implementation. On our online docs we describe how to use DoT on it. https://dietpi.com/docs/software/dns_servers/#unbound

Gill-Bates, 2021-09-22:
SFTP also needs a valid Certificate :-)

MichaIng, 2021-09-22:
SFTP does not require an HTTPS certificate. It uses SSH and hence requires only either user/password or a key pair for user authentication while the SSH servers have their own host keys for server authentication and the encryption is done with secret session based generated keys.

I'm wondering whether the snakeoil certificate shipped/generated by this package has any advantage or disadvantage over a manually generated certificate via OpenSSL. It shouldn't contain SAN e.g. so that it shouldn't help much to import it into an OS trusted CA storage to mute or lower browser/client warnings. So I guess a self-signed certificate generator would be more helpful, where you can include hostnames and IPs for SAN and this CA flag required to satisfy e.g. macOS.

So to my knowledge the snakeoil cert is really only for testing and if you require LAN HTTPS while a pubic cert is not applicable for some reason, then a manually generated cert e.g. via `openssl` with proper SAN should be superior.

And yes for DoT or DoH it isn't requires either since your resolver is the client. Of course it would be required of you have a local DoT/DoH server where other local DoT/DoH capable clients connect to. But that is quite uncommon and would mean multiple Unbound (does dnsmasq support DoT or DoH?) instances and at least separately encrypted connections along the route to the upstream provider.