MichaIng/DietPi #4768: Missing spotify related software on fresh install

ezekini, 2021-09-22:
### Creating a bug report/issue

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
- Distro version: bullseye 0
- Kernel version: Linux DietPi 5.10.60-v8+ # 1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021 aarch64 GNU/Linux
- SBC model: RPi 3 Model B+ (aarch64)

#### Additional Information (if applicable)
- Software title : Raspotify / Spotifyd
- Was the software title installed freshly or updated/migrated? Fresh
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Fresh install on rpi3B+
2. Search for Spotifyd or Raspotify in dietpi-software. No software found

Joulinar, 2021-09-22:
Hi,

these software title are disabled on 64bit aarch64

```
ID 141 | =0 | Spotify Connect Web: web interface for spotify premium | +ALSA DISABLED for aarch64 | https://dietpi.com/docs/software/media/#spotify-connect-web
ID 167 | =0 | Raspotify: spotify connect client | +ALSA DISABLED for aarch64 | https://dietpi.com/docs/software/media/#raspotify
ID 199 | =0 | Spotifyd: Open source Spotify client running as UNIX daemon | +ALSA DISABLED for aarch64 | https://dietpi.com/docs/software/media/#spotifyd
```

ezekini, 2021-09-22:
> Hi,
>
> these software title are disabled on 64bit aarch64
>
> ```
> ID 141 | =0 | Spotify Connect Web: web interface for spotify premium | +ALSA DISABLED for aarch64 | [...]

Thank you for the reply. Does that mean that it doesn't work or just not tested? I was tempted to try to install from source (curl -sL https://dtcooper.github.io/raspotify/install.sh | sh), but wanted to ask here first.

Joulinar, 2021-09-22:
it's a limitation of Raspotify as it did not support arm64

`repository 'https://dtcooper.github.io/raspotify raspotify InRelease' doesn't support architecture 'arm64'`

MichaIng, 2021-09-22:
The repository which is installed by the installer and where Raspotify is installed from is 32-bit (ARMv6) only, so that doesn't work on the 64-bit image.

Spotifyd currently does not ship aarch64 binaries: https://github.com/Spotifyd/spotifyd/releases
But it seems it is about to be implemented:
- https://github.com/Spotifyd/spotifyd/issues/280
- https://github.com/Spotifyd/spotifyd/pull/992

Spotify Connect Web didn't see some development the last years: https://github.com/Fornoth/spotify-connect-web
The pre-compiled binaries provided with the releases are armv7l only, but I see that all required shared binaries are packaged together with the archive, so it may run on aarch64 and mostly independent of the host system anyway. This is something you might want to test:
- Remove this code block form `/boot/dietpi/dietpi-software`: https://github.com/MichaIng/DietPi/blob/3e6763e/dietpi/dietpi-software#L547-L552
- Then it will appear in the software list.

ezekini, 2021-09-22:
OK, thank you. Googling a little bit ended up finding [spocon](https://github.com/spocon/spocon). Installed and configured in one minute, been using it for a few hours, working as expected; in case anyone finds the same issue.

ezekini, 2021-09-22:
Thank you for the tip-command. htop shows around 8% while streaming, no idea what's acceptable.

[image]

MichaIng, 2021-09-22:
Thanks, as expected, very high memory usage of > 200 MiB, the price of using Java applications in most cases. For comparison: Spotifyd uses 11 MiB and a constant 0.0% CPU usage on idle. Or is it currently streaming music in your case (7.3% CPU usage)?

However, sufficient memory is available, so not a big issue in your case 🙂.

closed 2021-10-06 by MichaIng