MichaIng/DietPi #4771: Boot - Drop NTP call without Inet connection

Axellarator, 2021-09-22:
#### During an installation, neither Wlan nor Eth was coming up, simply stupidity on my side. But in BOOT line 64 the nntp program is always called and the loop is 60 times. And calling an Inet service without Inet is normally useless. It is not mission critical, but if you could modify it ...

Joulinar, 2021-09-22:
You could configure system time sync in `dietpi.txt`. it could be disabled complete if needed.

MichaIng, 2021-09-22:
If I understand it right, it was only at first boot due to known mistake, so isn't any issue anymore, and of course disabling time sync causes larger issues than a delayed boot.

The plan is to move time sync into a background service so that boot continues, basically like if the `systemd-timesyncd` service was enabled, with the only difference that it stops running once the sync has completed.

Axellarator, 2021-09-22:
Okay, an OS installation without the Inet is not very useful. But ```it``` happens.

closed 2021-09-22 by Axellarator

Joulinar, 2021-09-22:
usually without internet, you should not be able to complete initial setup.

MichaIng, 2021-09-22:
But you can configure it when accessing `dietpi-config` from the related error prompt or when exiting it manually from console. Rare case, but generally I agree that it doesn't make sense to try a network time sync when it is known the network isn't up and it doesn't make sense to forcefully delay the whole boot while waiting for network, now that the systemd network and network-online targets are given more meaning: https://github.com/MichaIng/DietPi/blob/master/rootfs/etc/systemd/system/ifup%40.service.d/dietpi.conf

DietPi-PostBoot can wait for this target instead and DietPi-Boot is then basically obsolete and can be removed.