MichaIng/DietPi #4773: DietPie update from v7.0.2 to v7.6.2 fails

mixpc, 2021-09-23:
#### Details:
- Date           | Thu Sep 23 10:40:08 CEST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. run command root@DietPi:~# dietpi-update
#### Expected behaviour:
DietPie should update from v7.0.2 to v7.6.2
#### Actual behaviour:
Update does not follow and dietpie-update quits with error message
#### Extra details:
#### Additional logs:
```
Hit:1 https://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Reading package lists...
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
E: Repository 'http://raspbian.raspberrypi.org/raspbian bullseye InRelease' changed its 'Suite' value from 'testing' to 'stable'
```

Please, let me know if you need additonal logging or testing.

Thank you and regards

Joulinar, 2021-09-23:
Hi,

pls can your run `apt update && apt upgrade`

MichaIng, 2021-09-23:
Hmm, shall I re-add the pre-patch to accept the suite change automatically when detecting an `apt` package version below `1.8.2.3`?

Joulinar, 2021-09-23:
might be a good idea.

MichaIng, 2021-09-29:
Done: https://github.com/MichaIng/DietPi/commit/4bfc669b9ebacb28ef904145bf90c765e08c920c
I also found a way to automate this on Stretch. So we now check for the `apt` package version and in case accept the suite change non-interactively on all distro versions.

closed 2021-09-29 by MichaIng