MichaIng/DietPi #4774: Error while installing Docker on Bullseye

FrancTheCat, 2021-09-23:
#### Details:
- Date           | Thu Sep 23 20:56:13 BST 2021
- Bug report     | 5f165240-455f-455f-9ee4-6c1f316232b7
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. I tried to install Docker uing the DietPi launcher on a fresh DietPi installation on my RPi 4.
2. No other software is installed
#### Expected behaviour:

- Docker should install
#### Actual behaviour:

- I get an error message
#### Extra details:

- Honesyly, i have no which extra details I could give you, I will answer any specific questions you might have!
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Ign:6 https://download.docker.com/linux/debian 11 InRelease
Err:7 https://download.docker.com/linux/debian 11 Release
  404  Not Found [IP: 65.9.73.8 443]
Reading package lists...
E: The repository 'https://download.docker.com/linux/debian 11 Release' does not have a Release file.
```

MichaIng, 2021-09-23:
Many thanks for your report.

Can you add some more log lines before those, is still present? I wonder whether this is at the start of after the Docker repository has been added. Because this is how we add the line:
```sh
echo 'deb https://download.docker.com/linux/debian/ ${G_DISTRO_NAME/bookworm/bullseye} stable' > /etc/apt/sources.list.d/docker.list
```
And from the generated issue template `Distro | bullseye (ID=6,RASPBIAN=0)` I see that `$G_DISTRO_NAME` is indeeed "bullseye" as expected, so it should be impossible that an `11` is present instead in your case, there is simply no single part in our code where an `11` could come from.

But when you use some other installer of follow some 3rd party guides, some of them may still expect that `/etc/debian_version` contains `bullseye/sid`, while after the release now it contains `11.0`. So that is where an `11` could come from.

To solve the issue, run the above command and redo the install.

Joulinar, 2021-09-24:
I did a test installation on RPi4B ARMv8 Bullseye and apt source list was created correctly.

```
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Docker: Build, ship, and run distributed applications

[  OK  ] DietPi-Software | eval curl -sSfL 'https://download.docker.com/linux/debian/gpg' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-docker.gpg --yes
[  OK  ] DietPi-Software | eval echo 'deb https://download.docker.com/linux/debian/ bullseye stable' > /etc/apt/sources.list.d/docker.list
[ INFO ] DietPi-Software | APT update, please wait...
Get:1 https://download.docker.com/linux/debian bullseye InRelease [43.3 kB]
Hit:2 https://deb.debian.org/debian bullseye InRelease
Hit:3 https://deb.debian.org/debian bullseye-updates InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Get:7 https://download.docker.com/linux/debian bullseye/stable arm64 Packages [4061 B]
Fetched 47.4 kB in 2s (28.1 kB/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update
[ INFO ] DietPi-Software | APT install for: docker-ce, please wait...

```

FrancTheCat, 2021-09-24:
I tried it again today on another fresh install, I got the notice, that the Kernel has just been updated. After rebooting and trying again, it worked! Thank you very much for your time and effortt!

Joulinar, 2021-09-24:
ok good it is working now. 

Yeah the kernel got upgraded due to the new installation and is an expected behaviour. But has no relation to the original issue where source list was incorrect. 

I will close the issue for now. Feel free to reopen if needed.

closed 2021-09-24 by Joulinar