MichaIng/DietPi #4775: FreshRSS - infinite child 'p' folders after Buster to Bullseye upgrade

kinoushe, 2021-09-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | v7.6.2
- Distro version | Bullseye
- Kernel version | Linux 5.10.60-v7+
- SBC model | RPi 3 Model B+ (armv7l)
- Power supply used | Raspberry Pi official
- SDcard used | EVOPlus

#### Additional Information (if applicable)
- Software title | FreshRSS
- Was the software title installed freshly or updated/migrated? Freshly with Buster

#### Steps to reproduce
1. Install FreshRSS on Buster
2. Upgrade to Bullseye

#### Expected behaviour
- /var/www/freshrss/p/ (single level)

#### Actual behaviour
- 3. /var/www/freshrss/p/p/p/p/p/p/p/... and so on

#### Extra details
- I  went to create a tar backup of my dietpi backup and noticed this:
- /mnt/dietpi-backup/data/var/www/freshrss/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/p/themes/Mapco/icons/non-starred-white.svg and so on...
- I checked the backup prior to the Bullseye upgrade and there is only a single level: /var/www/freshrss/p/

MichaIng, 2021-09-24:
Many thanks for your report.

Strange thing, like if there was a symlink look 😄. I'll try to replicate to see what is going on.

kinoushe, 2021-09-24:
Indeed! 😄 Thanks for taking a look @MichaIng!

The only problem I had after the upgrade was losing ethernet, but was directed to your `net.ifnames=0` predictable network interface names fix on the forum.

During the upgrade I also ran the reinstall commands suggested in the post before rebooting to Bullseye:
`dietpi-software reinstall 38 40 47 48 56 83 84 85 88 89 93 114 143`

Joulinar, 2021-09-24:
could you share

`ls -la /mnt/dietpi-backup/data/var/www/freshrss`

kinoushe, 2021-09-24:
`lrwxrwxrwx 1 root root 15 Oct 21  2019 /mnt/dietpi-backup/data/var/www/freshrss -> /opt/FreshRSS/p`

MichaIng, 2021-09-24:
I think I know what might be the issue 😄. When creating a symlink, one must be careful to not allow it being placed into the directory the existing symlink is pointing to. Else one gets another iteration every time it is created. I'll have a look into the code and fix the faulty line. I'll also add a patch to next DietPi update to clean this up for existing installs.

kinoushe, 2021-09-24:
Phew! For a moment I thought I had wasted your time, because I was using the tar -cvhf command (removing the 'h' allows the backup to complete), but yes, the reiterating 'p' folder is still there in source.

Now I'm also not sure which is the correct tar command to make a copy of the dietpi-backup folder to store externally?

MichaIng, 2021-10-16:
I found a bug on reinstalls where the new directory is moved into the old one on a reinstall. Since it has an internal updater, if the install dir exists already, it will now be skipped with a hint that if a repair is wanted, it needs to be removed manually after backing up possibly changed config and data files (`/opt/FreshRSS/data`).

An now I also found the reason for the nested `p`:
- `ln -sf /opt/FreshRSS/p /var/www/freshrss` puts the the symlink into `/var/www/freshrss/` if that one exists already and links to a directory.

For both issues the solution is simply:
```sh
rm -R /opt/FreshRSS/FreshRSS-master /opt/FreshRSS/p/p
```
I'll apply a patch as well to clean this up.

MichaIng, 2021-10-16:
Fixed with: https://github.com/MichaIng/DietPi/commit/dc3b23c6943184c43501fa57b337c99550f78460

closed 2021-10-16 by MichaIng