MichaIng/DietPi #4777: Shairport-sync issue after tvOS 15 installed

audiobruce, 2021-09-26:
Hi Mike,

Have been using DietPi + RoonBridge + Shairport-sync for years.

Upgraded AppleTV to tvOS 15 which instantly stuffed up AirPlay to RPi: 

- when hit play the sound starts out in sync with video as per normal
- sound very rapidly gets out of sync in a matter of seconds
- ends up with massive audio / video disparity
- video stutters cyclically about every 5 seconds...
 
AppleTV plays normally direct to TV (not using AirPlay).

thanks,
Bruce

#### Required Information
- DietPi version | 7.6.2
- Distro version | buster 1
- Kernel version | Linux DietPi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux
- SBC model | RPi 3 Model B (armv7l)

root@DietPi:~# shairport-sync -V
3.3.8-OpenSSL-Avahi-ALSA-soxr-metadata-mqtt-dbus-mpris-sysconfdir:/usr/local/etc

MichaIng, 2021-09-26:
Many thanks for your report.

Can you post the Shairport Sync logs, please:
```sh
journalctl -u shairport-sync
```
Probably not related but there is another report about issues with tvOS 15 on the Shairport Sync GitHub repo: https://github.com/mikebrady/shairport-sync/issues/1292

audiobruce, 2021-09-26:
Shairport Sync playing and nothing here...

-- Logs begin at Thu 2019-02-14 21:11:58 AEDT, end at Mon 2021-09-27 07:28:12 AEST. --
Sep 27 07:25:14 DietPi systemd[1]: Started Shairport Sync - AirPlay Audio Receiver.

MichaIng, 2021-09-28:
Now there is a matching report at the Shairport Sync repo: https://github.com/mikebrady/shairport-sync/issues/1297
Since I don't have an AppleTV for testing, let's keep discussion over there.

mikebrady, 2021-09-28:
Sadly, I don't have a recent ATV either. What would be interesting is to see what the behaviour is with, say, an old Apple Airport Express -- one that does AirPlay 1 only -- to see if the same thing happens to it. If that was so, then it would indicate that `tvOS 15` is moving away from AirPlay 1 support...

MichaIng, 2021-09-28:
As said, let's contribute any ideas to the issue over at Shairport Sync to avoid split information on the matter 😉.

MichaIng, 2022-10-15:
New packages are available which should have the issue fixed:
```sh
dietpi-software reinstall 37
```

closed 2022-10-15 by MichaIng