MichaIng/DietPi #4778: Medusa not working after install

igorkulman, 2021-09-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7.6.2`
- Distro version | `buster 1`
- Kernel version | `Linux DietPi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux`
- SBC model | `RPi 2 Model B (armv7l)`
- Power supply used | 3A
- SDcard used | USB Flash drive

#### Additional Information (if applicable)
Medusa, freshly installed

#### Steps to reproduce

1. Install Medusa

#### Expected behaviour

- Medusa runs on port 8081

#### Actual behaviour

- Medusa not running on port 8081

#### Extra details

- running `systemctl status medusa` showed that everything was OK
- running `lsof -i -P -n | grep LISTEN` did not show any process listening on `8081`. 
- looking at `cat /mnt/dietpi_userdata/medusa/Logs/application.log` I see only some warnings.

Looking at `journalctl -u medusa` I see

```bash
Sep 26 19:31:33 DietPi systemd[1]: Started Medusa Daemon.
Sep 26 19:31:51 DietPi python3[9515]: Traceback (most recent call last):
Sep 26 19:31:51 DietPi python3[9515]:   File "/mnt/dietpi_userdata/medusa/start.py", line 9, in <module>
Sep 26 19:31:51 DietPi python3[9515]:     main()
Sep 26 19:31:51 DietPi python3[9515]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 2352, in main
Sep 26 19:31:51 DietPi python3[9515]:     application.start(sys.argv[1:])
Sep 26 19:31:51 DietPi python3[9515]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 350, in start
Sep 26 19:31:51 DietPi python3[9515]:     self.initialize(console_logging=self.console_logging)
Sep 26 19:31:51 DietPi python3[9515]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 1184, in initialize
Sep 26 19:31:51 DietPi python3[9515]:     db.upgradeDatabase(main_db_con, main_db.InitialSchema)
Sep 26 19:31:51 DietPi python3[9515]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 453, in upgradeDatabase
Sep 26 19:31:51 DietPi python3[9515]:     _processUpgrade(connection, schema)
Sep 26 19:31:51 DietPi python3[9515]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 476, in _processUpgrade
Sep 26 19:31:51 DietPi python3[9515]:     _processUpgrade(connection, upgradeSubClass)
Sep 26 19:31:51 DietPi python3[9515]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 463, in _processUpgrade
Sep 26 19:31:51 DietPi python3[9515]:     if not instance.test():
Sep 26 19:31:51 DietPi python3[9515]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/databases/main_db.py", line 284, in test
Sep 26 19:31:51 DietPi python3[9515]:     return self.checkMajorDBVersion() >= 40
Sep 26 19:31:51 DietPi python3[9515]: TypeError: '>=' not supported between instances of 'NoneType' and 'int'
Sep 26 19:31:52 DietPi systemd[1]: medusa.service: Main process exited, code=exited, status=1/FAILURE

Sep 26 19:31:52 DietPi systemd[1]: medusa.service: Failed with result 'exit-code'.
Sep 26 19:31:52 DietPi systemd[1]: medusa.service: Service RestartSec=100ms expired, scheduling restart.
Sep 26 19:31:52 DietPi systemd[1]: medusa.service: Scheduled restart job, restart counter is at 1.
Sep 26 19:31:52 DietPi systemd[1]: Stopped Medusa Daemon.
Sep 26 19:31:52 DietPi systemd[1]: Started Medusa Daemon.
Sep 26 19:32:09 DietPi python3[9525]: Traceback (most recent call last):
Sep 26 19:32:09 DietPi python3[9525]:   File "/mnt/dietpi_userdata/medusa/start.py", line 9, in <module>
Sep 26 19:32:09 DietPi python3[9525]:     main()
Sep 26 19:32:09 DietPi python3[9525]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 2352, in main
Sep 26 19:32:09 DietPi python3[9525]:     application.start(sys.argv[1:])
Sep 26 19:32:09 DietPi python3[9525]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 350, in start
Sep 26 19:32:09 DietPi python3[9525]:     self.initialize(console_logging=self.console_logging)
Sep 26 19:32:09 DietPi python3[9525]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 1184, in initialize
Sep 26 19:32:09 DietPi python3[9525]:     db.upgradeDatabase(main_db_con, main_db.InitialSchema)
Sep 26 19:32:09 DietPi python3[9525]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 453, in upgradeDatabase
Sep 26 19:32:09 DietPi python3[9525]:     _processUpgrade(connection, schema)
Sep 26 19:32:09 DietPi python3[9525]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 476, in _processUpgrade
Sep 26 19:32:09 DietPi python3[9525]:     _processUpgrade(connection, upgradeSubClass)
Sep 26 19:32:09 DietPi python3[9525]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 463, in _processUpgrade
Sep 26 19:32:09 DietPi python3[9525]:     if not instance.test():
Sep 26 19:32:09 DietPi python3[9525]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/databases/main_db.py", line 284, in test
Sep 26 19:32:09 DietPi python3[9525]:     return self.checkMajorDBVersion() >= 40
Sep 26 19:32:09 DietPi python3[9525]: TypeError: '>=' not supported between instances of 'NoneType' and 'int'
Sep 26 19:32:10 DietPi systemd[1]: medusa.service: Main process exited, code=exited, status=1/FAILURE

Sep 26 19:32:10 DietPi systemd[1]: medusa.service: Failed with result 'exit-code'.
Sep 26 19:32:10 DietPi systemd[1]: medusa.service: Service RestartSec=100ms expired, scheduling restart.
Sep 26 19:32:10 DietPi systemd[1]: medusa.service: Scheduled restart job, restart counter is at 2.
Sep 26 19:32:10 DietPi systemd[1]: Stopped Medusa Daemon.
Sep 26 19:32:10 DietPi systemd[1]: Started Medusa Daemon.
Sep 26 19:32:27 DietPi python3[9539]: Traceback (most recent call last):
Sep 26 19:32:27 DietPi python3[9539]:   File "/mnt/dietpi_userdata/medusa/start.py", line 9, in <module>
Sep 26 19:32:27 DietPi python3[9539]:     main()
Sep 26 19:32:27 DietPi python3[9539]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 2352, in main
Sep 26 19:32:27 DietPi python3[9539]:     application.start(sys.argv[1:])
Sep 26 19:32:27 DietPi python3[9539]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 350, in start
Sep 26 19:32:27 DietPi python3[9539]:     self.initialize(console_logging=self.console_logging)
Sep 26 19:32:27 DietPi python3[9539]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/__main__.py", line 1184, in initialize
Sep 26 19:32:27 DietPi python3[9539]:     db.upgradeDatabase(main_db_con, main_db.InitialSchema)
Sep 26 19:32:27 DietPi python3[9539]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 453, in upgradeDatabase
Sep 26 19:32:27 DietPi python3[9539]:     _processUpgrade(connection, schema)
Sep 26 19:32:27 DietPi python3[9539]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 476, in _processUpgrade
Sep 26 19:32:27 DietPi python3[9539]:     _processUpgrade(connection, upgradeSubClass)
Sep 26 19:32:27 DietPi python3[9539]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/db.py", line 463, in _processUpgrade
Sep 26 19:32:27 DietPi python3[9539]:     if not instance.test():
Sep 26 19:32:27 DietPi python3[9539]:   File "/mnt/flash/dietpi_userdata/medusa/medusa/databases/main_db.py", line 284, in test
Sep 26 19:32:27 DietPi python3[9539]:     return self.checkMajorDBVersion() >= 40
Sep 26 19:32:27 DietPi python3[9539]: TypeError: '>=' not supported between instances of 'NoneType' and 'int'
Sep 26 19:32:28 DietPi systemd[1]: medusa.service: Main process exited, code=exited, status=1/FAILURE

Sep 26 19:32:28 DietPi systemd[1]: medusa.service: Failed with result 'exit-code'.
Sep 26 19:32:29 DietPi systemd[1]: medusa.service: Service RestartSec=100ms expired, scheduling restart.
Sep 26 19:32:29 DietPi systemd[1]: medusa.service: Scheduled restart job, restart counter is at 3.
Sep 26 19:32:29 DietPi systemd[1]: Stopped Medusa Daemon.
Sep 26 19:32:29 DietPi systemd[1]: medusa.service: Start request repeated too quickly.
Sep 26 19:32:29 DietPi systemd[1]: medusa.service: Failed with result 'exit-code'.
Sep 26 19:32:29 DietPi systemd[1]: Failed to start Medusa Daemon.
```

MichaIng, 2021-09-28:
Many thanks. I'll try to replicate.

igorkulman, 2021-09-29:
Works after deleting and installing again. Not sure what happened the first time. Sorry for the false alarm.

closed 2021-09-29 by igorkulman