MichaIng/DietPi #4779: Sonarr/Radarr not importing completed downloads

ZiggyPi, 2021-09-26:
This week i began having issues with Sonarr and Radarr not importing completed downloads from Deluge.

The sonarr log gives an error "unable to create folder".
So that seems to be a permission issue.

So i tried the obvious... chmod -R 777 /HTPC
HTPC is the link to my external HDD with HTPC/Movies and HTPC/Series.

Both are root:root, i also tried sonarr:root, sonarr:dietpi and dietpi:dietpi

But i still get the same error.

I have a suspicious this happend after an update.

Any help is appreciated. I now have to create folders and move files manually :cry:

MichaIng, 2021-09-28:
Many thanks for your report.

What is the filesystem type on the HDD?
```
findmnt -no FSTYPE -T /HTPC/Movies
```

ZiggyPi, 2021-09-28:
Hi Micha,

root@DietPi-HTPC:~# findmnt -no FSTYPE -T /HTPC/Movies
autofs
fusebl

MichaIng, 2021-09-28:
Ah, can you check again:
```sh
blkid -s TYPE -o value -c /dev/null "$(findmnt -Ufnro SOURCE -T /HTPC/Movies)"
```
It seems to be NTFS formatted, isn't it? NTFS does not support UNIX permissions natively and Mono had an issue in such case to write files. But that should be actually solved by Sonarr v3 and the .NET based Radarr v3.

Which DietPi version do you use and which SBC/device? Generally filling the bug report template is advised, we have it for reasons 😉.

ZiggyPi, 2021-09-28:
`DietPi v7.6.2 Device model : RPi 4 Model B (armv7l)`

And yes it is NTFS, but i have been using it for years and never had this problem before.

`root@DietPi-HTPC:~# blkid -s TYPE -o value -c /dev/null "$(findmnt -Ufnro SOURCE -T /HTPC/Movies)"
ntfs`

MichaIng, 2021-09-28:
The issue appeared with Mono v6, so in theory a package upgrade could have caused it.

Are Radarr and Sonarr at latest version (updated via their internal updater)?

MichaIng, 2021-09-28:
> The last time i used the internal updater i had to reinstall.

But are they on latest v3 now both? As said, that was an issue on older versions, so updating Sonarr and Radarr is very likely the solution.

ZiggyPi, 2021-09-28:
Both up to date
Sonarr : 3.0.6.1265
Radarr : 3.2.2.5080

MichaIng, 2021-09-28:
Okay that should be fine.

And the permissions/modes you applied to the media directories are reported back as expected?
```sh
ls -l /HTPC/Movies | mawk '{print $1,$3,$4}'
```

ZiggyPi, 2021-09-28:
`root@DietPi-HTPC:~# ls -l /HTPC/Series | mawk '{print $1,$3,$4}'
total
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
root@DietPi-HTPC:~# ls -l /HTPC/Movies | mawk '{print $1,$3,$4}'           total
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxr-xr-x root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
drwxrwxrwx root root
root@DietPi-HTPC:~#`

MichaIng, 2021-09-28:
Okay as of mode 777 Radarr and Sonarr should have write permissions. Could you try to change owner to Sonarr, check back whether this has been applied, and then import a series via Sonarr?
```sh
chown -R sonarr: /HTPC/Series
ls -l /HTPC/Series | mawk '{print $1,$3,$4}' | head -5
```

ZiggyPi, 2021-09-28:
Permissions were set correct (sonarr:dietpi)
But still not importing. See log from sonarr

2021-09-28 16:02:39.7|Warn|ImportApprovedEpisodes|Couldn't import episode /HTPC/Downloads/Complete/Breaking.Bad.S01E02.DSR.XviD-0TV.avi

[v3.0.6.1265] System.IO.DirectoryNotFoundException: Could not find a part of the path "/HTPC/Series/Breaking Bad/Season 1/Breaking.Bad.S01E02.DSR.XviD-0TV.avi".
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x00164] in <533173d24dae460899d2b10975534bb0>:0 
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.IO.FileOptions options) [0x00000] in <533173d24dae460899d2b10975534bb0>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int,System.IO.FileOptions)
  at System.IO.FileSystem.CopyFile (System.String sourceFullPath, System.String destFullPath, System.Boolean overwrite) [0x00035] in <533173d24dae460899d2b10975534bb0>:0 
  at System.IO.File.Copy (System.String sourceFileName, System.String destFileName, System.Boolean overwrite) [0x00062] in <533173d24dae460899d2b10975534bb0>:0 
  at NzbDrone.Common.Disk.DiskProviderBase.CopyFileInternal (System.String source, System.String destination, System.Boolean overwrite) [0x00000] in <11136e1590e2463e903086340d35290e>:0 
  at NzbDrone.Mono.Disk.DiskProvider.TransferFilePatched (System.String source, System.String destination, System.Boolean overwrite, System.Boolean move) [0x00057] in <7834a0bc7d0841f6b5d32356705ed00a>:0 
  at NzbDrone.Mono.Disk.DiskProvider.CopyFileInternal (System.String source, System.String destination, System.Boolean overwrite) [0x0009c] in <7834a0bc7d0841f6b5d32356705ed00a>:0 
  at NzbDrone.Common.Disk.DiskProviderBase.CopyFile (System.String source, System.String destination, System.Boolean overwrite) [0x000ba] in <11136e1590e2463e903086340d35290e>:0 
  at NzbDrone.Common.Disk.DiskTransferService.TryCopyFileVerified (System.String sourcePath, System.String targetPath, System.Int64 originalSize) [0x00047] in <11136e1590e2463e903086340d35290e>:0 
  at NzbDrone.Common.Disk.DiskTransferService.TransferFile (System.String sourcePath, System.String targetPath, NzbDrone.Common.Disk.TransferMode mode, System.Boolean overwrite) [0x003ad] in <11136e1590e2463e903086340d35290e>:0 
  at NzbDrone.Core.MediaFiles.EpisodeFileMovingService.TransferFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Tv.Series series, System.Collections.Generic.List`1[T] episodes, System.String destinationFilePath, NzbDrone.Common.Disk.TransferMode mode) [0x00129] in <b2ccbaa5d8e7430c8f2be515a642b6b9>:0 
  at NzbDrone.Core.MediaFiles.EpisodeFileMovingService.CopyEpisodeFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Parser.Model.LocalEpisode localEpisode) [0x00061] in <b2ccbaa5d8e7430c8f2be515a642b6b9>:0 
  at NzbDrone.Core.MediaFiles.UpgradeMediaFileService.UpgradeEpisodeFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Parser.Model.LocalEpisode localEpisode, System.Boolean copyOnly) [0x00196] in <b2ccbaa5d8e7430c8f2be515a642b6b9>:0 
  at NzbDrone.Core.MediaFiles.EpisodeImport.ImportApprovedEpisodes.Import (System.Collections.Generic.List`1[T] decisions, System.Boolean newDownload, NzbDrone.Core.Download.DownloadClientItem downloadClientItem, NzbDrone.Core.MediaFiles.EpisodeImport.ImportMode importMode) [0x0029b] in <b2ccbaa5d8e7430c8f2be515a642b6b9>:0

MichaIng, 2021-09-28:
Hmm, this indicates indeed an issue with setting the file modes. Very strange as this shouldn't happen anymore. Can you check whether Mono is up-to-date?
```
apt install mono-runtime mono-complete
dpkg -l | grep mono
```

And if you try the same for Radarr (which does not use Mono), it is the same issue? Can you share again the full logs of this?

ZiggyPi, 2021-09-28:
`oot@DietPi-HTPC:~# apt install mono-runtime mono-complete
Reading package lists... Done
Building dependency tree
Reading state information... Done
mono-complete is already the newest version (6.12.0.122-0xamarin1+raspbian10b1).
mono-runtime is already the newest version (6.12.0.122-0xamarin1+raspbian10b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

Radarr same issue (see log)

`2021-09-28 16:24:17.9|Warn|ImportApprovedMovie|Couldn't import movie /HTPC/Downloads/Complete/Old.2021.1080p.WEBRip.DD5.1.x264-NOGRP/Old.2021.1080p.WEBRip.DD5.1.x264-NOGRP.mkv

[v3.2.2.5080] System.IO.IOException: Read-only file system
   at System.IO.FileSystem.LinkOrCopyFile(String sourceFullPath, String destFullPath)
   at System.IO.FileSystem.MoveFile(String sourceFullPath, String destFullPath, Boolean overwrite)
   at System.IO.File.Move(String sourceFileName, String destFileName, Boolean overwrite)
   at NzbDrone.Mono.Disk.DiskProvider.TransferFilePatched(String source, String destination, Boolean overwrite, Boolean move) in D:\a\1\s\src\NzbDrone.Mono\Disk\DiskProvider.cs:line 333
   at NzbDrone.Mono.Disk.DiskProvider.MoveFileInternal(String source, String destination) in D:\a\1\s\src\NzbDrone.Mono\Disk\DiskProvider.cs:line 300
   at NzbDrone.Common.Disk.DiskProviderBase.MoveFile(String source, String destination, Boolean overwrite) in D:\a\1\s\src\NzbDrone.Common\Disk\DiskProviderBase.cs:line 254
   at NzbDrone.Common.Disk.DiskTransferService.TryMoveFileVerified(String sourcePath, String targetPath, Int64 originalSize) in D:\a\1\s\src\NzbDrone.Common\Disk\DiskTransferService.cs:line 495
   at NzbDrone.Common.Disk.DiskTransferService.TransferFile(String sourcePath, String targetPath, TransferMode mode, Boolean overwrite) in D:\a\1\s\src\NzbDrone.Common\Disk\DiskTransferService.cs:line 296
   at NzbDrone.Core.MediaFiles.MovieFileMovingService.TransferFile(MovieFile movieFile, Movie movie, String destinationFilePath, TransferMode mode) in D:\a\1\s\src\NzbDrone.Core\MediaFiles\MovieFileMovingService.cs:line 134
   at NzbDrone.Core.MediaFiles.MovieFileMovingService.MoveMovieFile(MovieFile movieFile, LocalMovie localMovie) in D:\a\1\s\src\NzbDrone.Core\MediaFiles\MovieFileMovingService.cs:line 79
   at NzbDrone.Core.MediaFiles.UpgradeMediaFileService.UpgradeMovieFile(MovieFile movieFile, LocalMovie localMovie, Boolean copyOnly) in D:\a\1\s\src\NzbDrone.Core\MediaFiles\UpgradeMediaFileService.cs:line 51
   at NzbDrone.Core.MediaFiles.MovieImport.ImportApprovedMovie.Import(List`1 decisions, Boolean newDownload, DownloadClientItem downloadClientItem, ImportMode importMode) in D:\a\1\s\src\NzbDrone.Core\MediaFiles\MovieImport\ImportApprovedMovie.cs:line 123`

MichaIng, 2021-09-28:
Logs/pastes from console and files are btw nicer to read when wrapped into code fences 🙂:
````
```
paste multiple
lines here
```
````

Indeed the issue seems to be underlying the same. Probably latest .NET core broke things again and found it's way into latest Mono already 😞...

But strange is that Mono has not seen any update since March, so I'd have expected more reports. I'll try to replicate it.

ZiggyPi, 2021-09-28:
Micha thanks inadvance.
If it helps Teamviewer is also an option.

ZiggyPi, 2021-10-01:
@MichaIng i now see an available update in sonarr itself. Is this a possible fix for the problem. And is it at all possible to update in sonarr itself without messing things up. Apt update didnt find any updates...

MichaIng, 2021-10-01:
Where do you see the Sonarr update? It might take a while until a release is pushed to the APT repository, so keep an eye on the banner notification (if you have daily APT checks enabled) or `apt update`.

And yes, of course you can and should upgrade the Sonarr APT package as fast as an update is available, and as long as Sonarr guys don't do a mistake, it won't mess things up.

ZiggyPi, 2021-10-01:
I'm a bit hesitant as the last time i update through sonarr/radarr i had to reinstall and add every movie/serie again.

[image]

MichaIng, 2021-10-01:
Hmm, was it probably the upgrade from Sonarr v2 to v3? Movies etc are stored in the database which should not be touched neither by the package upgrade nor by the internal updater. Interesting actually that the internal updater can be used when the package was installed via APT, as this may involve a downgrade when the package is reinstalled. Not sure how this is solved.

If unsure, backup the data directory first:
```sh
cp -a /mnt/dietpi_userdata/sonarr{_bak}
```

ZiggyPi, 2021-10-01:
Okay, updated it through sonarr. But still not importing completed episodes

ZiggyPi, 2021-10-04:
@MichaIng any progress on a solution?

MichaIng, 2021-10-04:
It works fine here, importing via Sonarr into an NTFS mount (formatted and mounted via `dietpi-drive_manager`) after:
```sh
chgrp -R dietpi /mnt/media
chmod g=u -R dietpi /mnt/media
```
You have the ntfs-3g driver installed, right?
```sh
apt install ntfs-3g
```
In case, the `fstab` entry:
```
UUID=2B514D5B573D8190 /mnt/media ntfs noatime,lazytime,rw,permissions,big_writes,nofail,noauto,x-systemd.automount
```

ZiggyPi, 2021-10-04:
Yep ntfs-3g newest version and my fstab looks the same the only diffetrence is the "/mnt/media" mine says "/HTPC" if i look in /dietpi/mnt/ i cant see my mount. It is under /dietpi/root/HTPC.

But as far as i know it has always been that way

MichaIng, 2021-10-04:
Ah dammit, I totally forgot about the systemd-wise hardening. Please do the following:
- Either you change the mount point to something below `/mnt` or below `/media`.
- Or:
    ```sh
    mkdir -p /etc/systemd/system/{sonarr,radarr}.service.d
    echo -e '[Service]\nReadWritePaths=-/HTPC' > /etc/systemd/system/sonarr.service.d/extend-write-paths.conf
    echo -e '[Service]\nReadWritePaths=-/HTPC' > /etc/systemd/system/radarr.service.d/extend-write-paths.conf
    systemctl daemon-reload
    systemctl restart sonarr radarr
    ```

ZiggyPi, 2021-10-05:
Okay so i changed my HTPC mount to /mnt/HTPC and changed the symlink accordingly.

Now sonarr imports the episodes but it doesnt delete them from deluge. They stay queued.

MichaIng, 2021-10-05:
The letter is done via Deluge API, so that shouldn't be affected by the restrictions. I cannot test it currently as on my test systems Deluge does not start downloads and I don't want to open ports to testing systems 😉. I see downloads were done to `/HTPC/Downloads`, so the same mount and now hence `/mnt/HTPC/Downloads`?

Isn't it also a setting in Sonarr/Radarr whether to keep (for seeding) or remove important downloads?

closed 2021-10-05 by ZiggyPi

MichaIng, 2021-10-05:
Great. Probably we should relax the Sonarr/Radarr/Lidarr hardenings a bit to not R/W protect everything outside of the `ReadWritePaths` list 🤔.

ZiggyPi, 2021-10-05:
Yeah thanks very much for your help.

And yes i think that relaxing the hardening would have saved me a lot of headache. But yet again, good oppertunity to learn some new stuff and setting up my mount better 😁