MichaIng/DietPi #4780: Bazarr problem after upgrading from buster to bullseye

Tooloco, 2021-09-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
G_LIVE_PATCH_STATUS[3]='applied'

- Distro version | `bullseye 1`
- Kernel version | `Linux DietPi3 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | [Labists 5.1V 3A](https://labists.com/products/raspberry-pi-4-power-supply)
- SDcard used | SanDisk ultra 32GB

#### Additional Information (if applicable)
- Software title | Bazarr
- Was the software title installed freshly or updated/migrated? Updated 
- Can this issue be replicated on a fresh installation of DietPi? Not sure

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Upgrade from buster to bullseye
2. Bazarr will fail to initialise due to missing pip package to install requirements
3. After installing python3-pip the following error message is prompted: 

> ERROR   |root                            |BAZARR requirements.txt installation result: b'ERROR: Could not find a version that satisfies the requirement gevent>=21\nERROR: No matching distribution found for gevent>=21\n'|'Traceback (most recent call last):\n  File "/opt/bazarr/bazarr/init.py", line 57, in <module>\n    import lxml, numpy, webrtcvad, gevent, geventwebsocket, setuptools\nModuleNotFoundError: No module named \'webrtcvad\'\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File "/opt/bazarr/bazarr/init.py", line 74, in <module>\n    subprocess.check_output(pip_command, stderr=subprocess.STDOUT)\n  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output\n    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,\n  File "/usr/lib/python3.9/subprocess.py", line 528, in run\n    raise CalledProcessError(retcode, process.args,\nsubprocess.CalledProcessError: Command \'[\'/usr/bin/python3\', \'-m\', \'pip\', \'install\', \'--user\', \'-qq\', \'--disable-pip-version-check\', \'-r\', \'/opt/bazarr/requirements.txt\']\' returned non-zero exit status 1.'|

#### Expected behaviour

- Bazarr should initialise correctly

#### Actual behaviour

- Bazarr is failing to initialise due to a missing requirement

#### Extra details

Tooloco, 2021-09-27:
Ok, So following this I launched bazarr again by `dietpi-services start bazarr' and it has started working again although I can see this error message:

> |root                            |Error trying to get releases from Github. Connection Error.|'Traceback (most recent call last):\n  File "/opt/bazarr/bazarr/../libs/urllib3/connection.py", line 159, in _new_conn\n    conn = connection.create_connection(\n  File "/opt/bazarr/bazarr/../libs/subliminal_patch/http.py", line 366, in patched_create_connection\n    return _orig_create_connection((host, port), *args, **kwargs)\n  File "/opt/bazarr/bazarr/../libs/urllib3/util/connection.py", line 80, in create_connection\n    raise err\n  File "/opt/bazarr/bazarr/../libs/urllib3/util/connection.py", line 70, in create_connection\n    sock.connect(sa)\n  File "/mnt/dietpi_userdata/bazarr/.local/lib/python3.9/site-packages/gevent/_socketcommon.py", line 607, in connect\n    raise _SocketError(err, strerror(err))\nTimeoutError: [Errno 110] Connection timed out\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File "/opt/bazarr/bazarr/../libs/urllib3/connectionpool.py", line 600, in urlopen\n    httplib_response = self._make_request(conn, method, url,\n  File "/opt/bazarr/bazarr/../libs/urllib3/connectionpool.py", line 344, in _make_request\n    self._validate_conn(conn)\n  File "/opt/bazarr/bazarr/../libs/urllib3/connectionpool.py", line 843, in _validate_conn\n    conn.connect()\n  File "/opt/bazarr/bazarr/../libs/urllib3/connection.py", line 316, in connect\n    conn = self._new_conn()\n  File "/opt/bazarr/bazarr/../libs/urllib3/connection.py", line 168, in _new_conn\n    raise NewConnectionError(\nurllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0xb0ce7118>: Failed to establish a new connection: [Errno 110] Connection timed out\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File "/opt/bazarr/bazarr/../libs/requests/adapters.py", line 439, in send\n    resp = conn.urlopen(\n  File "/opt/bazarr/bazarr/../libs/urllib3/connectionpool.py", line 640, in urlopen\n    retries = retries.increment(method, url, error=e, _pool=self,\n  File "/opt/bazarr/bazarr/../libs/urllib3/util/retry.py", line 399, in increment\n    raise MaxRetryError(_pool, url, error or ResponseError(cause))\nurllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host=\'api.github.com\', port=443): Max retries exceeded with url: /repos/morpheus65535/Bazarr/releases?per_page=100 (Caused by NewConnectionError(\'<urllib3.connection.VerifiedHTTPSConnection object at 0xb0ce7118>: Failed to establish a new connection: [Errno 110] Connection timed out\'))\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n  File "/opt/bazarr/bazarr/check_update.py", line 22, in check_releases\n    r = requests.get(url_releases, allow_redirects=True)\n  File "/opt/bazarr/bazarr/../libs/requests/api.py", line 76, in get\n    return request(\'get\', url, params=params, **kwargs)\n  File "/opt/bazarr/bazarr/../libs/requests/api.py", line 61, in request\n    return session.request(method=method, url=url, **kwargs)\n  File "/opt/bazarr/bazarr/../libs/requests/sessions.py", line 542, in request\n    resp = self.send(prep, **send_kwargs)\n  File "/opt/bazarr/bazarr/../libs/requests/sessions.py", line 655, in send\n    r = adapter.send(request, **kwargs)\n  File "/opt/bazarr/bazarr/../libs/requests/adapters.py", line 516, in send\n    raise ConnectionError(e, request=request)\nrequests.exceptions.ConnectionError: HTTPSConnectionPool(host=\'api.github.com\', port=443): Max retries exceeded with url: /repos/morpheus65535/Bazarr/releases?per_page=100 (Caused by NewConnectionError(\'<urllib3.connection.VerifiedHTTPSConnection object at 0xb0ce7118>: Failed to establish a new connection: [Errno 110] Connection timed out\'))'|

Which could be a network error not necessarily a bazarr one

Joulinar, 2021-09-27:
looks like your system is trying to connect to GitHub to check for new version and this is going to `timed out`

`That's the url I could see on your error message`
https://api.github.com/repos/morpheus65535/Bazarr/releases?per_page=100

Maybe a temp issue. Maybe retry or check if you could open url from a different system.

Tooloco, 2021-09-27:
Yes, it seems it was a temp issue as now there's no errors being thrown. Im more worried about the initial upgrade from buster as it seems that pip3 gets autoremoved (I presume, not sure) and that means that bazarr cant start unless you manually install pip3.

MichaIng, 2021-09-28:
The Python 3 pip should stay installed during the upgrade, also as it is no APT package but installed via https://bootstrap.pypa.io/get-pip.py.

So the issue is not present anymore and the issue can hence be closed?

Tooloco, 2021-09-28:
I did install pip through APT though (`apt install python3-pip`) and I presume it was uninstalled automatically when upgrading as bazarr did work perfectly before the upgrade. It seems that installing pip through APT did fix the issue so you can close the issue if you want. Just wanted to inform you guys that the upgrade might break bazarr. btw I followed these instructions for upgrading: https://dietpi.com/blog/?p=811

MichaIng, 2021-09-28:
Ah okay. I'll try to replicate that this package is removed during the distro upgrade. I remember the same happened to MariaDB. This may be a good into to add to the upgrade guide.

MichaIng, 2021-11-11:
Python 3 with pip is now part of the reinstalled software in the blog post. Also in my case all Python 3 packages were migrated during the APT upgrade, but haven't tested `python3-pip` since `dietpi-software` installs the pip module differently. Bazarr works fine after the upgrade, I'll hence mark this as closed.

closed 2021-11-11 by MichaIng