MichaIng/DietPi #4781: something during update removed nginx log file so it failed to start

G2G2G2G, 2021-09-28:
I've solved this issue below, just posting maybe have a fix so people who are new to linux or something don't run into this issue and they don't know what to do.

### Creating a bug report/issue

#### Required Information
- DietPi version | `upgrading from 1 version to other, see logs below`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`  <--- `lsb_release -a` might be better for more platforms

```
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

- Kernel version | `5.10.63-rockchip64 #21.08.2 SMP PREEMPT Wed Sep 8 10:57:23 UTC 2021 aarch64 GNU/Linux`
- SBC model | `NanoPi M4 (old model)`
- Power supply used | https://www.newegg.com/hp-578322-b21-1200w/p/N82E16817292058
- SDcard used | SSD

```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 7.6.2
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v7.5.2
[ INFO ] DietPi-Update | Latest version  : v7.6.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=26863 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : syncthing
[  OK  ] DietPi-Services | stop : nginx
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : nfs-kernel-server
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:2 https://deb.debian.org/debian buster InRelease
Get:3 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Get:1 http://armbian.tnahosting.net/apt buster InRelease [18.3 kB]
Get:4 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Get:5 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Get:6 https://deb.debian.org/debian buster-updates/main arm64 Packages.diff/Index [8116 B]
Get:7 https://deb.debian.org/debian buster-updates/main arm64 Packages 2021-09-18-2009.26.pdiff [1099 B]
Get:7 https://deb.debian.org/debian buster-updates/main arm64 Packages 2021-09-18-2009.26.pdiff [1099 B]
Get:8 http://armbian.tnahosting.net/apt buster/main arm64 Packages [252 kB]
Get:9 https://deb.debian.org/debian-security buster/updates/main arm64 Packages [299 kB]
Get:10 https://deb.debian.org/debian buster-backports/main arm64 Packages.diff/Index [27.8 kB]
Get:11 https://deb.debian.org/debian buster-backports/main arm64 Packages 2021-09-21-1403.11.pdiff [356 B]
Get:11 https://deb.debian.org/debian buster-backports/main arm64 Packages 2021-09-21-1403.11.pdiff [356 B]
Fetched 771 kB in 6s (131 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
[  OK  ] DietPi-Update | APT upgrade

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Installing new DietPi code

[  OK  ] DietPi-Update | Downloading update archive
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Installing new DietPi scripts
[  OK  ] DietPi-Update | Installing new DietPi system files
[  OK  ] DietPi-Update | Setting execute permissions for DietPi scripts
[ SUB1 ] DietPi-Set_software > verify_dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] verify_dietpi.txt  | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches

[ INFO ] DietPi-Update | Current version : v7.5.2
[ INFO ] DietPi-Update | Latest version  : v7.6.2
[ INFO ] DietPi-Patch | Patching to DietPi v7.6...
[  OK  ] DietPi-Patch | Patched to DietPi v7.6
[ INFO ] DietPi-Update | APT autopurge, please wait...
[  OK  ] DietPi-Update | APT autopurge
[  OK  ] DietPi-Update | Incremental patching to v7.6.2 completed
[ INFO ] DietPi-Update | Checking for available live patches
[ INFO ] DietPi-Update | Found valid live patch 0:
 - Description: dummy
 - Condition: false
 - Patch: :
[ INFO ] DietPi-Update | Live patch 0 is not applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[0]='not applicable' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Found valid live patch 1:
 - Description: Fixes a bug in DietPi-LetsEncrypt which makes it fail when ownCloud or Nextcloud are installed: https://github.com/MichaIng/DietPi/issues/4752
 - Condition: grep -Eq "s/'http://localhost/(own|next)cloud'/'https://\\$primary_domain/(own|next)cloud'/" /boot/dietpi/dietpi-letsencrypt
 - Patch: sed -Ei "s#s/'http://localhost/(own|next)cloud'/'https://\\$primary_domain/(own|next)cloud'/#s|'http://localhost/\1cloud'|'https://\$primary_domain/\1cloud'|#" /boot/dietpi/dietpi-letsencrypt
[ INFO ] DietPi-Update | Live patch 1 is applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[1]='not applied' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Found valid live patch 2:
 - Description: Fixes a bug in DietPi-Software where the Buster => Bullseye upgrade, following our guide, fails if HTTPS was enabled via DietPi-LetsEncrypt before: https://dietpi.com/phpbb/viewtopic.php?t=9477
 - Condition: grep -q "lighty-enable-mod openssl" /boot/dietpi/dietpi-software
 - Patch: sed -i "s/lighty-enable-mod openssl/lighty-enable-mod ssl/" /boot/dietpi/dietpi-software
[ INFO ] DietPi-Update | Live patch 2 is applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[2]='not applied' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Found valid live patch 3:
 - Description: Fixes a bug in DietPi-Software where Kodi fails on RPi ARMv8/64-bit systems when installed without a desktop: https://dietpi.com/phpbb/viewtopic.php?p=38079#p38079
 - Condition: grep -q "\$G_HW_MODEL -le 9 || \$G_HW_MODEL == 1\[56\]" /boot/dietpi/dietpi-software
 - Patch: sed -i "s/\$G_HW_MODEL -le 9 || \$G_HW_MODEL == 1\[56\]/\$G_HW_MODEL -le 9 \&\& \$(dpkg --print-architecture) == 'armhf' || \$G_HW_MODEL == 1\[56\]/" /boot/dietpi/dietpi-software
[ INFO ] DietPi-Update | Live patch 3 is applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[3]='not applied' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Applying live patch 1
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[1]='applied'
[ INFO ] DietPi-Update | Applying live patch 2
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[2]='applied'
[ INFO ] DietPi-Update | Applying live patch 3
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[3]='applied'
```

what happened:
```
[ INFO ] DietPi-Update | Current version : v7.6.2
[ INFO ] DietPi-Update | Latest version  : v7.6.2
[  OK  ] DietPi-Survey | Purging survey data
[ SUB1 ] DietPi-Services > restart 
[  OK  ] DietPi-Services | restart : nmbd
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : nfs-kernel-server
[  OK  ] DietPi-Services | restart : php7.3-fpm
[FAILED] DietPi-Services | restart : nginx
[  OK  ] DietPi-Services | restart : syncthing
[  OK  ] DietPi-Services | restart : cron
```

```
nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
2021/09/27 18:43:19 [emerg] 2265#2265: open() "/var/log/nginx/error.log" failed (2: No such file or directory)
```

really annoying about nginx... It's always been like this, not sure why it's so hard to add mkdir -p /path/to/log/ so it doesn't fail ever (on their part, not yours)

**method to fix:**
```
mkdir /var/log/nginx/
```

MichaIng, 2021-09-28:
> really annoying about nginx... It's always been like this, not sure why it's so hard to add mkdir -p /path/to/log/ so it doesn't fail ever (on their part, not yours)

At least there is nothing which would actively remove the Nginx log directory. It is created once during package install and should then stay present. Yes it would be trivial on Buster and Bullseye to include log directory pre-creation natively via the systemd unit. We may suggest that to the Debian maintainer, adding `LogsDirectory=nginx` to the units's `[Service]` block.

If you loose that directory, then probably the tmpfs got unmounted somehow, not sure. Is everything else still present there?
```sh
ls -l /var/log
```

> Just realized it doesn't even mention that pi-hole is on this system

Where do you expect it to be listed? In `dietpi-software` you should see it, e.g. when entering the uninstall sub menu. It is not listed as one of the `DietPi-Services` handled services as it does not handle it:
- As users may choose to resolve the services own DNS requests via Pi-hole, `DietPi-Services` would otherwise break the systems own network name resolution. So it needs to stay enabled during maintenance tasks, installs, updates, etc.
- Pi-hole is instead started at boot by systemd and then not touched anymore by any DietPi script.

Joulinar, 2021-09-28:
Pihole service should be shown if you run `dietpi-services status`

G2G2G2G, 2021-09-28:
>Where do you expect it to be listed? In dietpi-software you should see it, e.g. when entering the uninstall sub menu. It is not listed as one of the DietPi-Services handled services as it does not handle it:

that is where I expected it to be listed (you seem to know that is where I expected it to be listed because you answered why it is not listed there)
It's running fine, that wasn't an issue, just was giving more info as to why/how nginx is installed on my specific system, (it is installed via your pihole software package) (in case somethings are different from that vs others)
>As users may choose to resolve the services own DNS requests via Pi-hole, DietPi-Services would otherwise break the systems own network name resolution. So it needs to stay enabled during maintenance tasks, installs, updates, etc.

yea this makes sense, when running I watched via another ssh session to see if it shut off because I was curious (months ago) so I knew it didn't shut off, just never really look at the update output so never noticed it wasn't there

Nothing else seemed to have issues at all, I already checked that at the time of posting. It was fine then and I have rebooted since but it looks like pretty much how it was. (and I did see it was tmpfs via `df -h` at the time as well)
(I ran this list 2 mins ago not at the time of posting so some dates are today)
```
ls -thal /var/log/
total 12K
-rw-rw-r--  1 root   utmp   290K Sep 28 10:41 lastlog
-rw-rw-r--  1 root   utmp    400 Sep 28 10:41 wtmp
-rw-r--r--  1 pihole pihole    0 Sep 28 10:17 pihole-FTL.log
-rw-r--r--  1 pihole pihole    0 Sep 28 00:17 pihole-FTL.log.1
drwxr-xr-t  6 root   root    380 Sep 28 00:00 ./
-rw-r--r--  1 pihole pihole    0 Sep 27 19:17 pihole.log
-rw-------  1 root   root      0 Sep 27 19:17 php7.3-fpm.log
-rw-rw----  1 root   utmp      0 Sep 27 19:06 btmp
drwxr-xr-x  2 root   root     60 Sep 27 18:44 nginx/
drwxr-x---  3 root   adm     380 Sep 27 18:41 samba/
-rw-r--r--  1 pihole pihole    0 Sep 27 18:41 pihole.log.1
-rw-r--r--  1 root   root      0 Sep 15 08:34 dpkg.log
drwxr-xr-x  2 root   root    100 Sep 15 08:32 apt/
-rw-r-----  1 root   adm       0 Sep  5 00:00 fail2ban.log
-rw-rw----  1 root   utmp      0 Aug 30 23:56 btmp.1
-rw-r--r--  1 root   root      0 Aug 29 21:17 alternatives.log
-rw-r-----  1 root   adm       0 Aug 29 00:01 fail2ban.log.1
drwx------  2 root   root     40 Aug  8  2020 private/
```

All looks good after adding that mkdir and rebooting to make sure it all was working when reboot too so it was back to normal. Not sure where else to look

MichaIng, 2021-09-28:
Okay, as it seems to work now, it should be fine. Check back when the Nginx log directory is missing again. A possible explanation would be if you installed Nginx and did an unclean reboot/power off afterwards, as in this case the /var/log directory structure is not stored to disk to be re-loaded on boot.

closed 2021-11-11 by MichaIng