MichaIng/DietPi #4782: Chromium Kiosk broken

Loader23, 2021-09-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7`
- Distro version | `bullseye`
- Kernel version | `Linux DietPi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux`
- SBC model | `RPi4` or (EG: RPi3)
- Power supply used | Official RPi
- SDcard used | SanDisk Extreme

#### Additional Information (if applicable)
- Software title | Chromium
- Was the software title installed freshly or updated/migrated? fresh installation
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Install Chromium
2. Setup Autoboot to Chromium

#### Expected behaviour

- Chromium should start

#### Actual behaviour

This error:

```
Build Date: 05 August 2021  11:35:55AM
xorg-server 2:1.20.11-1+rpt1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Sep 28 09:14:55 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
/usr/bin/chromium-browser: line 134: lsb_release: command not found
/usr/bin/chromium-browser: line 135: lsb_release: command not found
dpkg-query: no packages found matching bluealsa
[0928/091456.599447:ERROR:elf_dynamic_array_reader.h(61)] tag not found
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
```

#### Extra details

When installing lsb-release the two lsb errors are gone but the other ones remain.

MichaIng, 2021-09-28:
Many thanks for your report.

Can you check the X.org logs, please:
```
cat /var/log/Xorg.0.log
```

Loader23, 2021-09-28:
Here it is :-)
```
[    14.761]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[    14.761] Build Operating System: linux Raspbian
[    14.761] Current Operating System: Linux DietPi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l
[    14.761] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:F6:E9:72 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=PARTUUID=80ee19d2-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=ttyS0,115200 console=tty1
[    14.762] Build Date: 05 August 2021  11:35:55AM
[    14.762] xorg-server 2:1.20.11-1+rpt1 (https://www.debian.org/support)
[    14.762] Current version of pixman: 0.40.0
[    14.762]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    14.762] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    14.762] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Sep 28 14:19:53 2021
[    14.775] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    14.775] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    14.777] (==) No Layout section.  Using the first Screen section.
[    14.777] (==) No screen section available. Using defaults.
[    14.777] (**) |-->Screen "Default Screen Section" (0)
[    14.777] (**) |   |-->Monitor "<default monitor>"
[    14.781] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    14.781] (**) Option "BlankTime" "0"
[    14.781] (==) Automatically adding devices
[    14.781] (==) Automatically enabling devices
[    14.781] (==) Automatically adding GPU devices
[    14.785] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    14.790] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    14.790]    Entry deleted from font path.
[    14.790] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    14.790]    Entry deleted from font path.
[    14.790] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    14.790]    Entry deleted from font path.
[    14.790] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    14.790]    Entry deleted from font path.
[    14.790] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    14.790]    Entry deleted from font path.
[    14.790] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    14.790]    Entry deleted from font path.
[    14.790] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[    14.790] (==) ModulePath set to "/usr/lib/xorg/modules"
[    14.790] (**) Extension "DPMS" is disabled
[    14.790] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    14.790] (II) Loader magic: 0x20cf40
[    14.790] (II) Module ABI versions:
[    14.790]    X.Org ANSI C Emulation: 0.4
[    14.790]    X.Org Video Driver: 24.1
[    14.790]    X.Org XInput driver : 24.1
[    14.790]    X.Org Server Extension : 10.0
[    14.791] (--) using VT number 2

[    14.791] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    14.793] (II) no primary bus or device found
[    14.793] (II) LoadModule: "glx"
[    14.794] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    14.830] (II) Module glx: vendor="X.Org Foundation"
[    14.830]    compiled for 1.20.11, module version = 1.0.0
[    14.830]    ABI class: X.Org Server Extension, version 10.0
[    14.830] (==) Matched modesetting as autoconfigured driver 0
[    14.830] (==) Matched fbdev as autoconfigured driver 1
[    14.830] (==) Assigned the driver to the xf86ConfigLayout
[    14.831] (II) LoadModule: "modesetting"
[    14.832] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[    14.835] (II) Module modesetting: vendor="X.Org Foundation"
[... 62 lines cut ...]
[    14.879] (II) Initializing extension RENDER
[    14.880] (II) Initializing extension RANDR
[    14.881] (II) Initializing extension COMPOSITE
[    14.881] (II) Initializing extension DAMAGE
[    14.882] (II) Initializing extension MIT-SCREEN-SAVER
[    14.882] (II) Initializing extension DOUBLE-BUFFER
[    14.883] (II) Initializing extension RECORD
[    14.883] (II) Initializing extension Present
[    14.884] (II) Initializing extension DRI3
[    14.884] (II) Initializing extension X-Resource
[    14.884] (II) Initializing extension XVideo
[    14.885] (II) Initializing extension XVideo-MotionCompensation
[    14.885] (II) Initializing extension SELinux
[    14.885] (II) SELinux: Disabled on system
[    14.885] (II) Initializing extension GLX
[    14.885] (II) AIGLX: Screen 0 is not DRI2 capable
[    15.826] (II) IGLX: Loaded and initialized swrast
[    15.826] (II) GLX: Initialized DRISWRAST GL provider for screen 0
[    15.826] (II) Initializing extension XFree86-VidModeExtension
[    15.826] (II) Initializing extension XFree86-DGA
[    15.827] (II) Initializing extension XFree86-DRI
[    15.829] (II) Initializing extension DRI2
[    18.527] (II) Server terminated successfully (0). Closing log file.

```

MichaIng, 2021-09-28:
It tires to load the X server via fbdev. That should generally work as well, but could you try to switch to FKMS driver:
```sh
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-fkms-v3d
reboot
```

Loader23, 2021-09-28:
No luck:

```

[    12.364]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[    12.364] Build Operating System: linux Raspbian
[    12.364] Current Operating System: Linux DietPi 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l
[    12.364] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:F6:E9:72 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=PARTUUID=80ee19d2-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=ttyS0,115200 console=tty1
[    12.365] Build Date: 05 August 2021  11:35:55AM
[    12.365] xorg-server 2:1.20.11-1+rpt1 (https://www.debian.org/support)
[    12.365] Current version of pixman: 0.40.0
[    12.365]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    12.365] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    12.366] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Sep 28 14:40:27 2021
[    12.386] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    12.386] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    12.389] (==) No Layout section.  Using the first Screen section.
[    12.389] (==) No screen section available. Using defaults.
[    12.389] (**) |-->Screen "Default Screen Section" (0)
[    12.389] (**) |   |-->Monitor "<default monitor>"
[    12.393] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    12.393] (**) Option "BlankTime" "0"
[    12.393] (==) Automatically adding devices
[    12.393] (==) Automatically enabling devices
[    12.393] (==) Automatically adding GPU devices
[    12.397] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    12.402] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    12.402]    Entry deleted from font path.
[    12.402] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    12.402]    Entry deleted from font path.
[    12.402] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    12.402]    Entry deleted from font path.
[    12.403] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    12.403]    Entry deleted from font path.
[    12.403] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    12.403]    Entry deleted from font path.
[    12.403] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    12.403]    Entry deleted from font path.
[    12.403] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[    12.403] (==) ModulePath set to "/usr/lib/xorg/modules"
[    12.403] (**) Extension "DPMS" is disabled
[    12.403] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    12.403] (II) Loader magic: 0x20cf40
[    12.403] (II) Module ABI versions:
[    12.403]    X.Org ANSI C Emulation: 0.4
[    12.403]    X.Org Video Driver: 24.1
[    12.403]    X.Org XInput driver : 24.1
[    12.403]    X.Org Server Extension : 10.0
[    12.404] (--) using VT number 2

[    12.404] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    12.406] (II) xfree86: Adding drm device (/dev/dri/card1)
[    12.417] (II) xfree86: Adding drm device (/dev/dri/card0)
[    12.419] (II) no primary bus or device found
[    12.419]    falling back to /sys/devices/platform/gpu/drm/card1
[    12.419] (II) LoadModule: "glx"
[    12.420] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    12.456] (II) Module glx: vendor="X.Org Foundation"
[    12.456]    compiled for 1.20.11, module version = 1.0.0
[    12.456]    ABI class: X.Org Server Extension, version 10.0
[    12.456] (==) Matched modesetting as autoconfigured driver 0
[... 168 lines cut ...]
[    14.309] (II) Initializing extension RANDR
[    14.310] (II) Initializing extension COMPOSITE
[    14.311] (II) Initializing extension DAMAGE
[    14.311] (II) Initializing extension MIT-SCREEN-SAVER
[    14.312] (II) Initializing extension DOUBLE-BUFFER
[    14.312] (II) Initializing extension RECORD
[    14.313] (II) Initializing extension Present
[    14.313] (II) Initializing extension DRI3
[    14.314] (II) Initializing extension X-Resource
[    14.314] (II) Initializing extension XVideo
[    14.315] (II) Initializing extension XVideo-MotionCompensation
[    14.315] (II) Initializing extension SELinux
[    14.315] (II) SELinux: Disabled on system
[    14.315] (II) Initializing extension GLX
[    14.681] (II) AIGLX: Loaded and initialized vc4
[    14.681] (II) GLX: Initialized DRI2 GL provider for screen 0
[    14.681] (II) Initializing extension XFree86-VidModeExtension
[    14.682] (II) Initializing extension XFree86-DGA
[    14.682] (II) Initializing extension XFree86-DRI
[    14.684] (II) Initializing extension DRI2
[    14.687] (II) modeset(0): Damage tracking initialized
[    14.687] (II) modeset(0): Setting screen physical size to 508 x 285
[    17.354] (II) Server terminated successfully (0). Closing log file.

```

MichaIng, 2021-09-28:
Strange is that it does not show any error, just "Server terminated successfully (0).". Can you try to start Chromium manually from console, please:
```sh
xinit "$(command -v chromium-browser)"
# and
xinit "$(command -v chromium-browser)" --use-gl=egl
```

Loader23, 2021-09-29:
Same, does not start.

Loader23, 2021-10-01:
Any Updates on this? Should be easy to reproduce.

MichaIng, 2021-10-13:
If you have a desktop installed, can you try to call `chromium-browser` from a terminal emulator? I get an `illegal instruction` on ARMv6 and from console (outside desktop) the same symptoms like you. Will further test on RPi 2 so we have covered the major RPi SoCs.

MichaIng, 2021-10-13:
Reported: https://github.com/RPi-Distro/chromium-browser/issues/22
I'll add it manually until it got added to the packages's dependencies.

MichaIng, 2021-10-13:
Workaround added with: https://github.com/MichaIng/DietPi/commit/ac69f15ed16ca464d4f87d3f5b9192a657747bf4

Loader23, 2021-10-13:
Just tested it and its working fine now, thanks alot :-)

albone3000, 2021-10-16:
installing libgtk-3-0 on my system fixed this problem as well.

closed 2021-10-16 by MichaIng

MichaIng, 2021-12-07:
Now solved in the Debian package.