MichaIng/DietPi #4783: Amiberry image with Hstwb installer support

symbios24, 2021-09-28:
Hi there
Is it possible to add support for the hstwb installer in the Amiberry Image??
To make easier the installation of amigaos,workench's etc

https://hstwb.firstrealize.com/

https://github.com/henrikstengaard/hstwb-installer/wiki/HstWB-Installer-for-Amibian

Thanks

MichaIng, 2021-09-28:
Can this be used with Amiberry directly somehow? Sorry I'm not very experienced with Amiga emulation 😅. Else this might be better a separate install option.

symbios24, 2021-09-28:
It can be used but in amibian distro that is very old and based in raspbian stretch,now i dont know its its going to work in Amiberry image in dietpi,i can try to install it one of this days and let you know,it will be great if you can include it anyway,it automates the amiga workbenxh instalations.it can be included as an extra option in Amiberry installation.