MichaIng/DietPi #4784: Bazarr fail to start

Dr0bac, 2021-09-29:
Rpi3 buster v7.6.2

I tried to go to bazarr web page, and i saw this message `This site can’t be reached`
After that i ssh to rpi3 and in htop bazarr was missing.
So i entered `systemctl status bazarr.service`

````
root@DietPi:~# systemctl status bazarr.service
● bazarr.service - Bazarr Daemon (DietPi)
   Loaded: loaded (/etc/systemd/system/bazarr.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2021-09-29 01:57:11 CEST; 7min ago
  Process: 631 ExecStart=/usr/bin/python3 /opt/bazarr/bazarr.py -c /mnt/dietpi_userdata/bazarr (code=exited, status=2)
 Main PID: 631 (code=exited, status=2)

Sep 29 01:57:10 DietPi systemd[1]: Started Bazarr Daemon (DietPi).
Sep 29 01:57:11 DietPi Bazarr[631]: /usr/bin/python3: can't open file '/opt/bazarr/bazarr.py': [Errno 2] No such file or directory
Sep 29 01:57:11 DietPi systemd[1]: bazarr.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Sep 29 01:57:11 DietPi systemd[1]: bazarr.service: Failed with result 'exit-code'.
root@DietPi:~# cd /opt/bazarr/
root@DietPi:/opt/bazarr# ls -l
total 8
drwxr-xr-x 6 root   root   4096 Jul 23 10:29 data
drwxr-xr-x 8 bazarr dietpi 4096 Sep 21 09:37 libs
````

i see that bazarr.py is missing, how is that possible?

Joulinar, 2021-09-29:
It would be helpful if you could fill the bug report.

```
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)
```

Dr0bac, 2021-09-29:
````
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='not applied'
G_LIVE_PATCH_STATUS[2]='not applied'
G_LIVE_PATCH_STATUS[3]='not applied'
````

````
buster 1 
````

````
Linux DietPi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux
````

````
RPi 3 Model B (armv7l)  
````

Official raspberry pi 3 power adapter

Sandisk extreme

Joulinar, 2021-09-29:
Do you have a backup available? You could try to copy missing files over.

Dr0bac, 2021-09-29:
No, i didn't backup.

I tried to find bazarr.py but it doesn't exist.

````
root@DietPi:/# find . -name bazarr.py                                           root@DietPi:/# find . -name bazarr                                              ./mnt/database/dietpi_userdata/bazarr
./var/tmp/dietpi/logs/dietpi-ramlog_store/bazarr
./var/log/bazarr
./opt/bazarr
root@DietPi:/# cd /opt/bazarr/                                                  root@DietPi:/opt/bazarr# ls -al
total 16
drwxr-xr-x 4 bazarr dietpi 4096 Sep 21 09:38 .
drwxr-xr-x 5 root   root   4096 Jan  2  2021 ..
drwxr-xr-x 6 root   root   4096 Jul 23 10:29 data
drwxr-xr-x 8 bazarr dietpi 4096 Sep 21 09:37 libs
````

Joulinar, 2021-09-29:
Probably a reinstallation would be needed if Bazarr scripts are missing

```
dietpi-software reinstall 180
```

MichaIng, 2021-09-29:
Maybe its internal updater failed?

Dr0bac, 2021-09-30:
I asked in bazarr discord server and developer told me this:

_@Boris there's been an issue with Github API when we tried to release 0.9.10-beta.4 and it released a broken package. Sorry about that but I hope it wont happen again._

So i will reinstall bazarr then, do i need to go trough uninstall process in dietpi-software and install it again with dietpi-software or can i just enter `dietpi-software reinstall 180`?

MichaIng, 2021-09-30:
You can do a reinstall, but you need to remove the install directory first, else `dietpi-software` will tell you to do so or use the internal updater:
```sh
rm -R /opt/bazarr
dietpi-software reinstall 180
```

closed 2021-09-30 by Dr0bac