MichaIng/DietPi #4785: Deluge 2.0.3 on Bulleye

quyentruong, 2021-09-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.6.2
- Distro version | bullseye 1
- Kernel version | Linux DietPi 5.10.60-v7l+ # 1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)

#### Additional Information (if applicable)
- Software title | Deluge
- Was the software title installed freshly or updated/migrated? Updated

#### Steps to reproduce

1. Follow this guild to upgrade OS to Bulleye https://dietpi.com/blog/?p=811#2.2-manual-upgrade
2. dietpi-software reinstall 45

#### Expected behaviour

- deluge should work

#### Actual behaviour

- deluge-web service doesn't run

#### Extra details

- I tried this https://forum.deluge-torrent.org/viewtopic.php?t=55379 to add -d option to deluge-web.service
- deluge-web service is working now, but deluged service fails when I go to the deluge website.

Joulinar, 2021-09-29:
Pls can you share error message of the service.

quyentruong, 2021-09-29:
[image]

[image]

MichaIng, 2021-09-29:
Fix is up: #4787

Joulinar, 2021-09-29:
Ok this fix the missing `-d` option on the web service but I guess `deluged.service` is still failing. Could you share some more logs

`journalctl -u deluged`

MichaIng, 2021-09-29:
Ah yes that is true.

quyentruong, 2021-09-29:
I guess the issue is `ModuleNotFoundError: No module named 'deluge.libtorrent'`

[deluged.txt](https://github.com/MichaIng/DietPi/files/7252321/deluged.txt)

Joulinar, 2021-09-29:
There is a Debian bug report from end of last year https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=975891

Looks like it's related to package `libtorrent-rasterbar` but should have been fixed by package maintainer already. 🤔

Joulinar, 2021-09-29:
I guess issue was closed automatically due to merging the service fix.

MichaIng, 2021-09-29:
Check the package versions:
```sh
dpkg -l | grep deluge
```

Probably it is somehow a similar issue than what we saw with curl when doing the Bullseye => Buster upgrade, just the other way round, somehow a badly manipulated `python3-libtorrent` or `libtorrent-rasterbar10` file. Please try to fix it via reinstall:
```sh
apt install --reinstall python3-libtorrent libtorrent-rasterbar10
```

Joulinar, 2021-09-29:
@MichaIng 
I did a fresh install and it's failing with same error.

MichaIng, 2021-09-29:
On Raspberry Pi? Because on VM it works well here.

Joulinar, 2021-09-29:
yes correct. RPi3B+ ARMv6 Bullseye

MichaIng, 2021-09-29:
Verified.
```
ImportError: /usr/lib/arm-linux-gnueabihf/libtorrent-rasterbar.so.10: undefined symbol: __atomic_fetch_add_8
```
Installing `libatomic1` does not solve it, which is an issue in a few other cases on ARM.

`libatomic1` likely is required, however, at least on ARMv6, as it is a dependency on Debian for `armel` (=ARMv6, though non hard-float): https://packages.debian.org/bullseye/libtorrent-rasterbar10
But again, installing it doesn't help here.

Found the bug report at Deluge: https://dev.deluge-torrent.org/ticket/3437

MichaIng, 2021-09-29:
Bug reported at Raspbian: https://bugs.launchpad.net/raspbian/+bug/1945530

MichaIng, 2021-10-16:
There is two issues combined here, one fixed, the other one pending but needs to be fixed by the Raspbian package maintainers. I'll keep this issue open to keep track on this and in case push the fix.

MichaIng, 2021-11-08:
External bug split off: #4944

closed 2021-11-08 by MichaIng