MichaIng/DietPi #4786: `.hushlogin` has no effect

dnknth, 2021-09-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator | DietPi Core Team
- Pre-image | from scratch
- Hardware | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux
- Distro | bullseye (ID=6,RASPBIAN=0)

#### Steps to reproduce
1. touch ~/.hushlogin
2. Log out
3. Log in again

#### Expected behaviour
- System summary and Dietpi related information should not appear, just a plain shell prompt.

#### Actual behaviour
- Screen is cleared and filled with system summary and Dietpi related information.

MichaIng, 2021-09-29:
Many thanks for your request.

I never heard of it, but `~/.hushlogin` seems to prevent common shells from printing `/etc/motd` to the console. We want to merge the DietPi banner into `/etc/motd`, so then it will work naturally, but until then we can implement it manually into the [dietpi-login](https://github.com/MichaIng/DietPi/blob/872b98/dietpi/dietpi-login#L248) script. Makes sense to respect common UNIX methods.

An additional nice thing would be to have an option in `dietpi-banner` itself to disable it completely.

dnknth, 2021-09-29:
It normally suppresses any blurb when logging in, both remotely via SSH or at the console.

MichaIng, 2021-09-29:
It's usually about `/etc/motd`, which is the common way to have information printed to console on login. But I'm sure other tools who do not use `/etc/motd` (e.g. when dynamic content like in the DietPi banner is shown) may also respect it manually.

Just implemented for next version: https://github.com/MichaIng/DietPi/commit/6115fcd3d121fa4c9b0a51283f6d1c78430826a5

closed 2021-09-29 by dnknth