MichaIng/DietPi #4794: Trying to update DietPi from v7.3.2 to v7.6.2

Ianszh, 2021-10-02:
#### Details:
- Date           | Sat Oct  2 15:13:07 CEST 2021
- DietPi version | v7.3.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. Run `DietPi-Update`
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://repos.influxdata.com/debian stretch InRelease
Hit:3 https://packages.grafana.com/oss/deb stable InRelease
Ign:4 https://repo.mosquitto.org/debian stretch InRelease
Hit:5 https://archive.raspberrypi.org/debian stretch InRelease
Err:6 https://repo.mosquitto.org/debian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists...
E: The repository 'https://repo.mosquitto.org/debian stretch Release' does no longer have a Release file.
```

MichaIng, 2021-10-02:
Many thanks for your report.

Please try the following:
```sh
apt install ca-certificates
apt update
apt upgrade
```

Ianszh, 2021-10-02:
Thanks that did the trick, but unfortunately now InfluxDB will no longer start!

`Phase: Completed

[ INFO ] DietPi-Update | Current version : v7.6.2
[ INFO ] DietPi-Update | Latest version  : v7.6.2
[  OK  ] DietPi-Survey | Sending survey data
[ SUB1 ] DietPi-Services > restart 
[  OK  ] DietPi-Services | restart : proftpd
[  OK  ] DietPi-Services | restart : nmbd
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : mosquitto
[  OK  ] DietPi-Services | restart : node-red
[FAILED] DietPi-Services | restart : influxdb
[  OK  ] DietPi-Services | restart : grafana-server
[  OK  ] DietPi-Services | restart : cron
root@DietPi:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status 

[  OK  ] DietPi-Services | proftpd		active (running) since Sat 2021-10-02 15:33:58 CEST; 26s ago
[  OK  ] DietPi-Services | nmbd			active (running) since Sat 2021-10-02 15:33:58 CEST; 26s ago
[  OK  ] DietPi-Services | smbd			active (running) since Sat 2021-10-02 15:33:58 CEST; 26s ago
[  OK  ] DietPi-Services | mosquitto		active (running) since Sat 2021-10-02 15:33:59 CEST; 25s ago
[  OK  ] DietPi-Services | node-red		active (running) since Sat 2021-10-02 15:33:59 CEST; 26s ago
[  OK  ] DietPi-Services | influxdb		activating (start) since Sat 2021-10-02 15:34:21 CEST; 3s ago
[  OK  ] DietPi-Services | grafana-server	active (running) since Sat 2021-10-02 15:34:10 CEST; 15s ago
[  OK  ] DietPi-Services | cron			active (running) since Sat 2021-10-02 15:34:10 CEST; 15s ago
[  OK  ] DietPi-Services | dropbear		active (running) since Sun 2021-06-27 12:13:45 CEST; 3 months 5 days ago
[  OK  ] DietPi-Services | openvpn		active (exited) since Thu 2016-11-03 18:16:46 CET; 4 years 10 months ago
[  OK  ] DietPi-Services | nxserver		active (running) since Thu 2016-11-03 18:16:46 CET; 4 years 10 months ago
[ INFO ] DietPi-Services | dietpi-vpn		inactive (dead)
[  OK  ] DietPi-Services | dietpi-ramlog	active (exited) since Thu 2016-11-03 18:16:46 CET; 4 years 10 months ago
[  OK  ] DietPi-Services | dietpi-preboot	active (exited) since Thu 2016-11-03 18:16:46 CET; 4 years 10 months ago
[  OK  ] DietPi-Services | dietpi-boot		active (exited) since Sun 2021-06-27 12:13:45 CEST; 3 months 5 days ago
[  OK  ] DietPi-Services | dietpi-postboot	active (exited) since Sun 2021-06-27 12:13:45 CEST; 3 months 5 days ago`

Joulinar, 2021-10-02:
Just a side node and not related to the issue. You are running oldold Debian Stretch version. You migh consider to upgrade to Buster https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

Joulinar, 2021-10-02:
Did you tried to reboot?

MichaIng, 2021-10-02:
Please paste the service logs for InfluxDB:
```sh
journalctl -u influxdb
```

Ianszh, 2021-10-02:
Yes, but it's stuck at `activating (start)` 

It's quite long so here's part of it

`-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sat 2021-10-02 15:43:44 CEST. --
Nov 03 18:16:46 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.162248Z lvl=info msg="InfluxDB starting" log_id=0WxJa6dW000 ver
sion=1.8.9 branch=1.8 commit=d9b56321d579
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.162358Z lvl=info msg="Go runtime" log_id=0WxJa6dW000 version=go
1.13.8 maxprocs=4
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.302656Z lvl=info msg="Using data dir" log_id=0WxJa6dW000 servic
e=store path=/var/lib/influxdb/data
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.302900Z lvl=info msg="Compaction settings" log_id=0WxJa6dW000 s
ervice=store max_concurrent_compactions=2 throughput_bytes_per_second=50331648 throughput_bytes_per_second_burst=50331648
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.302986Z lvl=info msg="Open store (start)" log_id=0WxJa6dW000 se
rvice=store trace_id=0WxJa7BW000 op_name=tsdb_open op_event=start
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.526586Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=ts
m1 service=filestore path=/var/lib/influxdb/data/Balcony/autogen/1064/000000130-000000002.tsm id=0 duration=18.620ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.545644Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=ts
m1 service=filestore path=/var/lib/influxdb/data/Balcony/autogen/1040/000000004-000000002.tsm id=0 duration=18.487ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.547325Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=ts
m1 service=filestore path=/var/lib/influxdb/data/Balcony/autogen/1094/000000047-000000002.tsm id=0 duration=39.319ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.549280Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=ts
m1 service=filestore path=/var/lib/influxdb/data/Balcony/autogen/1079/000000192-000000002.tsm id=0 duration=39.104ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.896090Z lvl=info msg="Opened shard" log_id=0WxJa6dW000 service=
store trace_id=0WxJa7BW000 op_name=tsdb_open index_version=inmem path=/var/lib/influxdb/data/Balcony/autogen/1040 duration=542.035ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.896301Z lvl=info msg="Opened shard" log_id=0WxJa6dW000 service=
store trace_id=0WxJa7BW000 op_name=tsdb_open index_version=inmem path=/var/lib/influxdb/data/Balcony/autogen/1094 duration=542.173ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.896280Z lvl=info msg="Opened shard" log_id=0WxJa6dW000 service=
store trace_id=0WxJa7BW000 op_name=tsdb_open index_version=inmem path=/var/lib/influxdb/data/Balcony/autogen/1064 duration=542.226ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.896322Z lvl=info msg="Opened shard" log_id=0WxJa6dW000 service=
store trace_id=0WxJa7BW000 op_name=tsdb_open index_version=inmem path=/var/lib/influxdb/data/Balcony/autogen/1079 duration=542.186ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.904774Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=ts
m1 service=filestore path=/var/lib/influxdb/data/CPO/autogen/1004/000000002-000000002.tsm id=0 duration=1.300ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.908535Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=ts
m1 service=filestore path=/var/lib/influxdb/data/CPO/autogen/590/000000003-000000002.tsm id=0 duration=4.392ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.914859Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=ts
m1 service=filestore path=/var/lib/influxdb/data/CPO/autogen/597/000000002-000000002.tsm id=0 duration=7.997ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.917409Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=ts
m1 service=filestore path=/var/lib/influxdb/data/Balcony/autogen/1049/000000033-000000002.tsm id=0 duration=6.670ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.922338Z lvl=info msg="Opened shard" log_id=0WxJa6dW000 service=
store trace_id=0WxJa7BW000 op_name=tsdb_open index_version=inmem path=/var/lib/influxdb/data/CPO/autogen/1004 duration=24.272ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.928560Z lvl=info msg="Opened shard" log_id=0WxJa6dW000 service=
store trace_id=0WxJa7BW000 op_name=tsdb_open index_version=inmem path=/var/lib/influxdb/data/CPO/autogen/590 duration=30.443ms
Oct 02 15:40:16 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:16.939525Z lvl=info msg="Opened shard" log_id=0WxJa6dW000 service=
store trace_id=0WxJa7BW000 op_name=tsdb_open index_version=inmem path=/var/lib/influxdb/data/CPO/autogen/597 duration=41.512ms
[... cut ...]
Oct 02 15:40:17 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:17.148597Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=tsm1 service=filestore p
ath=/var/lib/influxdb/data/EnviroPlus/autogen/1108/000000001-000000001.tsm id=0 duration=7.467ms
Oct 02 15:40:17 DietPi influxd-systemd-start.sh[498]: ts=2021-10-02T13:40:17.150932Z lvl=info msg="Opened file" log_id=0WxJa6dW000 engine=tsm1 service=filestore p
ath=/var/lib/influxdb/data/EnviroPlus/autogen/1048/000000009-000000002.tsm id=0 duration=9.492ms`

MichaIng, 2021-10-02:
Please show the lower end (newest) logs as here I cannot see any error. It may make sense to reduce the log level to not show "info" but only warnings and errors:
```sh
G_CONFIG_INJECT 'level = ' 'level = "warn"' /etc/influxdb/influxdb.conf
```

Ianszh, 2021-10-02:
Sorry if you could help me here, I the logging level to `warn` but it still emits the whole log, do I have to enter the whole line 

`G_CONFIG_INJECT 'level = ' 'level = "warn"'`?

`###
### [logging]
###
### Controls how the logger emits logs to the output.
###

[logging]
  # Determines which log encoder to use for logs. Available options
  # are auto, logfmt, and json. auto will use a more a more user-friendly
  # output format if the output terminal is a TTY, but the format is not as
  # easily machine-readable. When the output is a non-TTY, auto will use
  # logfmt.
  # format = "auto"

  # Determines which level of logs will be emitted. The available levels
  # are error, warn, info, and debug. Logs that are equal to or above the
  # specified level will be emitted.
  # level = "warn"

  # Suppresses the logo output that is printed when the program is started.
  # The logo is always suppressed if STDOUT is not a TTY.
  # suppress-logo = false
`

Ianszh, 2021-10-02:
Sorry if you could help me here, I the logging level to warn but it still emits the whole log, do I have to enter the whole line

G_CONFIG_INJECT 'level = ' 'level = "warn"'?

`###
### [logging]
###
### Controls how the logger emits logs to the output.
###

[logging]
  # Determines which log encoder to use for logs. Available options
  # are auto, logfmt, and json. auto will use a more a more user-friendly
  # output format if the output terminal is a TTY, but the format is not as
  # easily machine-readable. When the output is a non-TTY, auto will use
  # logfmt.
  # format = "auto"

  # Determines which level of logs will be emitted. The available levels
  # are error, warn, info, and debug. Logs that are equal to or above the
  # specified level will be emitted.
  # level = "warn"

  # Suppresses the logo output that is printed when the program is started.
  # The logo is always suppressed if STDOUT is not a TTY.
  # suppress-logo = false
`

Ianszh, 2021-10-02:
Sorry if you could help me here, I the logging level to warn but it still emits the whole log, do I have to enter the whole line

`G_CONFIG_INJECT 'level = ' 'level = "warn"'`?

`# Determines which level of logs will be emitted. The available levels
  # are error, warn, info, and debug. Logs that are equal to or above the
  # specified level will be emitted.
  # level = "warn"`

Joulinar, 2021-10-02:
Restart the system should clear the log and create a fresh one

Ianszh, 2021-10-02:
> Just a side node and not related to the issue. You are running oldold Debian Stretch version. You migh consider to upgrade to Buster https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

Will do thanks

MichaIng, 2021-10-02:
That is the right setting you pasted. The command should uncomment it (remove the leading `#`). And yes, a service restart is required to have it taking effect, but since it fails anyway, no need to do that manually 😄. Still, the end of the journal logs are still required to see what the issue actually is.

Ianszh, 2021-10-02:
`root@DietPi:~# journalctl -u influxdb
-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sat 2021-10-02 16:16:20 CEST. --
Nov 03 18:16:46 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 02 16:16:10 DietPi influxd-systemd-start.sh[484]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 16:16:10 DietPi influxd-systemd-start.sh[484]: Merging with configuration at: /etc/influxdb/influxdb.conf
`

Ianszh, 2021-10-02:
`-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sat 2021-10-02 16:17:55 CEST. --
Nov 03 18:16:46 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 02 16:16:10 DietPi influxd-systemd-start.sh[484]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 16:16:10 DietPi influxd-systemd-start.sh[484]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 16:16:22 DietPi systemd[1]: influxdb.service: Control process exited, code=exited status=1
Oct 02 16:16:22 DietPi systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Oct 02 16:16:22 DietPi systemd[1]: influxdb.service: Unit entered failed state.
Oct 02 16:16:22 DietPi systemd[1]: influxdb.service: Failed with result 'exit-code'.
Oct 02 16:16:22 DietPi systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Oct 02 16:16:22 DietPi systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Oct 02 16:16:22 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 02 16:16:22 DietPi influxd-systemd-start.sh[1110]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 16:16:22 DietPi influxd-systemd-start.sh[1110]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 16:16:33 DietPi influxd-systemd-start.sh[1110]: Failed to reach influxdb http endpoint at http://localhost:8086/health
Oct 02 16:16:33 DietPi systemd[1]: influxdb.service: Control process exited, code=exited status=1
Oct 02 16:16:33 DietPi systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Oct 02 16:16:33 DietPi systemd[1]: influxdb.service: Unit entered failed state.
Oct 02 16:16:33 DietPi systemd[1]: influxdb.service: Failed with result 'exit-code'.
Oct 02 16:16:33 DietPi systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Oct 02 16:16:33 DietPi systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Oct 02 16:16:33 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 02 16:16:34 DietPi influxd-systemd-start.sh[1286]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 16:16:34 DietPi influxd-systemd-start.sh[1286]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 02 16:16:45 DietPi systemd[1]: influxdb.service: Control process exited, code=exited status=1
Oct 02 16:16:45 DietPi systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
`

MichaIng, 2021-10-02:
It could be simply that it takes more than 10 seconds before the database HTTP endpoint is reachable, in which case it fails. Latest InfluxDB has a pretty crippled systemd startup wrapper script. There would be much better way to achieve the intention. Can you please try the following:
```sh
sudo -u influxdb /usr/bin/influxd -config /etc/influxdb/influxdb.conf
```
What hardware you're running this on?

Ianszh, 2021-10-02:
Thanks, still not running I'm afraid:

`run: open server: listen: listen tcp 127.0.0.1:8088: bind: address already in use`
`OK  ] DietPi-Services | influxdb		activating (start) since Sat 2021-10-02 17:10:24 CEST; 10s ago`

No communication on my Grafana graph

I'm running on a RPi 3 Model B (armv7l)

Joulinar, 2021-10-02:
Service would need to be stopped first before starting manually. I guess the failing service is trying to restart again and again blocking the port

MichaIng, 2021-10-02:
> run: open server: listen: listen tcp 127.0.0.1:8088: bind: address already in use

This means there is an instance up already. Not sure whether InfluxDB is in an endless loop. Please try again:
```sh
systemctl stop influxdb
sudo -u influxdb /usr/bin/influxd -config /etc/influxdb/influxdb.conf
```

Ianszh, 2021-10-02:
Great thanks now it's running again, it did seem to me to be in an endless loop.

The database is relatively small but a couple of years data ( running the Pi on an SSD)

Joulinar, 2021-10-02:
Is it running manually started or as service? Did you tried a reboot test to ensure all running fine?

MichaIng, 2021-10-02:
Okay, so the only explanation then indeed is the timeout. Nasty is that the script of the package does not match any version of the script on GitHub, so I cannot send a PR: https://github.com/influxdata/influxdb/blob/master/scripts/influxd-systemd-start.sh

This never had a timeout but an unlimited loop, and the latest version contains meaningful logs while waiting. I suggest using this script to check how long startup actually takes. Based on this we can open an issue to suggest raising the timeout, or add a setting for it via `/etc/default/influxdb` to point to on timeout:
```sh
curl -sSfL 'https://raw.githubusercontent.com/influxdata/influxdb/master/scripts/influxd-systemd-start.sh' -o /usr/lib/influxdb/scripts/influxd-systemd-start.sh
# Abort the manually started InfluxDB instance
systemct restart influxdb
# Then e.g. wait 20 seconds and check the logs which should tell you how long it took
sleep 20
journalctl -u influxdb
```

Ianszh, 2021-10-02:
Well even though I'm getting data on my Grafana, dietpi-services status says:
`INFO ] DietPi-Services | influxdb		inactive (dead) since Sat 2021-10-02 17:22:17 CEST; 11min ago`

> Is it running manually started or as service? Did you tried a reboot test to ensure all running fine?

dietpi-services status did say it was dead but I received data in Grafana, after a reboot back to same as before.
It's normally running as a service

MichaIng, 2021-10-02:
It was the foreground instance you started which served for Grafana, if you didn't abort it (`CTRL+C`). This was only intended as a test whether it generally is able to function. Apply the above to get the actual systemd service to work.

Joulinar, 2021-10-02:
It was working because you started database manually. And the service was dead because we stopped it before. Pls follow the suggestion done by @MichaIng to check how long the database takes to startup.

Ianszh, 2021-10-02:
`root@DietPi:~# systemct restart influxdb
-bash: systemct: command not found`

I'm afraid I have to do something, do you mind if I get back to you both sometime tomorrow?

MichaIng, 2021-10-02:
> do you mind if I get back to you both sometime tomorrow?

Of course.

> systemct: command not found

Whoops, there was an 'l' missing (fixed above):
```sh
systemctl restart influxdb
```

Ianszh, 2021-10-02:
I should have seen that 🙄

Thanks for your help and see you

Ianszh, 2021-10-03:
Hello,
Updating 7.4.2 -> 7.6.2 is what caused this problem

Ianszh, 2021-10-03:
`root@DietPi:~# systemctl restart influxdb
Job for influxdb.service failed because the control process exited with error code.
See "systemctl status influxdb.service" and "journalctl -xe" for details.`

Ianszh, 2021-10-03:
`-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sun 2021-10-03 16:50:05 CEST. --
Nov 03 18:16:46 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 03 16:43:42 DietPi influxd-systemd-start.sh[485]: unknown command "print-config"
Oct 03 16:43:42 DietPi influxd-systemd-start.sh[485]: Run 'influxd help' for usage
Oct 03 16:43:42 DietPi systemd[1]: influxdb.service: Control process exited, code=exited status=1
Oct 03 16:43:42 DietPi systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Oct 03 16:43:42 DietPi systemd[1]: influxdb.service: Unit entered failed state.
Oct 03 16:43:42 DietPi systemd[1]: influxdb.service: Failed with result 'exit-code'.
Oct 03 16:43:42 DietPi systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Oct 03 16:43:42 DietPi systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Oct 03 16:43:42 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 03 16:43:42 DietPi influxd-systemd-start.sh[539]: unknown command "print-config"
Oct 03 16:43:42 DietPi influxd-systemd-start.sh[539]: Run 'influxd help' for usage
Oct 03 16:43:42 DietPi systemd[1]: influxdb.service: Control process exited, code=exited status=1
`

Ianszh, 2021-10-03:
is there a way to roll back the InfluxDb update?

Joulinar, 2021-10-03:
Did you created a backup before?

Ianszh, 2021-10-03:
Yes I did create a backup

Joulinar, 2021-10-03:
In this case you could simply restore.

Ianszh, 2021-10-03:
Ok yes I will, but I've also updated another RPi with the same problem after updating, but that dosen't have a backup

MichaIng, 2021-10-03:
Let's try to fix this forward. It will affect other users as well, so keeping an older InfluxDB version is not a good solution. Also any `apt upgrade` will break it the same way. It has nothing to do with the DietPi version but simply this new buggy startup script introduced by InfluxDB.

MichaIng, 2021-10-03:
Okay yes sorry the script linked above is for InfluxDB v2 while we use v1.8. Please do the following:
```sh
curl -sSfL 'https://raw.githubusercontent.com/influxdata/influxdb/1.8/scripts/influxd-systemd-start.sh' -o /usr/lib/influxdb/scripts/influxd-systemd-start.sh
# Abort the manually started InfluxDB instance
systemctl restart influxdb
# Then e.g. wait 20 seconds and check the logs which should tell you how long it took
sleep 20
journalctl -u influxdb
```

Ianszh, 2021-10-03:
still the same I'm afraid

`-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sun 2021-10-03 17:17:21 CEST. --
Nov 03 18:16:46 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 03 16:43:43 DietPi influxd-systemd-start.sh[476]: unknown command "print-config"
Oct 03 16:43:43 DietPi influxd-systemd-start.sh[476]: Run 'influxd help' for usage
Oct 03 16:43:43 DietPi systemd[1]: influxdb.service: Control process exited, code=exited status=1
Oct 03 16:43:43 DietPi systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Oct 03 16:43:43 DietPi systemd[1]: influxdb.service: Unit entered failed state.
Oct 03 16:43:43 DietPi systemd[1]: influxdb.service: Failed with result 'exit-code'.
Oct 03 16:57:11 DietPi systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Oct 03 16:57:11 DietPi systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Oct 03 16:57:11 DietPi systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Oct 03 16:57:11 DietPi influxd-systemd-start.sh[531]: unknown command "print-config"
Oct 03 16:57:11 DietPi influxd-systemd-start.sh[531]: Run 'influxd help' for usage
Oct 03 16:57:11 DietPi systemd[1]: influxdb.service: Control process exited, code=exited status=1`

Joulinar, 2021-10-03:
@MichaIng 

Just a side note not. Today I installed Influx for some testing purposes to get a monitoring database. And it's working without issues on my RPi4B ARMv8 Bullseye.

Ianszh, 2021-10-03:
now it's running

`Oct 03 17:16:28 DietPi influxd-systemd-start.sh[1833]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 03 17:16:28 DietPi influxd-systemd-start.sh[1833]: Merging with configuration at: /etc/influxdb/influxdb.conf
Oct 03 17:16:28 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 1 attempts...
Oct 03 17:16:29 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 2 attempts...
Oct 03 17:16:30 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 3 attempts...
Oct 03 17:16:31 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 4 attempts...
Oct 03 17:16:32 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 5 attempts...
Oct 03 17:16:33 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 6 attempts...
Oct 03 17:16:34 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 7 attempts...
Oct 03 17:16:35 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 8 attempts...
Oct 03 17:16:36 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 9 attempts...
Oct 03 17:16:37 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 10 attempts...
Oct 03 17:16:38 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 11 attempts...
Oct 03 17:16:39 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 12 attempts...
Oct 03 17:16:40 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 13 attempts...
Oct 03 17:16:42 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 14 attempts...
Oct 03 17:16:43 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 15 attempts...
Oct 03 17:16:44 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 16 attempts...
Oct 03 17:16:45 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 17 attempts...
Oct 03 17:16:46 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 18 attempts...
Oct 03 17:16:47 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 19 attempts...
Oct 03 17:16:48 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 20 attempts...
Oct 03 17:16:49 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 21 attempts...
Oct 03 17:16:50 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 22 attempts...
Oct 03 17:16:51 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 23 attempts...
Oct 03 17:16:52 DietPi influxd-systemd-start.sh[1833]: InfluxDB API unavailable after 24 attempts...
Oct 03 17:16:53 DietPi influxd-systemd-start.sh[1833]: [httpd] 192.168.0.3 - root [03/Oct/2021:17:16:53 +0200] "POST /write?db=SolarPo
wer&p=%5BREDACTED%5D&precision=n&rp=&u=root HTTP/1.1 " 204 0 "-" "-" f077b7bc-245c-11ec-8001-b827eb29dece 10114
Oct 03 17:16:53 DietPi influxd-systemd-start.sh[1833]: [httpd] 192.168.0.3 - root [03/Oct/2021:17:16:53 +0200] "POST /write?db=SolarPo
wer&p=%5BREDACTED%5D&precision=n&rp=&u=root HTTP/1.1 " 204 0 "-" "-" f084b401-245c-11ec-8002-b827eb29dece 2463
Oct 03 17:16:53 DietPi influxd-systemd-start.sh[1833]: [httpd] 192.168.0.3 - root [03/Oct/2021:17:16:53 +0200] "POST /write?db=SolarPo
wer&p=%5BREDACTED%5D&precision=n&rp=&u=root HTTP/1.1 " 204 0 "-" "-" f093bd1e-245c-11ec-8003-b827eb29dece 2222
Oct 03 17:16:53 DietPi influxd-systemd-start.sh[1833]: [httpd] 192.168.0.3 - root [03/Oct/2021:17:16:53 +0200] "POST /write?db=SolarPo
wer&p=%5BREDACTED%5D&precision=n&rp=&u=root HTTP/1.1 " 204 0 "-" "-" f0a38588-245c-11ec-8004-b827eb29dece 2267
Oct 03 17:16:53 DietPi influxd-systemd-start.sh[1833]: [httpd] ::1 - - [03/Oct/2021:17:16:53 +0200] "GET /health HTTP/1.1" 200 106 "-"
 "curl/7.52.1" f0b2cdac-245c-11ec-8005-b827eb29dece 330
Oct 03 17:16:53 DietPi systemd[1]: Started InfluxDB is an open-source, distributed, time series database.`

MichaIng, 2021-10-03:
Okay, great 🥳. So it took 24 seconds while the new InfluxDB wrapper fails after a hardcoded 10 seconds. I'm afraid we'll see a lot more of such reports. I'll immediately report this to InfluxDB guys to have a package upgrade shipped ASAP.

MichaIng, 2021-10-03:
Issue was reported already here: https://github.com/influxdata/influxdb/issues/22576
I contributed to the discussion. I hope they'll go for the native systemd solution to have the startup delayed until the network listener has been established.