MichaIng/DietPi #4795: Trying to update 7.4.2 -> 7.6.2

breadman100, 2021-10-03:
│
                                                          │ Err:1 https://packages.sury.org/php stretch InRelease                                                                │
                                                          │   The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key               │
                                                          │ <deb@sury.org>                                                                                                       │
                                                          │ Ign:6 https://deb.debian.org/debian-security stretch/updates InRelease                                               │
                                                          │ Ign:7 https://deb.debian.org/debian stretch-backports InRelease                                                      │
                                                          │ Err:8 https://deb.debian.org/debian stretch Release                                                                  │
                                                          │   server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none                   │
                                                          │ Err:9 https://deb.debian.org/debian stretch-updates Release                                                          │
                                                          │   server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none                   │
                                                          │ Err:10 https://deb.debian.org/debian-security stretch/updates Release                                                │
                                                          │   server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none                   │
                                                          │ Err:11 https://deb.debian.org/debian stretch-backports Release                                                       │
                                                          │   server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none                   │
                                                          │ Reading package lists...                                                                                             │
                                                          │ W: An error occurred during the signature verification. The repository is not updated and the previous index files   │
                                                          │ will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid:     │
                                                          │ EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>                                         │
                                                          │ E: The repository 'https://deb.debian.org/debian stretch Release' does no longer have a Release file.                │
                                                          │ E: The repository 'https://deb.debian.org/debian stretch-updates Release' does no longer have a Release file.        │
                                                          │ E: The repository 'https://deb.debian.org/debian-security stretch/updates Release' does no longer have a Release     │
                                                          │ file.

Joulinar, 2021-10-03:
Basically you have 2 issue.

1st one is the invalid certificate #4794 and the 2nd is the expired PGP key for the PHP repository #4219 

```
apt install ca-certificates
apt-key del '95BD4743'
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

And a side node not related to the issue. You are running oldold Debian Stretch version. You migh consider to upgrade to Buster https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

Joulinar, 2021-10-03:
@MichaIng 
there is one more with `server certificate verification` issue https://dietpi.com/phpbb/viewtopic.php?f=11&t=9504
All Stretch. Not sure if we could do something on prepatch for next release.

MichaIng, 2021-10-03:
The old Let's Encrypt intermediate certificate expired, which was based on a 3rd party root. Based on all information I read this should however not affect a Debian Stretch system, but only VERY old systems. The `deb.debian.org` CDN certificate is not Let's Encrypt but based on the same R3 intermediate hence same changed root certificate, so it is affected by the same issue.

I cannot replicate the issue here, so I guess it is affecting Stretch systems which did not upgrade `ca-certificates` for a long time? Although DietPi v7.4 should be new enough.

Resolving the issue is a bootstrap issue: The APT lists cannot be updated, so the `ca-certificates` package cannot be upgraded, so the list cannot be updated... I'm not keen to download this essential package automatically via plain HTTP, so probably we find a working mirror. This should fail for the same reason, but let's see:
```sh
cd /tmp
curl -sSfLO 'https://deb.debian.org/debian-security/pool/main/c/ca-certificates/ca-certificates_20200601~deb9u2_all.deb'
dpkg -i ca-certificates_20200601~deb9u2_all.deb
rm ca-certificates_20200601~deb9u2_all.deb
```

breadman100, 2021-10-03:
> **EDIT:** Before applying the below steps manually, can you please run the following and paste the output here. It should do exactly the same and so we need to check why it doesn't work as it should:
> 
> ```shell
> bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"
> ```
> 
> Basically you have 2 issue.
> 
> 1st one is the invalid certificate #4794 and the 2nd is the expired PGP key for the PHP repository #4219
> 
> ```shell
> apt install ca-certificates
> apt-key del '95BD4743'
> curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
> /boot/dietpi/func/dietpi-set_software apt-cache clean
> apt update
> apt upgrade
> ```
> 
> And a side node not related to the issue. You are running oldold Debian Stretch version. You migh consider to upgrade to Buster https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

> ```shell
> bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"
> ```

Yes did the same after the above.

MichaIng, 2021-10-03:
> Yes did the same after the above.

And did it fix the "The following signatures were invalid" error when running:
```sh
apt update
```

breadman100, 2021-10-03:
> > Yes did the same after the above.
> 
> And did it fix the "The following signatures were invalid" error when running:
> 
> ```shell
> apt update
> ```

No I had the same issue, still received The following signatures were invalid.

I was able to fix by doing this following:

nano /etc/apt/sources.list
replace https with http
save and exit the file
apt update
apt upgrade
nano /etc/apt/sources.list
replace http with https
save and exit the file

Then I was able to upgrade to latest dietpi and then upgraded to Buster without issue.

Joulinar, 2021-10-03:
I would recommend to change back to HTTPS and verify `apt update` still working

MichaIng, 2021-10-03:
This solves the "server certificate verification failed" errors, not the "The following signatures were invalid" one. Okay than the second was indeed resolved. I just wonder why it didn't work in the first place since that script is executed on every DietPi update.

There were two separate issues:
1. Ondrej's PHP repository key expired some months ago. Our `pre-patches` script should have solved it, as it runs on every DietPi update. So I'm puzzled why this error still shows up for some users, especially since the exact same script now did fix it for you (as long as it was not the other commands, if you ran them manually). However, this has nothing to do with HTTPS but is about the signature of the APT repository.
2. The second issue is about the webservers HTTPS certificate. As mentioned just the last days an intermediate expired which was expected to cause issues with (very) old clients. Also here I'm puzzled how a Debian Stretch on DietPi v7.4, hence with the latest `ca-certificates` installed (that is from `2021-03-13`, hence from before DietPi v7.4 release) can have this issue.

Using plain HTTP is not a great solution, as it meas unauthenticated and unencrypted connection and data transfer. In case of APT, the authentication is not that relevant, as it has its own authentication mechanism (which is responsible for the first issue). But doing any remove HTTP connection unauthenticated and unencrypted is somehow completely nonsense and shouldn't ever be required in 2021, sadly there are obviously some edge cases...

So yes, please test if you can now revert to HTTPS. And can you please check/paste the DietPi update log to see which packages have been upgraded (especially whether `ca-certificates` was included or now):
```sh
cat /var/tmp/dietpi/logs/dietpi-update.log
```

breadman100, 2021-10-03:
> I would recommend to change back to HTTPS and verify `apt update` still working

Yes I did indicate I switched it http to https back in my response above already.

Here is my dietpi-update.log

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Ign:1 http://deb.debian.org/debian stretch InRelease
Hit:2 http://deb.debian.org/debian stretch-updates InRelease
Hit:3 http://deb.debian.org/debian-security stretch/updates InRelease
Hit:4 http://deb.debian.org/debian stretch-backports InRelease
Hit:5 http://deb.debian.org/debian stretch Release
Hit:6 https://packages.sury.org/php stretch InRelease
Hit:7 https://dietpi.com/meveric all InRelease
Hit:8 https://dietpi.com/meveric stretch InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
[  OK  ] DietPi-Update | APT upgrade

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Installing new DietPi code

[  OK  ] DietPi-Update | Downloading update archive
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Installing new DietPi scripts
[  OK  ] DietPi-Update | Installing new DietPi system files
[  OK  ] DietPi-Update | Setting execute permissions for DietPi scripts
[ SUB1 ] DietPi-Set_software > verify_dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] verify_dietpi.txt  | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches

[ INFO ] DietPi-Update | Current version : v7.4.2
[ INFO ] DietPi-Update | Latest version  : v7.6.2
[ INFO ] DietPi-Patch | Patching to DietPi v7.5...
[  OK  ] DietPi-Patch | Patched to DietPi v7.5
[ INFO ] DietPi-Patch | Patching to DietPi v7.6...
[  OK  ] DietPi-Patch | Patched to DietPi v7.6
[ INFO ] DietPi-Update | APT autopurge, please wait...
[  OK  ] DietPi-Update | APT autopurge
[  OK  ] DietPi-Update | Incremental patching to v7.6.2 completed
[ INFO ] DietPi-Update | Checking for available live patches
[ INFO ] DietPi-Update | Found valid live patch 0:
 - Description: dummy
 - Condition: false
 - Patch: :
[ INFO ] DietPi-Update | Live patch 0 is not applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[0]='not applicable' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Found valid live patch 1:
 - Description: Fixes a bug in DietPi-LetsEncrypt which makes it fail when ownCloud or Nextcloud are installed: https://github.com/MichaIng/DietPi/issues/4752
 - Condition: grep -Eq "s/'http://localhost/(own|next)cloud'/'https://\\$primary_domain/(own|next)cloud'/" /boot/dietpi/dietpi-letsencrypt
 - Patch: sed -Ei "s#s/'http://localhost/(own|next)cloud'/'https://\\$primary_domain/(own|next)cloud'/#s|'http://localhost/\1cloud'|'https://\$primary_domain/\1cloud'|#" /boot/dietpi/dietpi-letsencrypt
[ INFO ] DietPi-Update | Live patch 1 is applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[1]='not applied' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Found valid live patch 2:
 - Description: Fixes a bug in DietPi-Software where the Buster => Bullseye upgrade, following our guide, fails if HTTPS was enabled via DietPi-LetsEncrypt before: https://dietpi.com/phpbb/viewtopic.php?t=9477
 - Condition: grep -q "lighty-enable-mod openssl" /boot/dietpi/dietpi-software
 - Patch: sed -i "s/lighty-enable-mod openssl/lighty-enable-mod ssl/" /boot/dietpi/dietpi-software
[ INFO ] DietPi-Update | Live patch 2 is applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[2]='not applied' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Found valid live patch 3:
 - Description: Fixes a bug in DietPi-Software where Kodi fails on RPi ARMv8/64-bit systems when installed without a desktop: https://dietpi.com/phpbb/viewtopic.php?p=38079#p38079
 - Condition: grep -q "\$G_HW_MODEL -le 9 || \$G_HW_MODEL == 1\[56\]" /boot/dietpi/dietpi-software
 - Patch: sed -i "s/\$G_HW_MODEL -le 9 || \$G_HW_MODEL == 1\[56\]/\$G_HW_MODEL -le 9 \&\& \$(dpkg --print-architecture) == 'armhf' || \$G_HW_MODEL == 1\[56\]/" /boot/dietpi/dietpi-software
[ INFO ] DietPi-Update | Live patch 3 is applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[3]='not applied' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Found valid live patch 4:
 - Description: Fixes a bug in DietPi-Software where the Deluge web interface fails to start on Bullseye: https://github.com/MichaIng/DietPi/issues/4785
 - Condition: (( $G_DISTRO > 5 )) && ! grep -E "deluge-web)? -d" /boot/dietpi/dietpi-software
 - Patch: sed -i "s/deluge-web) -l/deluge-web) -d -l/" /boot/dietpi/dietpi-software
[ INFO ] DietPi-Update | Live patch 4 is not applicable on your system
[  OK  ] DietPi-Update | Added setting G_LIVE_PATCH_STATUS[4]='not applicable' to end of file /boot/dietpi/.version
[ INFO ] DietPi-Update | Applying live patch 1
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[1]='applied'
[ INFO ] DietPi-Update | Applying live patch 2
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[2]='applied'
[ INFO ] DietPi-Update | Applying live patch 3
[  OK  ] DietPi-Update | Setting in /boot/dietpi/.version adjusted: G_LIVE_PATCH_STATUS[3]='applied'

breadman100, 2021-10-03:
Yes apt update and apt upgrade work fine with HTTPS now.

root@DietPi7:~# apt update
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://dietpi.com/meveric all InRelease
Hit:5 https://dietpi.com/meveric bullseye InRelease
Hit:6 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
root@DietPi7:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi7:~#

MichaIng, 2021-10-03:
Mysterious as not a single package was upgraded. Did you manually run [these](https://github.com/MichaIng/DietPi/issues/4795#issuecomment-932942484) steps before?

breadman100, 2021-10-03:
> Mysterious as not a single package was upgraded. Did you manually run [these](https://github.com/MichaIng/DietPi/issues/4795#issuecomment-932942484) steps before?

Yes sorry for the confusion I did the run the steps you recommended first, but apt update failed still until I did this:

nano /etc/apt/sources.list
replace https with http
save and exit the file
apt update
apt upgrade
nano /etc/apt/sources.list
replace http with https
save and exit the file

MichaIng, 2021-10-03:
Hmm, so updating the lists once with HTTP solved it without any package upgrade... Like if the list files themselves (in `/var/lib/dpkg/lists` resp. `/tmp/apt/lists` when moved to RAM) contain something which causes this 🤔. Probably `/boot/dietpi/func/dietpi-set_software apt-cache clean` is then a solution. We have a second report, where this can be tested: #4796

breadman100, 2021-10-03:
I have 6 dietpi systems that were doing all the same thing..

Err:8 https://deb.debian.org/debian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:9 https://deb.debian.org/debian stretch-updates Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:10 https://deb.debian.org/debian-security stretch/updates Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:11 https://deb.debian.org/debian stretch-backports Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists...

I just verified doing just this fixes the issue:

nano /etc/apt/sources.list
replace https with http
save and exit the file
apt update
apt upgrade
nano /etc/apt/sources.list
replace http with https
save and exit the file
dietpi-update

Joulinar, 2021-10-03:
@MichaIng 
Packages are `upgrade` using HTTP. That's why non where upgraded during `dietpi-update`

MichaIng, 2021-10-03:
See the other issue here: https://github.com/MichaIng/DietPi/issues/4796#issuecomment-932986941
Only updating the list files once via HTTP and then reverting back to HTTPS without any package upgrades works as well. I have one affected system as well and trying to figure out where `apt update` stores whichever info which fixes it and aside of cache and list files (which are not the culprit).

MichaIng, 2021-10-03:
On the affected system:
```
# openssl s_client -connect deb.debian.org:443
CONNECTED(00000003)
depth=2 OU = GlobalSign Root CA - R3, O = GlobalSign, CN = GlobalSign
verify return:1
depth=1 C = BE, O = GlobalSign nv-sa, CN = GlobalSign Atlas R3 DV TLS CA 2020
verify return:1
depth=0 CN = j.sni-644-default.ssl.fastly.net
verify return:1
---
Certificate chain
 0 s:/CN=j.sni-644-default.ssl.fastly.net
   i:/C=BE/O=GlobalSign nv-sa/CN=GlobalSign Atlas R3 DV TLS CA 2020
 1 s:/C=BE/O=GlobalSign nv-sa/CN=GlobalSign Atlas R3 DV TLS CA 2020
   i:/OU=GlobalSign Root CA - R3/O=GlobalSign/CN=GlobalSign
---
```
On non affected system:
```
# openssl s_client -connect deb.debian.org:443
CONNECTED(00000003)
depth=2 C = US, O = Internet Security Research Group, CN = ISRG Root X1
verify return:1
depth=1 C = US, O = Let's Encrypt, CN = R3
verify return:1
depth=0 CN = cdn-fastly.deb.debian.org
verify return:1
---
Certificate chain
 0 s:CN = cdn-fastly.deb.debian.org
   i:C = US, O = Let's Encrypt, CN = R3
 1 s:C = US, O = Let's Encrypt, CN = R3
   i:C = US, O = Internet Security Research Group, CN = ISRG Root X1
 2 s:C = US, O = Internet Security Research Group, CN = ISRG Root X1
   i:O = Digital Signature Trust Co., CN = DST Root CA X3
---
```
Below is what also browsers show with the expected new `ISRG Root X1` while above is a wrong certificate received. How is that possible? Like the affected systems have certificates for hosts cached somewhere and do not try to actually receive the new one?

MichaIng, 2021-10-03:
Searching through the web for such issues leads to the assumption that it is not an issue with the client but with the server. But how is it possible that a Stretch container shows the wrong certificate while the host and all other containers (so far) show the right one. And I cannot imagine that is has something to do with the connection method (TLSv1.2 vs TLSv1.3). At least forcing an older TLS version with `curl` doesn't break it.

ravenclaw900, 2021-10-03:
I read about this earlier, I think it was a bug with an old version of OpenSSL (1.0.2), which stopped immediately on the first certificate, instead of going along the chain looking for newer ones.

MichaIng, 2021-10-03:
The same issue however was reported with a Buster system as well 🤔. I checked ssllabs and indeed the old certificate path seems to be still shipped as alternative: https://www.ssllabs.com/ssltest/analyze.html?d=deb.debian.org&s=2a04%3a4e42%3a6a%3a0%3a0%3a0%3a0%3a644

So it seems to be indeed a bug with the Debian CDN then, but how can we force OpenSSL to use the correct new path and why does connecting to the same host via plain HTTP once solve the issue 🤔.

MichaIng, 2021-10-03:
Ah okay, switching back to HTTPS at least in my case breaks it again. So that mystery is solved 😄.

MichaIng, 2021-10-03:
Ahh, there is a `libssl` upgrade available, which solves the issue. Okay, so following simple solution:
1. We switch to HTTP via pre-patches
2. We switch back to HTTPS via (post) patches

MichaIng, 2021-10-06:
Issue solved for `deb.debian.org`, the Debian System Administrators reacted to my mail 👍🙂.