MichaIng/DietPi #4796: Odroid C2 - Dietpi-Image FromScratch as always, but not updating and aborting

bbbxl68, 2021-10-03:
#### Details:
- Date           | Sun Oct  3 16:21:46 BST 2021
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Ign:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://dietpi.com/meveric all InRelease
Ign:3 https://deb.debian.org/debian buster-updates InRelease
Hit:4 https://dietpi.com/meveric buster InRelease
Ign:5 https://deb.debian.org/debian-security buster/updates InRelease
Ign:6 https://deb.debian.org/debian buster-backports InRelease
Err:7 https://deb.debian.org/debian buster Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 151.101.14.132 443]
Err:8 https://deb.debian.org/debian buster-updates Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 151.101.14.132 443]
Err:9 https://deb.debian.org/debian-security buster/updates Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 151.101.14.132 443]
Err:10 https://deb.debian.org/debian buster-backports Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 151.101.14.132 443]
Reading package lists...
E: The repository 'https://deb.debian.org/debian buster Release' does not have a Release file.
E: The repository 'https://deb.debian.org/debian buster-updates Release' does not have a Release file.
E: The repository 'https://deb.debian.org/debian-security buster/updates Release' does not have a Release file.
E: The repository 'https://deb.debian.org/debian buster-backports Release' does not have a Release file.
```

MichaIng, 2021-10-03:
Many thanks for your report. It duplicates with this one: #4795

Interesting that it can occur even on Buster. Can you test whether this solves it:
```sh
cd /tmp
curl -sSfLO 'https://deb.debian.org/debian/pool/main/c/ca-certificates/ca-certificates_20200601~deb10u2_all.deb'
dpkg -i ca-certificates_20200601~deb10u2_all.deb
rm ca-certificates_20200601~deb10u2_all.deb
```

bbbxl68, 2021-10-03:
After applying your Lines, exact same behaviour.

#### Details:
- Date           | Sun Oct  3 17:07:19 BST 2021
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Ign:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://dietpi.com/meveric all InRelease
Ign:3 https://deb.debian.org/debian buster-updates InRelease
Hit:4 https://dietpi.com/meveric buster InRelease
Ign:5 https://deb.debian.org/debian-security buster/updates InRelease
Ign:6 https://deb.debian.org/debian buster-backports InRelease
Err:7 https://deb.debian.org/debian buster Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 151.101.14.132 443]
Err:8 https://deb.debian.org/debian buster-updates Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 151.101.14.132 443]
Err:9 https://deb.debian.org/debian-security buster/updates Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 151.101.14.132 443]
Err:10 https://deb.debian.org/debian buster-backports Release
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 151.101.14.132 443]
Reading package lists...
E: The repository 'https://deb.debian.org/debian buster Release' does not have a Release file.
E: The repository 'https://deb.debian.org/debian buster-updates Release' does not have a Release file.
E: The repository 'https://deb.debian.org/debian-security buster/updates Release' does not have a Release file.
E: The repository 'https://deb.debian.org/debian buster-backports Release' does not have a Release file.
```

bbbxl68, 2021-10-03:
Forgotten to say "Thank you for your Help" although it´s not working yet.

MichaIng, 2021-10-03:
Can you try the following:
```sh
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

bbbxl68, 2021-10-03:
Unfortunately the same Error....

MichaIng, 2021-10-03:
Does the following help:
```sh
update-ca-certificates --fresh
apt update
```
Else:
```sh
sed -i 's/https:/http:/' /etc/apt/sources.list
apt update
sed -i 's/http:/https:/' /etc/apt/sources.list
apt update
```
This helped in case of the other report, I just don't know why 🤔.

bbbxl68, 2021-10-03:
update-ca-certificates --fresh
apt update

This did nothing. Same Error.

I´m right to do 
 
sed -i 's/https:/http:/' /etc/apt/sources.list
apt update

then do the dietpi-update and after that reverse it with

sed -i 's/http:/https:/' /etc/apt/sources.list
apt update

MichaIng, 2021-10-03:
The DietPi update does nothing relevant in this case aside of updating the APT lists, so probably you can run all four commands and check whether the last one does not show these errors anymore.

Good thing is that I now face the same error on one of my ARM build systems (while on the VM I didn't face it). So far I can replicate that none of the steps until the last four commands help. I'll try to figure out what the issue is and why changing back and forth between HTTP and HTTPS helps.

bbbxl68, 2021-10-03:
Ok i can confirm that does the trick. I´can now set the dietpi passwords and go on from there i hope. Thank you so far.

> sed -i 's/https:/http:/' /etc/apt/sources.list
apt update
sed -i 's/http:/https:/' /etc/apt/sources.list
apt update

MichaIng, 2021-10-03:
Many thanks for verifying. So there seems to be something stored with the list update aside of the APT cache and the list files (which are all removed by the `/boot/dietpi/func/dietpi-set_software apt-cache clean` command). I'll try to figure out what and where.

Joulinar, 2021-10-05:
One more https://dietpi.com/phpbb/viewtopic.php?p=38488#p38488

MichaIng, 2021-10-05:
I reported the issue to Debian now as well in hope that they update their certificate to have the expired path removed.

Joulinar, 2021-10-06:
We have quite some issues open for the certificate validation failing. Not sure which one is the leading on now. There is one more for Webmin on our forum https://dietpi.com/phpbb/viewtopic.php?p=38521#p38521

MichaIng, 2021-10-06:
Issue solved by Debian System Administrators.

closed 2021-10-06 by MichaIng