MichaIng/DietPi #4797: WAN IP no longer displaying upon boot - Raspberry Pi

bransond5225, 2021-10-03:
### Creating a bug report/issue

#### Required Information
- DietPi version:  v7.6.2
- Distro version:  bullseye 1
- Kernel version:  Linux DietPi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux
- SBC model:  RPi 3 Model B (armv7l)
- Power supply used:  Canakit 5V 2.5A
- SDcard used:  Samsung 32 EVO Plus

#### Steps to reproduce
SSH into DietPi OS and an error message appears in the WAN IP field

#### Expected behaviour
WAN IP address should display

#### Actual behaviour
The following error message:  curl: (35) error:1408F10B:SSL routines:ssl3_get_record:wrong version number

MichaIng, 2021-10-03:
Many thanks for your report. `curl` throws an error. Can you try:
```sh
curl -I https://dietpi.com/
```

bransond5225, 2021-10-03:
Output:

***@***.***:/$ curl -I https://dietpi.com/
HTTP/2 200
date: Sun, 03 Oct 2021 17:11:53 GMT
content-type: text/html; charset=utf-8
x-content-type-options: nosniff
x-frame-options: SAMEORIGIN
x-xss-protection: 1; mode=block
x-robots-tag: all
x-download-options: noopen
x-permitted-cross-domain-policies: none
referrer-policy: no-referrer-when-downgrade
expect-ct: enforce, max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
content-security-policy: upgrade-insecure-requests; frame-ancestors 'self'; form-action 'self'; base-uri https://dietpi.com/matomo/index.php https://dietpi.com/grafana/; default-src 'none'; object-src 'none'; style-src 'self' 'unsafe-inline' https://fonts.googleapis.com; font-src 'self' https://fonts.gstatic.com data:; img-src * data:; script-src 'self' 'unsafe-inline' 'unsafe-eval' https://unpkg.com https://google.com/recaptcha/ https://www.google.com/recaptcha/ https://www.gstatic.com/recaptcha/; frame-src 'self' https://www.youtube-nocookie.com https://google.com/recaptcha/ https://www.google.com/recaptcha/; manifest-src 'self'; connect-src 'self' https://api.github.com
permissions-policy: accelerometer=(), autoplay=(), camera=(), document-domain=(), encrypted-media=(), fullscreen=(self "https://www.youtube-nocookie.com"), geolocation=(), gyroscope=(), magnetometer=(), microphone=(), midi=(), payment=(), usb=(), screen-wake-lock=()
cache-control: public, max-age=86399
strict-transport-security: max-age=31536000; includeSubDomains; preload
last-modified: Thu, 30 Sep 2021 18:46:25 GMT
vary: Accept-Encoding
content-language: en
cf-cache-status: DYNAMIC
report-to: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v3?s=MARHXymlPvOdjt8s60Q7PALJUYx%2BjhDp0yhFR65Bb%2F2WC695wdoNyn2NfYK0dSTbir57vaTulRpHEtZ58T2MvWm4vbLykQ1rPL8ugu14iDFlFQna52pbTj3ttWD%2B"}],"group":"cf-nel","max_age":604800}
nel: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
server: cloudflare
cf-ray: 6987cacfca01f35d-ATL
alt-svc: h3=":443"; ma=86400, h3-29=":443"; ma=86400, h3-28=":443"; ma=86400, h3-27=":443"; ma=86400

***@***.***:/$ ***@***.***:/$

It used to display WAN IP but stopped a while back.

Thanks for quick replay.

David

MichaIng, 2021-10-03:
Okay that works. How about:
```sh
curl -sSfL https://freegeoip.app/csv/
```

bransond5225, 2021-10-03:
Had to leave for a bit, back now.

Output:

***@***.***:/$ curl -sSfL https://freegeoip.app/csv/
curl: (35) error:1408F10B:SSL routines:ssl3_get_record:wrong version number
***@***.***:/$ ***@***.***:/$

I tried going to it in a web browser got error:  

David

bransond5225, 2021-10-03:
continuation of last email:

This site can’t provide a secure connection

freegeoip.appsent an invalid response.

* Try running Windows Network Diagnostics.

ERR_SSL_PROTOCOL_ERROR

David

MichaIng, 2021-10-03:
Do you use Pi-hole or another way of DNS filtering?

bransond5225, 2021-10-03:
I use PiHole + Unbound as my DNS resolver/filter and my router as my DHCP server.

When I first set it all up, it worked fine for a long time.   Would pull WAN IP no problem.

David

MichaIng, 2021-10-03:
Can you try to add `freegeoip.app` to the whitelist? Probably a blocklist update now contains it. Sadly such public APIs are often found on blocklists as also tracking scripts by times make use of them. It is however executed client side, so your browser sees its own remote IP and not the ads provider (which has your IP anyway), so the reason for blocking those APIs is questionable.

bransond5225, 2021-10-03:
It is and has been whitelisted for quite a while.  I figured out early on that it was needed for DietPi.  I just now disabled and re-enabled it but that did not solve the issue.

I have even gone as far as to tell my router to use it's own DNS resolution thereby by-passing PiHole.  Also, I have disabled PiHole temporarily and all ad blocking on my system to no avail.

David

MichaIng, 2021-10-03:
Okay than this is not the issue. Can you show the verbose output:
```sh
curl -v https://freegeoip.app/csv/
```

bransond5225, 2021-10-03:
Output:

***@***.***:~$ curl -v https://freegeoip.app/csv/
* Trying 172.67.188.154:443...
* Connected to freegeoip.app (172.67.188.154) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
* CAfile: /etc/ssl/certs/ca-certificates.crt
* CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* error:1408F10B:SSL routines:ssl3_get_record:wrong version number
* Closing connection 0
curl: (35) error:1408F10B:SSL routines:ssl3_get_record:wrong version number
***@***.***:~$ ***@***.***:~$

David

MichaIng, 2021-10-03:
Strange, here it works well:
```
# curl -v https://freegeoip.app/csv/
*   Trying 104.21.19.200:443...
* Connected to freegeoip.app (104.21.19.200) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=California; L=San Francisco; O=Cloudflare, Inc.; CN=sni.cloudflaressl.com
*  start date: Aug 11 00:00:00 2021 GMT
*  expire date: Aug 10 23:59:59 2022 GMT
*  subjectAltName: host "freegeoip.app" matched cert's "freegeoip.app"
*  issuer: C=US; O=Cloudflare, Inc.; CN=Cloudflare Inc ECC CA-3
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* Using Stream ID: 1 (easy handle 0x5574d10e5f80)
> GET /csv/ HTTP/2
> Host: freegeoip.app
> user-agent: curl/7.74.0
> accept: */*
```
The server sits behind Cloudflare and your `172.67.188.154` is a Cloudflare IP as well, so that looks all correct. Are there package (libSSL/OpenSSL) available?
```sh
apt update
apt upgrade
```

bransond5225, 2021-10-03:
***@***.***:~$ sudo apt update
Hit:1 https://archive.raspberrypi.org/debian bullseye InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Fetched 15.0 kB in 2s (8663 B/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
***@***.***:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
***@***.***:~$ ***@***.***:~$

I guess the answer is no :)

David

MichaIng, 2021-10-03:
Probably that Cloudflare node is faulty. Does it work if you try to connect via IPv6 (if enabled on your system and network):
```sh
curl -v6 https://freegeoip.app/csv/
```

bransond5225, 2021-10-03:
No, Output:

***@***.***:~$ curl -v6 https://freegeoip.app/csv/
* Trying 2606:4700:3036::6815:13c8:443...
* Immediate connect fail for 2606:4700:3036::6815:13c8: Network is unreachable
* Trying 2606:4700:3033::ac43:bc9a:443...
* Immediate connect fail for 2606:4700:3033::ac43:bc9a: Network is unreachable
* Closing connection 0
curl: (7) Couldn't connect to server
***@***.***:~$ ***@***.***:~$

I was just reading where fregeoip.app is deprecated and now uses this:  https://freegeoip.live

Also, as stated on that page there is a maximum quota:

You're allowed up to50,000 queries per hour for one IP address. Once this limit is reached, all of your requests will result in HTTP 403, forbidden, until your quota is cleared.

I really don't know much about any of this, so please forgive me if I am wasting your time with this information.  Just trying to help.

David

MichaIng, 2021-10-03:
There seems to be no IPv6 route (probably the router does not provide it):
```sh
ip -6 r
```
This command needs to show a route starting with `default via` followed by the IPv6 LLA address of the router.

> You're allowed up to 50,000 queries per hour for one IP address.

I wonder if this applies to the Cloudflare server IPs or respects the CF headers to apply to the actual client. However, the curl error does not indicate that it's this issue. And of course I doubt that any actual client is doing 50,000 requests in one hour 😄.

Not sure how to solve this issue at the moment, I guess it is a temporary one. If you don't rely on it, let's wait for one day and see if it is still present. I may report this to Cloudflare if it persists.

bransond5225, 2021-10-03:
Output:

***@***.***:~$ ip -6 r
::1 dev lo proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
***@***.***:~$ ***@***.***:~$

I looked at the PiHole query log and confirmed it is indeed replying with an IP address.  I am also able to get a WAN IP from sites such as whatsmyip.org and freegeoip.live.  So it appears to me, it is the curl command format that is in question.

Thanks for trying and your time.  Maybe it will get sorted out in the future.

David

MichaIng, 2021-10-03:
So you don't have an IPv6 route as expected.

> I looked at the PiHole query log and confirmed it is indeed replying with an IP address.

Yes we see that in the outputs above already, also you get an IPv6 address which is independent of IPv6 being functional: You can ask a DNS server for an AAAA record (IPv6) via IPv4 and independent of the fact whether IPv6 is functional at all or not.

>  I am also able to get a WAN IP from sites such as whatsmyip.org and freegeoip.live.

Try to solve this by adding them to the whitelist, respectively check whether Pi-hole is blocking those. Also which upstream DNS are you using? Some have an internal DNS blocking.

> So it appears to me, it is the curl command format that is in question.

It is a completely regular curl command, nothing special, also as you could see above even a simply `curl https://freegeoip.app/csv/` fails the same way, so the flags have nothing to do with the issue,

bransond5225, 2021-10-03:
I added them to the whitelist and confirmed they are not being blocked by pihole, however, did not solve the issue.

My question is why don't I have IPv6 response?  I don't recall ever disabling that anywhere.

Let's call it a day as I need to attend to something else.

Thanks again,

David

MichaIng, 2021-10-03:
Which upstream DNS do you use in Pi-hole? Ah, sorry you said already that you use Unbound. You use Unbound as recursive DNS server or did you enabled DoT or similar?
... ah actually the resolved IPv4 seems correct (a Cloudflare address), so DNS blocking isn't the issue for `freegeoip.app` and probably neither for the other two API hosts. No idea whether some firewall elsewhere may be the issue then, something that blocks on IP level, not on DNS level 🤔.

> My question is why don't I have IPv6 response?  I don't recall ever disabling that anywhere.

It seems it is disabled in your router. Probably you don't even have a public IPv6 address? Or do you manage DHCP via Pi-hole? Your router needs to send router advertisements (RA) which contain the public IPv6 prefix, so clients can attach themselves a global unicast address (GUA) and the routers IPv6 address as gateway. You can check the IPv6 addresses of your system via:
```
ip -6 a
```
I guess there is only one starting with `fe80::`, which is a local link address (LLA) which can only be used to connect to the router itself but not to a remove IPv6 host.

bransond5225, 2021-10-03:
Output:

***@***.***:~$ ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 state UNKNOWN qlen 1000
inet6 ::1/128 scope host
valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
inet6 fe80::eab9:51b9:d5dd:2adc/64 scope link
valid_lft forever preferred_lft forever
***@***.***:~$ ***@***.***:~$

Pihole points to 127.0.0.1#5335 for unbound.

I use my router for DHCP on the local network.

My router is running DD-WRT and I've made no changes to it recently since first setting up pihole.

David

Joulinar, 2021-10-03:
as stated above by @bransond5225 issue is with `freegeoip.app`. The new name `freegeoip.live` is working fine.

```
root@DietPi3:~# curl -sSfLm 100 https://freegeoip.app/csv/ 2>&1 | mawk -F, '($5){r=$5" "}{print $1" "r$3}'
curl: (7) Failed to connect to freegeoip.app port 443: Connection refused
root@DietPi3:~# curl -sSfLm 100 https://freegeoip.live/csv/ 2>&1 | mawk -F, '($5){r=$5" "}{print $1" "r$3}'
93.x.x.x xxx Germany
root@DietPi3:~#
```

MichaIng, 2021-10-03:
> My router is running DD-WRT

Probably you can browse the DD-WRT settings to check whether you got a public IPv6 address by your ISP and whether the router is providing IPv6 addresses to the local network accordingly.

bransond5225, 2021-10-03:
I will take a look at dd-wrt settings tomorrow.

Thank-you again for all your assistance.

David

MichaIng, 2021-10-03:
> issue is with `freegeoip.app`. The new name `freegeoip.live` is working fine.

On all my systemd `freegeoip.app` works fine. Any hint which says that `freegeoip.live` is "new" or superseding the first? To me it looks like a different provider. Also your error is a different one.

Joulinar, 2021-10-03:
sorry need to revert my statement. I had some human error. Old one is working fine. Just located me wrongly in another state. But in general it's working fine.

```
root@DietPi3:~# curl -sSfLm 100 https://freegeoip.app/csv/ 2>&1 | mawk -F, '($5){r=$5" "}{print $1" "r$3}'
93.x.x.x. xxx Germany
root@DietPi3:~#
```

MichaIng, 2021-10-03:
Nice would be to allow choosing a different provider, but all have different response formats when the location is included 🤔. So only possible way would be if users enter the `mawk` as well, which makes is pretty unusable for 95% of our users or so 😄.

MichaIng, 2021-10-04:
Since we are behind Cloudflare, I found a simple way to provide GeoIP info via our own domain via Cloudflare worker. If this get's used by too many clients outside of our scripts, we may add a custom request header or user agent and block all requests which do not match it. But our old myip.php script wasn't overloaded with unintended requests either, so for now it should be fine without any filter.

MichaIng, 2021-10-04:
But let's see if this really solves the issue:
```sh
curl -sSf https://dietpi.com/geoip
```

bransond5225, 2021-10-04:
I don't know if you had already implemented this change earlier or not, but when I logged in to my DietPi system at 6:30 am this morning everything was working again.  

Also, the curl command works fine from my side.

Thank-you for all the assistance.  I really like DietPi OS a lot and I made a financial donation last week to show my appreciation and support.

David

MichaIng, 2021-10-04:
Okay, then our guess that it was a temporary issue, probably with the Cloudflare node, or something on ISP side, turned out to be true 🙂. However, the step to not rely on 3rd party providers/APIs but provide own ones for our uses is reasonable anyway. And it was a constant issue that those public APIs often land on blocklists.

closed 2021-10-04 by bransond5225

Joulinar, 2021-10-04:
@MichaIng 
for me it looks like this using our service now. 

```
root@DietPiProd:~# curl -sSf https://dietpi.com/geoip
93.x.x.x xx DE
```

MichaIng, 2021-10-04:
Jep, same from our server, so a region was not found by the service which Cloudflare internally uses. From my home it shows:
```
xxxx:yyyy:... Land Berlin DE
```
which is not accurate but it matches the result of the old API, despite `DE` instead of `Germany`.

Joulinar, 2021-10-04:
myself are located in S-A which is as well wrong for my state. But yeah same as the old API.

MichaIng, 2021-10-04:
Okay 😄. I remember we used the city in the past, which was wrong even more often. I think the region is accurate in larger countries like the US, where it matches the state, but in Germany IPs seem to be not reliably attached to specific federal states.

ravenclaw900, 2021-10-04:
Yeah, it gets my state correct.