MichaIng/DietPi #4799: MongoDB

Zenthae, 2021-10-04:
### Creating a software request

#### Formal software information
- MongoDB | 
- Good nosql DB | 
- https://www.mongodb.com | 
- GitHub URL (if available) | 
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?

-  no

#### What makes your requested software better than the above solutions, if available?

- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

Only the amd64 version is available on mongo repo, I think it's needed to compile it to arm64

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- 

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

Joulinar, 2021-10-04:
Hi,

I don't think this will happen. MongoDB did not support arm64 platform https://docs.mongodb.com/manual/administration/production-notes/#arm64

MichaIng, 2021-10-04:
Was discussed here: #3708
An ARMv8/arm64 build was requested, but the linked notes seem to indicate that it is a deeper incompatibility, means it cannot be compiled either 🤔.

Zenthae, 2021-10-05:
it would be a sad thing if it can be done because i really want to use DietPI and i don't really want to bother setup a relational DB

Joulinar, 2021-10-05:
This is not a limitation of DietPi. Starting MongoDB 5, they did not support ARM architecture anymore. Only possible way is to us Docker.

Zenthae, 2021-10-05:
could Docker run on a RPI without consuming too much resources and overheating ?

Joulinar, 2021-10-05:
Of course there is some overhead. How much? I don't know. I would simply give it a try. Docker/ Docker Compose are available via our software catalogue.

MichaIng, 2021-10-05:
Hmm, if the Docker container runs on arm64 then obviously the binary does. So as long as they are not cheating and it's an armhf one actually, it seems to compile on arm64. We may have a look into their sources, probably some hack required.

Joulinar, 2021-10-05:
If I'm not mistaken they use Ubuntu for their container. Will have a look and check what is inside.

Joulinar, 2021-10-05:
ok after playing around whole evening on Docker, no chance to get MongoDB v5 running on ARMv8. Simply it's not working on this CPU/architecture. 

```
root@DietPi4:~# docker logs mongodb
/usr/local/bin/docker-entrypoint.sh: line 392:    27 Illegal instruction     "${mongodHackedArgs[@]}" --fork
/usr/local/bin/docker-entrypoint.sh: line 392:    28 Illegal instruction     "${mongodHackedArgs[@]}" --fork
/usr/local/bin/docker-entrypoint.sh: line 392:    27 Illegal instruction     "${mongodHackedArgs[@]}" --fork
/usr/local/bin/docker-entrypoint.sh: line 392:    28 Illegal instruction     "${mongodHackedArgs[@]}" --fork
/usr/local/bin/docker-entrypoint.sh: line 392:    28 Illegal instruction     "${mongodHackedArgs[@]}" --fork
/usr/local/bin/docker-entrypoint.sh: line 392:    30 Illegal instruction     "${mongodHackedArgs[@]}" --fork
```

Latest image I was able to start was `mongo:4.4.9-focal`. This is how it looks like

```
root@DietPi4:~# docker ps -a
CONTAINER ID   IMAGE                    COMMAND                  CREATED          STATUS                        PORTS                                        NAMES
c46993d821db   mongo-express            "tini -- /docker-ent…"   11 minutes ago   Up About a minute (healthy)   0.0.0.0:8081->8081/tcp                       mongo-express
a52588edb6e7   mongo:4.4.9-focal        "docker-entrypoint.s…"   12 minutes ago   Up 2 minutes (healthy)        0.0.0.0:27017->27017/tcp                     mongodb
21a9a41e8ea8   portainer/portainer-ce   "/portainer"             10 hours ago     Up About an hour              8000/tcp, 9443/tcp, 0.0.0.0:9002->9000/tcp   portainer
root@DietPi4:~#
```

Mongo-Express shows all databases
[image]

A small hack was needed to get system data displayed in Mongo-Express https://github.com/mongo-express/mongo-express/issues/743

Resource consumption on MongoDB is around 100MB memory usage
[image]

MichaIng, 2021-10-05:
Wow, even that it is the "official" image, good and sad to know, where ARMv8 is the standard of nearly all new SoCs 😞. Confusing that MongoDB with this step basically abandoned ARM support, while ARM is the growing architecture, becoming increasingly relevant in all areas, server marked, and, well, Apple M1.

Joulinar, 2021-10-05:
Probably they do not believe in SBC's?? In general they support ARMv8 as they offer an image for Mongo v5 but they require ARMv8.2-A or later microarchitecture.

MichaIng, 2021-10-05:
> ARMv8.2-A

Ah true, that was the point.

Joulinar, 2021-10-06:
ok I continued playing with Mongo and I did some relay `stupid` thinks. I just installed MongoDB 4.4 from Mongo Ubuntu repository. What to say, it's working so far.

It's going to download packages from Mongo repository only.

```
root@DietPi4:~# apt install mongodb-org
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  mongodb-database-tools mongodb-org-database-tools-extra mongodb-org-mongos mongodb-org-server mongodb-org-shell mongodb-org-tools
The following NEW packages will be installed:
  mongodb-database-tools mongodb-org mongodb-org-database-tools-extra mongodb-org-mongos mongodb-org-server mongodb-org-shell mongodb-org-tools
0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 90.1 MB of archives.
After this operation, 193 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse arm64 mongodb-database-tools arm64 100.5.0 [44.5 MB]
Get:2 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse arm64 mongodb-org-shell arm64 4.4.9 [12.3 MB]
Get:3 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse arm64 mongodb-org-server arm64 4.4.9 [18.8 MB]
Get:4 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse arm64 mongodb-org-mongos arm64 4.4.9 [14.6 MB]
Get:5 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse arm64 mongodb-org-database-tools-extra arm64 4.4.9 [7752 B]
Get:6 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse arm64 mongodb-org-tools arm64 4.4.9 [2888 B]
Get:7 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4/multiverse arm64 mongodb-org arm64 4.4.9 [3520 B]
Fetched 90.1 MB in 20s (4401 kB/s)
```

Service is active after startup

```
root@DietPi4:~# systemctl status mongod
● mongod.service - MongoDB Database Server
     Loaded: loaded (/lib/systemd/system/mongod.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2021-10-06 23:05:51 CEST; 21s ago
       Docs: https://docs.mongodb.org/manual
   Main PID: 8146 (mongod)
        CPU: 2.026s
     CGroup: /system.slice/mongod.service
             └─8146 /usr/bin/mongod --config /etc/mongod.conf

Oct 06 23:05:51 DietPi4 systemd[1]: Started MongoDB Database Server.
```

MongoDB is `LISTEN` to localhost only

```
root@DietPi4:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      4096       127.0.0.1:27017      0.0.0.0:*    users:(("mongod",pid=8146,fd=11))
tcp   LISTEN 0      1000         0.0.0.0:22         0.0.0.0:*    users:(("dropbear",pid=517,fd=4))
tcp   LISTEN 0      1000            [::]:22            [::]:*    users:(("dropbear",pid=517,fd=5))
```

MongoDB 4.4.9 has been installed

```
root@DietPi4:~# mongod --version
db version v4.4.9
Build Info: {
    "version": "4.4.9",
    "gitVersion": "b4048e19814bfebac717cf5a880076aa69aba481",
    "openSSLVersion": "OpenSSL 1.1.1k  25 Mar 2021",
    "modules": [],
    "allocator": "tcmalloc",
    "environment": {
        "distmod": "ubuntu2004",
        "distarch": "aarch64",
        "target_arch": "aarch64"
    }
}
```

Resource consumption is as well around 100MB memory usage but without docker overhead

```
  0[                                  0.0%] Mem[||||||||||                  113M/7.68G]
  1[                                  0.0%] Swp[                                 0K/0K]
  2[||                                2.0%] Tasks: 14; 1 running
  3[|                                 0.7%] Load average: 0.04 0.07 0.08
Avg[||                                0.7%] Uptime: 01:40:38
    PID USER        RES▽ NI CPU%   TIME+  Command
   8146 mongodb   97316   0  0.0  0:07.73 /usr/bin/mongod --config /etc/mongod.conf
      1 root       9488   0  0.0  0:05.14 /sbin/init
```

tindecken, 2021-12-29:
Hi, I'm also able to install MongoDB v4.4.10 on Dietpi 6.4 bit
My last attempt was not successful and I posted here: https://github.com/MichaIng/DietPi/issues/4532

I create one notion sharing with everyone if someone needs

https://tindecken.notion.site/Install-Mongodb-ec2bf409fdbc46daa5dfa744d46ac3da

[image]