MichaIng/DietPi #4801: dietpi-update 7.4.2 to 7.6.2 failed

Bluehmchen, 2021-10-05:
- Date           | Tue Oct  5 08:25:17 CEST 2021
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux rp3b 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. dietpi-update
.
#### Expected behaviour:

- ...
#### Actual behaviour:

- stop with error.
```
Ign:1 http://apt.tvheadend.org/stable raspbian-stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 http://apt.tvheadend.org/stable raspbian-stretch Release
Get:4 https://packages.sury.org/php stretch InRelease [6839 B]
Ign:5 https://repo.mosquitto.org/debian stretch InRelease
Hit:6 https://archive.raspberrypi.org/debian stretch InRelease
Get:7 https://download.docker.com/linux/raspbian stretch InRelease [31.1 kB]
Err:8 https://repo.mosquitto.org/debian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:4 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Reading package lists...
E: The repository 'https://repo.mosquitto.org/debian stretch Release' does no longer have a Release file.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
```

Joulinar, 2021-10-05:
Basically there are 2 issues and both are known. Could you post following output plus before we go to fix them.

```
bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"
```

Bluehmchen, 2021-10-05:
root@rp3b:~# bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"
root@rp3b:~#

Bluehmchen, 2021-10-05:
root@rp3b:# bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"
root@rp3b:#

Joulinar, 2021-10-05:
Fix for both issues would be

```
sed -i 's/https:/http:/' /etc/apt/sources.list.d/mosquitto.conf
apt-key del '95BD4743'
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
sed -i 's/http:/https:/' /etc/apt/sources.list.d/mosquitto.conf
apt update
```

I hope I catched correct location of `mosquitto.conf` file

And a side node not related to the issue. You are running oldold Debian Stretch version. You migh consider to upgrade to Buster https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

Bluehmchen, 2021-10-05:
Thank you, it worked. However, the file in sources.list.d/ is called dietpi-mosquitto.list and has the following content:
deb https://repo.mosquitto.org/debian/ stretch main
Is that ok?

Joulinar, 2021-10-05:
Yeah I wasn't sure on the correct source file name. As I'm online with my mobile phone only, I couldn't verify it. 

`deb https://repo.mosquitto.org/debian/ stretch main` is fine once update completed.

MichaIng, 2021-10-05:
> root@rp3b:# bash -c "$(curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches)"
root@rp3b:#

So the contained patch does not run 🤔. Can you run the following, please, to help us checking for the possible reason:
```sh
apt-key list '95BD4743'
```

Probably we should use the full fingerprint instead of only those 8 characters.

Bluehmchen, 2021-10-05:
root@rp3b:# htop
root@rp3b:# apt-key list '95BD4743'
pub   rsa3072 2019-03-18 [SC] [expires: 2024-02-16]
      1505 8500 A023 5D97 F5D1  0063 B188 E2B6 95BD 4743
uid           [ unknown] DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
sub   rsa3072 2019-03-18 [E] [expires: 2024-02-16]

root@rp3b:~#

MichaIng, 2021-10-05:
That all looks pretty correct, no idea why the pre-patch was not applied: https://github.com/MichaIng/DietPi/blob/03bc475/.update/pre-patches#L27-L37

It checks for `2021` expiry date (the old key) and then prints a message and does the exact same steps. So `apt-key list '95BD4743' 2> /dev/null | grep -q 'expires: 2021-'` didn't return `true`, but I'm puzzled why as it does not with `2024` expiry date. Also to verify the general logic:
```
2021-10-05 14:00:41 root@VM-Stretch:/tmp# wget https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches
2021-10-05 14:00:44 root@VM-Stretch:/tmp# chmod +x pre-patches
2021-10-05 14:00:48 root@VM-Stretch:/tmp# sed -i 's/2021-/2024-/' pre-patches
2021-10-05 14:01:07 root@VM-Stretch:/tmp# ./pre-patches
[ INFO ] DietPi-Pre-patches | Updating PHP APT repo key: https://github.com/MichaIng/DietPi/issues/4219
[  OK  ] DietPi-Pre-patches | apt-key del 95BD4743
[  OK  ] DietPi-Pre-patches | curl -sSfL https://packages.sury.org/php/apt.gpg -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
[  OK  ] DietPi-Pre-patches | apt-get clean
[  OK  ] DietPi-Pre-patches | rm -Rf /var/lib/apt/lists /tmp/apt/lists
```
@Joulinar 
On next report we need the output of `apt-key list '95BD4743'` before the manual fix is applied.

Joulinar, 2021-10-05:
Ok I will hold back resolution to get some deeper investigation done.

MichaIng, 2021-10-05:
Okay second issue, the Mosquitto repository suffers from the same issue as `deb.debian.org` with the false certification path selected by OpenSSL, fixed with the latest package upgrade? We should report this to Mosquitto guys and ask them to renew/recreate the certificate, since AFAIU both certification paths are still provided by their HTTPS server then, with the one being expired.

MichaIng, 2021-10-05:
Yes it is: https://www.ssllabs.com/ssltest/analyze.html?d=repo.mosquitto.org&s=85.119.83.194&latest
#### Path #1: Trusted
1 | Sent by server | repo.mosquitto.orgFingerprint SHA256: 09c7e86de19ba551a58d5c5aa8d7729958aabe42b37a3b07f2459819a9da4bacPin SHA256: 8T79+AGILqZ5hAfZwlFKYHonbAPmtCfO46a4rAG4I5I=RSA 2048 bits (e 65537) / SHA256withRSA
-- | -- | --
2 | Sent by server | R3Fingerprint SHA256: 67add1166b020ae61b8f5fc96813c04c2aa589960796865572a3c7e737613dfdPin SHA256: jQJTbIh0grw0/1TkHSumWb+Fs0Ggogr621gT3PvPKG0=RSA 2048 bits (e 65537) / SHA256withRSA
3 | In trust store | ISRG Root X1   Self-signedFingerprint SHA256: 96bcec06264976f37460779acf28c5a7cfe8a3c0aae11a8ffcee05c0bddf08c6Pin SHA256: C5+lpZ7tcVwmwQIMcRtPbsQtWLABXhQzejna0wHFr8M=RSA 4096 bits (e 65537) / SHA256withRSA

#### Path #2: Not trusted (invalid certificate [Fingerprint SHA256: 0687260331a72403d909f105e69bcf0d32e1bd2493ffc6d9206d11bcd6770739])
1 | Sent by server | repo.mosquitto.orgFingerprint SHA256: 09c7e86de19ba551a58d5c5aa8d7729958aabe42b37a3b07f2459819a9da4bacPin SHA256: 8T79+AGILqZ5hAfZwlFKYHonbAPmtCfO46a4rAG4I5I=RSA 2048 bits (e 65537) / SHA256withRSA
-- | -- | --
2 | Sent by server | R3Fingerprint SHA256: 67add1166b020ae61b8f5fc96813c04c2aa589960796865572a3c7e737613dfdPin SHA256: jQJTbIh0grw0/1TkHSumWb+Fs0Ggogr621gT3PvPKG0=RSA 2048 bits (e 65537) / SHA256withRSA
3 | Sent by server | ISRG Root X1Fingerprint SHA256: 6d99fb265eb1c5b3744765fcbc648f3cd8e1bffafdc4c2f99b9d47cf7ff1c24fPin SHA256: C5+lpZ7tcVwmwQIMcRtPbsQtWLABXhQzejna0wHFr8M=RSA 4096 bits (e 65537) / SHA256withRSACRL ERROR: HTTP request failed with status code 404: http://crl.identrust.com/DSTROOTCAX3CRL.crl
4 | In trust store | DST Root CA X3   Self-signedFingerprint SHA256: 0687260331a72403d909f105e69bcf0d32e1bd2493ffc6d9206d11bcd6770739Pin SHA256: Vjs8r4z+80wjNcr1YKepWQboSIRi63WsWXhIMN+eWys=RSA 2048 bits (e 65537) / SHA1withRSAValid until: Thu, 30 Sep 2021 14:01:15 UTCEXPIREDWeak or insecure signature, but no impact on root certificate

jneilliii, 2021-12-09:
Is there a manual work-around for this issue? I'm currently stuck on version 7.5.2 and attempting to update to 7.8.2 and get the same error. 

```
Reading package lists...
W: Failed to fetch https://repo.mosquitto.org/debian/dists/stretch/InRelease
Failed to connect to repo.mosquitto.org port 443: Connection refused
W: Some index files failed to download. They have been ignored, or old ones
used instead.
```

I attempted the suggested workaround to no avail:
```
sed -i 's/https:/http:/' /etc/apt/sources.list.d/dietpi-mosquitto.list
apt-key del '95BD4743'
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
sed -i 's/http:/https:/' /etc/apt/sources.list.d/dietpi-mosquitto.list
apt update
```
which gives the same error on that last update command

Joulinar, 2021-12-09:
Pls can you share the entire log message for  the command `apt update`

jneilliii, 2021-12-10:
```
dietpi@DietPi:~$ sudo apt update
Ign:1 https://deb.debian.org/debian stretch InRelease
Hit:2 https://packages.sury.org/php stretch InRelease
Hit:3 https://deb.debian.org/debian stretch-updates InRelease
Ign:4 https://repo.mosquitto.org/debian stretch InRelease
Get:5 https://deb.debian.org/debian-security stretch/updates InRelease [53.0 kB]
Get:6 https://deb.debian.org/debian stretch-backports InRelease [91.8 kB]
Hit:7 https://deb.debian.org/debian stretch Release
Err:8 https://repo.mosquitto.org/debian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists... Done
E: The repository 'https://repo.mosquitto.org/debian stretch Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

jneilliii, 2021-12-10:
If it helps I sent bug reports from the last time I tried as well as just trying again now, with the same error from dietpi-update. 

```
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Fri Dec 10 01:51:27 GMT 2021
- Bug report     | 00742d43-ec40-4aaa-9952-2fa70951452e
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Debian
- Hardware       | Virtual Machine (x86_64) (ID=20)
- Kernel version | `Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.144-3.1 (2019-02-19) x86_64 GNU/Linux`
- Distro         | stretch (ID=4)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:

Hit:1 https://packages.sury.org/php stretch InRelease
Ign:2 https://deb.debian.org/debian stretch InRelease
Hit:3 https://deb.debian.org/debian stretch-updates InRelease
Hit:4 https://deb.debian.org/debian-security stretch/updates InRelease
Ign:5 https://repo.mosquitto.org/debian stretch InRelease
Hit:6 https://deb.debian.org/debian stretch-backports InRelease
Hit:7 https://deb.debian.org/debian stretch Release
Ign:9 https://repo.mosquitto.org/debian stretch Release
Ign:10 https://repo.mosquitto.org/debian stretch/main amd64 Packages
Ign:11 https://repo.mosquitto.org/debian stretch/main all Packages
Ign:12 https://repo.mosquitto.org/debian stretch/main i386 Packages
Ign:10 https://repo.mosquitto.org/debian stretch/main amd64 Packages
Ign:11 https://repo.mosquitto.org/debian stretch/main all Packages
Ign:12 https://repo.mosquitto.org/debian stretch/main i386 Packages
Ign:10 https://repo.mosquitto.org/debian stretch/main amd64 Packages
Ign:11 https://repo.mosquitto.org/debian stretch/main all Packages
Ign:12 https://repo.mosquitto.org/debian stretch/main i386 Packages
Ign:10 https://repo.mosquitto.org/debian stretch/main amd64 Packages
Ign:11 https://repo.mosquitto.org/debian stretch/main all Packages
Ign:12 https://repo.mosquitto.org/debian stretch/main i386 Packages
Ign:10 https://repo.mosquitto.org/debian stretch/main amd64 Packages
Ign:11 https://repo.mosquitto.org/debian stretch/main all Packages
Ign:12 https://repo.mosquitto.org/debian stretch/main i386 Packages
Err:10 https://repo.mosquitto.org/debian stretch/main amd64 Packages
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Ign:11 https://repo.mosquitto.org/debian stretch/main all Packages
Ign:12 https://repo.mosquitto.org/debian stretch/main i386 Packages
Reading package lists...
W: The repository 'https://repo.mosquitto.org/debian stretch Release' does not have a Release file.
E: Failed to fetch https://repo.mosquitto.org/debian/dists/stretch/main/binary-amd64/Packages  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2021-12-10:
Please try the following:

```
apt install ca-certificates
apt update
apt upgrade
```

jneilliii, 2021-12-10:
```
root@DietPi:/home/dietpi# apt install ca-certificates
Reading package lists... Done
Building dependency tree
Reading state information... Done
ca-certificates is already the newest version (20200601~deb9u2).
0 upgraded, 0 newly installed, 0 to remove and 79 not upgraded.
root@DietPi:/home/dietpi# apt update
Ign:1 https://deb.debian.org/debian stretch InRelease
Hit:2 https://deb.debian.org/debian stretch-updates InRelease
Hit:3 https://packages.sury.org/php stretch InRelease
Hit:4 https://deb.debian.org/debian-security stretch/updates InRelease
Ign:5 https://repo.mosquitto.org/debian stretch InRelease
Hit:6 https://deb.debian.org/debian stretch-backports InRelease
Hit:7 https://deb.debian.org/debian stretch Release
Err:9 https://repo.mosquitto.org/debian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists... Done
E: The repository 'https://repo.mosquitto.org/debian stretch Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

jneilliii, 2021-12-10:
after the last apt upgrade it seems to now get through the dietpi-udate, thanks for the help.

Joulinar, 2021-12-10:
I'm wondering how this was fixed now because you still had the same issue while running `apt update` 🤔 

But good if it has been fixed. Btw: you are running oldold Debian Stretch. At least you should think of to upgrade to Buster as we will stop supporting Stretch systems soon and you will not receive further updates.

MichaIng, 2021-12-10:
`apt upgrade` upgraded the LibSSL/OpenSSL packages which solves the issue. Strange only that the pre-patch of `dietpi-update` didn't do it and the first manual repeat didn't do it either: https://github.com/MichaIng/DietPi/issues/4801#issuecomment-989533760

However, good when it's solved now. I'll mark this issue as closed. The three repositories have been informed about the issue, but it is not that trivial to solve as with common Certbot versions Let's Encrypt's ACME servers themselves still provide the invalid certification path for some reason. A workaround is in place, and at least generally it works (`apt upgrade`).

closed 2021-12-10 by MichaIng