MichaIng/DietPi #4802: Multiple UARTs via SC16IS752?

danergo, 2021-10-05:
Folks,

Once I've set this up, but now I can't figure out what the problem is. 

I ran this command to apply DT Overlay: 

```
# ls /dev/ttyS*
# dtoverlay sc16is752-spi0
# ls /dev/ttyS*
/dev/ttySC0  /dev/ttySC1
#
```

So I have two serial ports at least in theory. However no matter if I use `/dev/ttySC0` or `/dev/ttySC1`, in both cases only one channel of SC16IS752 will be used. 

SC16IS752 has two serial ports, A and B, and in both cases above, only serial port A works.

I'm using a ComputeModule1 for this, and a custom board with SC16IS752 (basically a breakout board for SPI functionality).

dmesg confirms everything is fine, could you please help me what could go wrong in this case?

I have used the same custom devices 2 month ago, and now it seems something has gone wrong.

I'm using  v7.4.2.

Thank you!

MichaIng, 2021-10-05:
Try to add `sc16is752-spi1` as well to enable it for SPI1 as well, if this is what you want to achieve. But I'm not sure whether these are attached to `/dev/ttySC0` and `/dev/ttySC1` respectively, or another console device pair. Check out `/boot/overlays/README` for details about the overlays. Else I suggest to ask such technical depth at the Raspberry Pi forum: https://www.raspberrypi.org/forums/
I'm no specialist in this and the amount of DietPi users with a compute module is pretty low. But when it's about the kernel or device tree, luckily DietPi == Raspberry Pi OS 🙂.

danergo, 2021-10-05:
Hm, but why adding `sc16is752-spi1` would help? My sc16is752 is attached to SPI0 with CE0: GPIO8 (CE0), 9 (MISO), 10 (MOSI), 11 (SCK).

sc16is752 is an SPI to dualUART, so it's channelB should be working on the same SPI as ChannelA. :)

(To be more exact I had two of these boards attached to SPI0, and I had 4 parallel UARTS working without any issues)

I'm pretty sure I'm missing something here. :)

danergo, 2021-10-05:
Hold on, I had a wiring issue. MISO and MOSI was interchanged. Very sorry for your time.

closed 2021-10-05 by danergo