MichaIng/DietPi #4803: unable to shutdown - bulleye 1

renaudl, 2021-10-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`   : see below
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN` :  bullseye 1
- Kernel version | `uname -a` : Linux DietPi 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3): RPi 3 Model B+ (armv7l)
- Power supply used | (EG: 5V 1A RAVpower) : standard
- SDcard used | (EG: SanDisk ultra) : scandisk

root@DietPi:~# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
G_LIVE_PATCH_STATUS[3]='applied'
G_LIVE_PATCH_STATUS[4]='applied'

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. install latest distrib
2. let updates and use all default
3. install raspotify shaireplay and audiophonics PI-SPC
4. shutdown command

#### Expected behaviour

- RPI do shutdown

#### Actual behaviour

- get:
Failed to set wall message, ignoring: Unit dbus-org.freedesktop.login1.service is masked.
Failed to call ScheduleShutdown in logind, no action will be taken: Unit dbus-org.freedesktop.login1.service is masked.
- don't shutdown

#### Extra details

- ...

Joulinar, 2021-10-05:
I guess it would be same as this? https://github.com/MichaIng/DietPi/issues/4555#issuecomment-933612730

renaudl, 2021-10-05:
Spotted!

closed 2021-10-05 by renaudl