MichaIng/DietPi #4804: Autorun, where is the best place to put these commands?

danergo, 2021-10-05:
Folks,

I know a lot of possible places to put my commands which I need to run at startup. (rc.local, systemd, init.d, etc)

I need to initialize a custom Bluetooth adapater on boot, preferrably as early as possible to not block any other processes which depends on Bluetooth.

For init I have to run this command:

`hciattach -s 921600 /dev/ttyXY any 921600 flow`

Where is it the best place to put this line in DietPi?

Thank you!

Joulinar, 2021-10-05:
Well you could put it into an own `systemd` service or use `crontab`. There you could specify `@reboot`

MichaIng, 2021-10-05:
Since cron starts at the end of the boot sequence, this will be pretty late, similar to putting it into `/var/lib/dietpi/postboot.d`. A systemd unit is the only way to have it starting ASAP:
```
cat << _EOF_ > /etc/systemd/system/hciattach.service
[Unit]
Description=Attach Bluetooth to UART

[Service]
Type=oneshot
ExecStart=$(command -v hciattach) -s 921600 /dev/ttyS0 any 921600 flow

[Install]
WantedBy=multi-user.target
_EOF_
systemctl enable --now hciattach
```
`Type=oneshot` means that the boot sequence continues when the command has finished. You can skip it, then the boot sequence continues when the command was just called: Other services then cannot reliably sort themselves after this service as they may start when the hciattach command is still processing, but concurrency is better.

MichaIng, 2021-10-06:
The full `cat` command I posted above would have expended `$(command -v hciattach)` to the actual full `hciattach` executable path (since I don't know where it is located), which is the recommended way in systemd units.

Ah yeah that's true, `oneshot` only assure that services which sort themselves after it wait for it to finish, not all services.

It seems the `bluetooth.service` needs to run before UART attachment can be done, which is the one you see as `bluetoothd` in the logs. At least in case of RPi we start that one first (when it's enabled). And also it should be assured that the UART device is ready. Likely this is also the reason for the timeouts and hence long startup time because it takes until `bluetoothd` has finished. Try to add the following to the units `[Unit]` section:
```
After=dev-ttyS0.device bluetooth.service
```

danergo, 2021-10-07:
Got some news:

if I change this: `After=dev-ttyS0.device bluetooth.service` to this: `After=dev-ttyS0.device`

From that point, manually starting (by `systemctl start hciattach.service`) takes the same time just as if I run the `hciattach` manually.

However, the hci interface is not present at all. And also, there are no errors printed into dmesg, so it's weird.

MichaIng, 2021-10-08:
Ahh okay, then this all makes sense 👍.

danergo, 2021-10-10:
Guys, sorry for reopening, but I have upgraded to Bullseye, and now this systemd service is not working anymore.

My service is still this:

```
# cat /etc/systemd/system/hciattach.service
[Unit]
Description=Attach Bluetooth to UART
After=dev-ttyBT.device
Before=bluetooth.service

[Service]
Type=forking
ExecStart=/usr/bin/hciattach -s 921600 /dev/ttyBT any 921600 flow

[Install]
WantedBy=multi-user.target
```

And it runs at startup, but fails:

```
# systemctl status hciuart.service
● hciuart.service - Attach Bluetooth to UART over HCI
     Loaded: loaded (/etc/systemd/system/hciattach.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2021-10-10 20:12:07 BST; 46s ago
    Process: 197 ExecStart=/usr/bin/hciattach -s 921600 /dev/ttyBT any 921600 flow (code=exited, status=1/FAILURE)
        CPU: 21ms

Oct 10 20:12:07 DietPi systemd[1]: Starting Attach Bluetooth to UART over HCI...
Oct 10 20:12:07 DietPi hciattach[197]: Can't open serial port: No such file or directory
Oct 10 20:12:07 DietPi hciattach[197]: Can't initialize device: No such file or directory
Oct 10 20:12:07 DietPi systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Oct 10 20:12:07 DietPi systemd[1]: hciuart.service: Failed with result 'exit-code'.
Oct 10 20:12:07 DietPi systemd[1]: Failed to start Attach Bluetooth to UART over HCI.
```

So it seems my "After=dev-ttyBT.device" directive is not being taken into account for some reason. Do we know about this issue in systemd in bullseye?

Thank you!

danergo, 2021-10-10:
yes.
and when I got chance to type anything I can manually start this systemd service and it succeeds then.

danergo, 2021-10-10:
I had to also define "Requires=[deviceDEFINITION]"

closed 2021-10-10 by danergo