MichaIng/DietPi #4805: No sound on Pine A64

Sciamano72, 2021-10-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
G_LIVE_PATCH_STATUS[3]='applied'
G_LIVE_PATCH_STATUS[4]='not applicable'

- Distro version | buster
- Kernel version | Linux DietPineA64 5.10.60-sunxi64 #21.08.1 SMP Wed Aug 25 18:29:57 UTC 2021 aarch64 GNU/Linux
- SBC model | PINE A64 (aarch64)
- Power supply used | 5V 2A UGreen
- SDcard used | SanDisk ultra 32Gb

#### Additional Information (if applicable)
- Software title | squeezelite
- Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
1. Enable Audio Card (no matter which one) in dietpi-config
2. Install squeezeboxserver + squeezelite
3. command 'squeezelite -l' which shows available cards will respond "null - Discard all samples (playback) or generate zero samples (capture)"
4. playing any audio file from squeezeboxserver (or locally, does not matter) does not result in any hearable output

#### Expected behaviour
- command 'squeezelite -l' should list available sound cards
- audio should be heard from the output

MichaIng, 2021-10-06:
Many thanks for your report. Is the sound card listed by ALSA?
```sh
aplay -l
```

Sciamano72, 2021-10-06:
Nope:

`aplay: device_list:272: no soundcards found...`

MichaIng, 2021-10-07:
So there is simply no sound device found. I checked the current A64 kernel and device tree overlays and there is at least no overlay to enable onboard sound, means that it "should" be enabled by default.

I checked the Armbian forum but couldn't fine something related so far: https://forum.armbian.com/forum/27-allwinner-a64-h5-h6-and-h616/

Does the kernel report any errors that may be related?
```sh
dmesg -l emerg,alert,crit,err
dmesg | grep -E 'audio|sound|snd'
```

Sciamano72, 2021-10-12:
Yes, I've rebooted many times already.

> root@DietPineA64:~# modprobe sunxi_codec
modprobe: FATAL: Module sunxi_codec not found in directory /lib/modules/5.10.60-sunxi64

> root@DietPineA64: # modprobe snd-soc-simple-card
root@DietPineA64:~#
(No output on the last one)

MichaIng, 2021-10-19:
Let me create a Bullseye image, probably the userland updates solve it, so we don't waste time in debugging an old Debian version.

Sciamano72, 2021-10-21:
Trying the new image:

```
root@DietPi:~# aplay - l
^CAborted by signal Interrupt...
aplay: read_header:2839: read error
```

this command hangs.

And after installing LMS+Squeezelite, still no audio output :-(

Sciamano72, 2021-10-21:
My latest post was related to the Bullseye image. Sorry for not being clear about that.

MichaIng, 2021-10-21:
If this is working on Armbian, then the kernel is not the issue. Does playing sound work as well (as you had the same output on the Buster image but no audio), e.g. via:
```sh
speaker-test
```

Sciamano72, 2021-10-21:
No output with ``speaker-test`` either :(

Here is what I see:
```
root@pine64:~# speaker-test

speaker-test 1.2.4

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 256 to 262144
Period size range from 128 to 32768
Using max buffer size 262144
Periods = 4
was set period_size = 32768
was set buffer_size = 262144
 0 - Front Left
Time per period = 5.469834
 0 - Front Left
Time per period = 5.472633
```
But I hear nothing. Hardware problem?

MichaIng, 2021-10-21:
I also checked the Armbian image, and at least there are no additional modules loaded or related module configurations applied. They have a generic sound config script, but it does not "create" a sound card, but only configures the found ones in all possible ways to assure that everything is unmuted and volumes sufficiently high, I guess. At least for the DAC case this could be tried, or then the onboard audio card is listed by `aplay -l`:
<details>
<summary>armbian-audio-config</summary>

```sh
#!/bin/sh

mixer() {
  parm=${4:-on}
  amixer -c "$1" sset "$2" "$3" $parm >/dev/null 2>&1
  amixer -c "$1" sset "$2" $parm >/dev/null 2>&1
}

if [ -f $HOME/.config/sound.conf ]; then

  alsactl restore -f $HOME/.config/sound.conf

else

# get card num
#card=`echo $1 | sed 's/[^0-9]*//g'`
card1=`aplay -l | grep "device 0" | awk '{print $3}'`
#echo $card

for card in $card1 
do

#echo $card

# set common mixer params
  mixer $card Master 0db
  mixer $card Front 100%
  mixer $card PCM 0db
  mixer $card Synth 100%

# mute CD, since using digital audio instead
  mixer $card CD 0% mute

# Only unmute Line and Aux if they are possibly used.
#  mixer $card Line 100%
#  mixer $card Aux 100%

# mute mic
  mixer $card Mic 0% mute

# ESS 1969 chipset has 2 PCM channels
  mixer $card PCM,1 100%

# Trident/YMFPCI/emu10k1
  mixer $card Wave 100%
  mixer $card Music 100%
  mixer $card AC97 100%
  mixer $card Surround 90%
  mixer $card 'Surround Digital' 90%
  mixer $card 'Wave Surround' 90%
  mixer $card 'Duplicate Front' 90%
  mixer $card 'Sigmatel 4-Speaker Stereo' 90%

# CS4237B chipset:
  mixer $card 'Master Digital' 100%

# DRC
  mixer $card 'Dynamic Range Compression' 90%

# Envy24 chips with analog outs
  mixer $card DAC 100%
  mixer $card DAC,0 100%
  mixer $card DAC,1 100%

# some notebooks use headphone instead of master
  mixer $card Headphone 100%
  mixer $card Speaker 100%
  mixer $card 'Internal Speaker' 0% mute
  mixer $card Playback 100%
  mixer $card Headphone 100%
  mixer $card Speaker 100%
  mixer $card Center 100%
  mixer $card LFE 100%
  mixer $card Center/LFE 100%

# Intel P4P800-MX  (Ubuntu bug #5813)
  mixer $card 'Master Playback Switch' on

# set digital output mixer params
  mixer $card 'IEC958' 100% on
  mixer $card 'IEC958 Output' 100%
  mixer $card 'IEC958 Coaxial' 100%
  mixer $card 'IEC958 LiveDrive' 100%
  mixer $card 'IEC958 Optical Raw' 100%
  mixer $card 'SPDIF Out' 100%
  mixer $card 'SPDIF Front' 100%
  mixer $card 'SPDIF Rear' 100%
  mixer $card 'SPDIF Center/LFE' 100%
  mixer $card 'Master Digital' 100%

  mixer $card 'Analog Front' 100%
  mixer $card 'Analog Rear' 100%
  mixer $card 'Analog Center/LFE' 100%

# ASRock ION 330 (and perhaps others) has 2 IEC958 channels
  mixer $card IEC958,0 on
  mixer $card IEC958,1 on

# some ION2 has much more IEC958 channels ...
  mixer $card IEC958,2 on
  mixer $card IEC958,3 on

# ASRock ION 330 has Master Front set to 0
  mixer $card 'Master Front' 100%

# Shuttle XS35GT needs this too
  mixer $card 'Master',0 100% on

# and this for various Fusion devices like Zotac ZBOX
  mixer $card 'Front',0 100% on

# NVidia CK804 sound devices
  mixer $card 'IEC958 Playback AC97-SPSA' 100%

# Allwinner H3 Analog
  mixer $card 'Line Out' 0db on

# Allwinner A20 Analog
  mixer $card 'Power Amplifier' 0db
  mixer $card 'Power Amplifier DAC' on
  mixer $card 'Power Amplifier Mute' on

# Allwinner A64 Analog
  mixer $card Headphone 0db on
  mixer $card 'AIF1 Slot 0 Digital DAC' on

# Amlogic G12 HDMI to PCM0
  mixer $card 'FRDDR_A SINK 1 SEL' 'OUT 1'
  mixer $card 'FRDDR_A SRC 1 EN' on
  mixer $card 'TDMOUT_B SRC SEL' 'IN 0'
  mixer $card 'TOHDMITX I2S SRC' 'I2S B'
  mixer $card 'TOHDMITX' on

# Amlogic G12 S/PDIF to PCM1
  mixer $card 'FRDDR_B SINK 1 SEL' 'OUT 3'
  mixer $card 'FRDDR_B SRC 1 EN' on
  mixer $card 'SPDIFOUT SRC SEL' 'IN 1'
  mixer $card 'SPDIFOUT Playback' on

# Amlogic GX HDMI and S/PDIF
  mixer $card 'AIU HDMI CTRL SRC' 'I2S'
  mixer $card 'AIU SPDIF SRC SEL' 'SPDIF'

# RockPI 4B Analog
  mixer $card 'Right Headphone Mixer Right DAC' on
  mixer $card 'Left Headphone Mixer Left DAC' on

# NanoPC T4 Analog
  mixer $card 'HPO L' on
  mixer $card 'HPO R' on
  mixer $card 'HPOVOL L' on
  mixer $card 'HPOVOL R' on
  mixer $card 'HPO MIX HPVOL' on
  mixer $card 'OUT MIXL DAC L1' on
  mixer $card 'OUT MIXR DAC R1' on
  mixer $card 'Stereo DAC MIXL DAC L1' on
  mixer $card 'Stereo DAC MIXR DAC R1' on

done

fi
```
</details>

MichaIng, 2021-11-15:
Is the issue with the USB DAC probably related to #4957 (only one USB port functional, the other one not), or do both work fine in your case?

MichaIng, 2022-04-03:
Marking as closed due to outstanding reply. But feel free to reopen when you found time to further investigate.

PineA64explore, 2022-11-12:
Hi,

I am trying to build a webradio with my Pine A64+ board and was unable to output any sound from the jack headphone connector during months (HDMI was working though). I tried just about every tip and trick (with alsa, amixer, alsamixer, pulseaudio, fiddling with asound.state files and so on) I could find, with no success and was up to the point of asking for RMA when I ran through this GitHub issue, and finally tried your suggestion:
_apt install linux-image-legacy-sunxi64 linux-dtb-legacy-sunxi64 linux-u-boot-pine64-legacy_
**And it worked ! Thanks.Thank. Thanks**

I would like to help tracking the issue and make sure that it disappears from future DietPi and Armbian builds.

Before installing the legacy images you proposed, I was running the last DietPi Bullseye image (before that I tried Armbian Bullseye). The playback devices seemed ok:

_root@DietPi:~# aplay -l

**** List of PLAYBACK Hardware Devices ****
card 0: sun50ia64audio [sun50i-a64-audio], device 0: 1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0 [1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sun9ihdmi [sun9i-hdmi], device 0: SUN9I-HDMI PCM i2s-hifi-0 [SUN9I-HDMI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0_

But I couldn't get any sound from jack.

One day, looking at dmesg during headphone plug-in and plug-out, what I found was intriging :
This is what I get when plugging IN :
_[ 3517.655643] jack: irq plug-out
[ 3517.663413] jack: irq plug-in
[ 3517.696086] jack: irq plug-out
[ 3517.762223] jack: plug-out reported_
=> so the headphone is reported being plugged out, when I plug it in.

This is what I get when plugging OUT :
_[ 3532.479203] jack: irq plug-in
[ 3532.487846] jack: irq plug-out
[ 3532.496029] jack: irq plug-in_
=> the headphone is reported being plugged in when I plug it out.

After installing the legacy images, these messages disappeared totally (no message at all about jack).
Maybe nothing to do with the sound problem, but still...

How can I help ? I have two of these boards, and can try any suggestion or build in order to track down the bug in the recent builds..
Thanks

MichaIng, 2022-11-12:
If you find time, you could try `edge` kernel, hence newer one instead of older:
```sh
G_AGP linux-{image,dtb}-{current,legacy}-sunxi64 linux-u-boot-pine64-{current,legacy}
G_AGI linux-{image,dtb}-edge-sunxi64 linux-u-boot-pine64-edge
```

PineA64explore, 2022-11-14:
Tried and compared legacy, current and edge:

**1) Current : no sound through jack connector**

root@DietPi:~# lsmod 
Module                  Size  Used by
snd_soc_hdmi_codec     24576  1
axp20x_adc             20480  0
joydev                 28672  0
axp20x_battery         16384  0
sun9i_hdmi_audio       16384  0
axp20x_ac_power        16384  0
snd_soc_simple_card    24576  0
snd_soc_simple_card_utils    24576  1 snd_soc_simple_card
dw_hdmi_cec            16384  0
sunxi_cedrus           49152  0
videobuf2_dma_contig    24576  1 sunxi_cedrus
dw_hdmi_i2s_audio      16384  0
v4l2_mem2mem           36864  1 sunxi_cedrus
videobuf2_memops       20480  1 videobuf2_dma_contig
sun4i_gpadc_iio        24576  0
lima                   57344  0
videobuf2_v4l2         24576  2 sunxi_cedrus,v4l2_mem2mem
gpu_sched              32768  1 lima
sun8i_mbus             20480  0
videobuf2_common       49152  5 sunxi_cedrus,videobuf2_dma_contig,videobuf2_v4l2,v4l2_mem2mem,videobuf2_memops
industrialio           73728  4 axp20x_battery,sun4i_gpadc_iio,axp20x_ac_power,axp20x_adc
sun4i_i2s              24576  4
videodev              204800  4 sunxi_cedrus,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
mc                     49152  5 sunxi_cedrus,videodev,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
display_connector      20480  0
cpufreq_dt             20480  0
fuse                  126976  1
pinctrl_axp209         16384  0
realtek                32768  1
dwmac_sun8i            28672  0
mdio_mux               16384  1 dwmac_sun8i
i2c_mv64xxx            24576  0

**2) Legacy : jack works perfectly**

root@DietPi:~# lsmod 
Module                  Size  Used by
snd_soc_hdmi_codec     20480  1
dw_hdmi_i2s_audio      16384  0
dw_hdmi_cec            16384  0
snd_soc_spdif_tx       16384  0
axp20x_adc             20480  0
axp20x_battery         16384  0
axp20x_ac_power        16384  0
sun50i_codec_analog    28672  1
snd_soc_simple_card    24576  0
snd_soc_simple_card_utils    20480  1 snd_soc_simple_card
sun8i_adda_pr_regmap    16384  1 sun50i_codec_analog
sun8i_codec            28672  1
sun4i_i2s              24576  4
snd_soc_core          176128  7 sun4i_i2s,snd_soc_spdif_tx,snd_soc_hdmi_codec,sun50i_codec_analog,sun8i_codec,snd_soc_simple_card_utils,snd_soc_simple_card
ac97_bus               16384  1 snd_soc_core
sun8i_drm_hdmi         24576  0
lima                   45056  0
dw_hdmi                40960  2 sun8i_drm_hdmi,dw_hdmi_i2s_audio
snd_pcm_dmaengine      16384  1 snd_soc_core
joydev                 32768  0
cec                    69632  2 dw_hdmi_cec,dw_hdmi
snd_pcm               110592  5 sun4i_i2s,snd_soc_hdmi_codec,sun8i_codec,snd_soc_core,snd_pcm_dmaengine
gpu_sched              28672  1 lima
snd_timer              40960  1 snd_pcm
snd                    86016  4 snd_soc_hdmi_codec,snd_timer,snd_soc_core,snd_pcm
sun4i_gpadc_iio        24576  0
soundcore              16384  1 snd
cpufreq_dt             20480  0
ip_tables              32768  0
x_tables               36864  1 ip_tables
autofs4                49152  2
realtek                24576  1
axp20x_regulator       45056  7
pinctrl_axp209         16384  0
axp20x_usb_power       16384  0
dwmac_sun8i            28672  0
mdio_mux               16384  1 dwmac_sun8i
i2c_mv64xxx            20480  0
sun8i_mixer            40960  0
sun4i_drm              20480  0
sun4i_frontend         20480  1 sun4i_drm
sun4i_tcon             32768  1 sun4i_drm
sun8i_tcon_top         16384  3 sun8i_drm_hdmi,sun4i_tcon,sun4i_drm

**3) Edge : no sound through jack connector**
Edge

root@DietPi:~# lsmod 
Module                  Size  Used by
snd_soc_hdmi_codec     24576  1
joydev                 32768  0
axp20x_adc             20480  0
snd_soc_simple_card    24576  0
axp20x_ac_power        16384  0
sun9i_hdmi_audio       16384  0
snd_soc_simple_card_utils    24576  1 snd_soc_simple_card
axp20x_battery         16384  0
sunxi_cedrus           49152  0
videobuf2_dma_contig    24576  1 sunxi_cedrus
v4l2_mem2mem           36864  1 sunxi_cedrus
lima                   57344  0
videobuf2_memops       20480  1 videobuf2_dma_contig
sun4i_gpadc_iio        24576  0
videobuf2_v4l2         24576  2 sunxi_cedrus,v4l2_mem2mem
gpu_sched              32768  1 lima
dw_hdmi_i2s_audio      16384  0
dw_hdmi_cec            16384  0
videobuf2_common       49152  5 sunxi_cedrus,videobuf2_dma_contig,videobuf2_v4l2,v4l2_mem2mem,videobuf2_memops
sun4i_i2s              24576  4
sun8i_a33_mbus         20480  0
drm_shmem_helper       20480  1 lima
industrialio           73728  4 axp20x_battery,sun4i_gpadc_iio,axp20x_ac_power,axp20x_adc
videodev              204800  4 sunxi_cedrus,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
mc                     49152  5 sunxi_cedrus,videodev,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
display_connector      20480  0
cpufreq_dt             20480  0
fuse                  126976  1
realtek                32768  1
pinctrl_axp209         16384  0
dwmac_sun8i            28672  0
mdio_mux               16384  1 dwmac_sun8i

Legacy seems to have many more modules activated. and they differ a lot from current and edge.

MichaIng, 2022-11-14:
I guess `sun50i_codec_analog` is the relevant module, and `snd_soc_simple_card` the non-functional fallback. Yeah, legacy most likely is the outdated but original PINE64 vendor kernel, while current and edge are patched mainline kernel builds.

I guess it will be the same issue on Pinebook.

Legacy is Linux 5.4, right?
```sh
uname -a
```

PineA64explore, 2022-11-15:
root@DietPi:~# uname -a
Linux DietPi 5.4.88-sunxi64 #21.02.3 SMP Mon Mar 8 08:07:33 UTC 2021 aarch64 GNU/Linux

PineA64explore, 2022-11-18:
Can I help with more tests ? Or do I need to open a new issue or is this one reopened ? Sorry, I am not that familiar with the process...

MichaIng, 2022-11-18:
I'll do some research tomorrow. If there is no solution in sight, I'll generate legacy kernel images.

MichaIng, 2022-12-04:
Based on case with ROCK Pi 4, as long as `aplay -l` does show the analogue jack audio device, tweaking `amixer`/`alsamixer` controls might fix it. It someone lands here or has time to test with our current image (and `current` kernel branch), please paste the output of the following command:
```sh
amixer -c 0 scontents
```

PineA64explore, 2022-12-11:
Just tried **current** again, still no sound. Here's what I get with **aplay** an a **amixer**:

_root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sun50ia64audio [sun50i-a64-audio], device 0: 1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0 [1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sun9ihdmi [sun9i-hdmi], device 0: SUN9I-HDMI PCM i2s-hifi-0 [SUN9I-HDMI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0_

After unmuting all devices devices with alsamixer, this is what amixer gives:

_root@DietPi:~# amixer -c 0 scontents
Simple mixer control 'Headphone',0
  Capabilities: pvolume pvolume-joined pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 63
  Mono:
  Front Left: Playback 56 [89%] [-7.00dB] [on]
  Front Right: Playback 56 [89%] [-7.00dB] [on]
Simple mixer control 'Headphone Source',0
  Capabilities: penum
  Items: 'DAC' 'Mixer'
  Item0: 'DAC'
  Item1: 'DAC'
Simple mixer control 'Line In',0
  Capabilities: pvolume pvolume-joined pswitch cswitch
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: Playback 0 - 7
  Front Left: Playback 5 [71%] [3.00dB] [on] Capture [off]
  Front Right: Playback 5 [71%] [3.00dB] [on] Capture [off]
Simple mixer control 'Line Out',0
  Capabilities: pvolume pvolume-joined pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 31
  Mono:
  Front Left: Playback 27 [87%] [-6.00dB] [on]
  Front Right: Playback 27 [87%] [-6.00dB] [on]
Simple mixer control 'Line Out Source',0
  Capabilities: penum
  Items: 'Stereo' 'Mono Differential'
  Item0: 'Stereo'
  Item1: 'Stereo'
Simple mixer control 'Mic1',0
  Capabilities: pvolume pvolume-joined pswitch cswitch
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: Playback 0 - 7
  Front Left: Playback 5 [71%] [3.00dB] [on] Capture [off]
  Front Right: Playback 5 [71%] [3.00dB] [on] Capture [off]
Simple mixer control 'Mic1 Boost',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 7
  Mono: 4 [57%] [33.00dB]
Simple mixer control 'Mic2',0
  Capabilities: pvolume pvolume-joined pswitch cswitch
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: Playback 0 - 7
  Front Left: Playback 5 [71%] [3.00dB] [on] Capture [off]
  Front Right: Playback 5 [71%] [3.00dB] [on] Capture [off]
Simple mixer control 'Mic2 Boost',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 7
[... 128 lines cut ...]
  Item0: 'None'
Simple mixer control 'DAC',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 192
  Mono:
  Front Left: Playback 178 [93%] [13.50dB] [on]
  Front Right: Playback 178 [93%] [13.50dB] [on]
Simple mixer control 'DAC Reversed',0
  Capabilities: pswitch
  Playback channels: Front Left - Front Right
  Mono:
  Front Left: Playback [on]
  Front Right: Playback [on]
Simple mixer control 'Earpiece',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 31
  Mono: Playback 24 [77%] [-10.50dB] [on]
Simple mixer control 'Earpiece Source',0
  Capabilities: penum
  Items: 'DACR' 'DACL' 'Right Mixer' 'Left Mixer'
  Item0: 'DACR'_

uname -a : Linux DietPi 5.15.80-sunxi64 #22.11.1 SMP Wed Nov 30 11:23:00 UTC 2022 aarch64 GNU/Linux

qtpies, 2022-12-17:
I'm having exactly the same issue on Armbian. Looking forward to any kind of resolution.

MichaIng, 2022-12-18:
Does this help?
```sh
amixer -c 0 set 'AIF1 DA0' '100%'
```

qtpies, 2022-12-18:
> Does this help?
> 
> ```shell
> amixer -c 0 set 'AIF1 DA0' '100%'
> ```

Unfortunately not. I also get the message below in dmesg.

> This is what I get when plugging OUT :
> [ 3532.479203] jack: irq plug-in
> [ 3532.487846] jack: irq plug-out
> [ 3532.496029] jack: irq plug-in
> => the headphone is reported being plugged in when I plug it out.

> One day, looking at dmesg during headphone plug-in and plug-out, what I found was intriging : This is what I get when plugging IN : _[ 3517.655643] jack: irq plug-out [ 3517.663413] jack: irq [...]

qtpies, 2022-12-20:
Thank you that would be great.

I still really appreciate PINE, but I do have to say that it is a bit 
weird for PINE to mention Armbian as the recommended OS 
(https://www.pine64.org/gettings-started/) when there is this level of 
incompatibility. The audio jack is not a minor feature. 'OS support not 
guaranteed' would be more accurate, or just mention the last known 
compatible kernel...

On 20-12-2022 04:09, MichaIng wrote:
> Since Armbian has no PINE A64 maintainer and I do not even find a 
> related report on their "unsupported SBCs" forum, and kernel driver 
> development goes beyond my capabilities, for now I think we need to live 
> with this, and using legacy kernel as workaround if analogue onboard 
> audio is needed. We'll host two images then. There are other cases where 
> using legacy kernel fixes specific issues, so we'll do this for some 
> more SBCs. I need to implement the ability to our build scripts to pass 
> a custom kernel branch.
>

MichaIng, 2022-12-21:
From Armbian side, the PINE A64 is currently not supported, i.e. there ia no SBC maintainer for it which could test hardware features with new kernel versions, in case fix stuff, moderate community reports/help etc. It was in the past, and the PINE64 wiki is likely not updated that regularly.

It's nasty that often ARM hardware support is not well integrated into mainline kernel but as part of a large number of kernel patches and 3rd party drivers which can break or conflict at any new kernel release. And Armbian just added another huge number of media-related patches from LibreELEC. Not sure if anyone will ever be able to properly review them, whether (still) required or not, possibly conflict or not, regularly. It currently looks like "they can be applied without error, so should be fine". I lack experience whether this is a save approach for the Linux kernel or not, but at least it isn't optimal.

PanderMusubi, 2023-04-09:
I am getting started for picking up support of PINE64 at Armbian. If I need to test anything, let me know.

Question for the community here, how are kernel modules selected/configured for the PINE64 builds?

Joulinar, 2023-04-09:
We don't do kernel development.

PanderMusubi, 2023-04-09:
I understand, but where can I look up what kernel module configuration is used?

MichaIng, 2023-04-10:
> I am getting started for picking up support of PINE64 at Armbian.

That is great to hear. Many thanks for this 🙂. An old SBC but still widely used.

> If I need to test anything, let me know.

As of this issue, onboard audio seems to not work. I actually haven't checked whether there is a dtoverlay which needs to be enabled but as the sound cards are detected already, just do not play sound with the mainline kernel based builds, that doesn't seem to be the issue.

> Question for the community here, how are kernel modules selected/configured for the PINE64 builds?

As we use the Armbian kernel builds for all PINE64 SBCs (except Quartz64), the config from here applies: https://github.com/armbian/build/blob/main/config/kernel/linux-rockchip-current.config

PanderMusubi, 2023-04-24:
It is a problem with the overlay, see https://forum.armbian.com/topic/15848-pine64-a64-kernel-59x-analog-sound-problem/ Will try to fix it in the build in the next weeks.

As bot kernel configuration, use the configs for the specific boards, e.g. https://github.com/armbian/build/blob/main/config/kernel/linux-sunxi64-current.config for Pine 64

MichaIng, 2023-04-24:
You mean a problem with the "device tree", not an "overlay". There was an overlay posted to fix the device tree. But we need to carefully compare with current rockchip64 kernel device tree as this was for Linux 5.9:
```dts
/dts-v1/;
/plugin/;

/ {
	compatible = "allwinner,sun50i-a64";

	fragment@0 {
		target = <&sound>;
		__overlay__ {
			status = "okay";
			/*simple-audio-card,name = "Pine64+";*/
			simple-audio-card,aux-devs = <&codec_analog>;
			simple-audio-card,widgets = "Microphone", "Microphone",
					"Headphone", "Headphone";
			simple-audio-card,routing =
					"Left DAC", "DACL",
					"Right DAC", "DACR",
					"Headphone", "HP",
					"ADCL", "Left ADC",
					"ADCR", "Right ADC",
					"Microphone", "HBIAS",
					"MIC2", "Microphone";

		};
	};

	fragment@1 {
		target = <&codec>;
		__overlay__ {
			allwinner,inverted-jack-detection;
			status = "okay";
		};
	};

	fragment@2 {
		target = <&codec_analog>;
		__overlay__ {
			cpvdd-supply = <&reg_eldo1>;
			allwinner,internal-bias-resistor;
			status = "okay";
		};
	};	

	fragment@3 {
		target = <&mixer1>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@4 {
		target = <&reg_eldo1>;
		__overlay__ {
			regulator-always-on;
		};
	};
};
```

PineA64explore, 2023-05-14:
Hi, good to see activity on this issue. I have two of these boards, and can make tests if any solution is proposed.
The device tree seems to be the right track : I also discovered that besides jack audio, GPIOs did not work with recent dietpi or armbian.
In the meantime I had to switch to an old RPI 1 for my project...

PanderMusubi, 2023-05-14:
The issue with GPIO might also explain why I wasn't able to get I2C temperature and humidity sensor working again. Any idea how to investigate/analyse that?

MichaIng, 2023-08-27:
This SBC receives a major kernel upgrade via `dietpi-update`. Worth to test whether this fixes the issues.

BJDwv, 2024-01-16:
Same kind of trouble trying bookworms version : fresh install for minimal pure image on Pine A64 SBC, no sound out of earphone jack whatever I try with JustBoom and alsamixer.

aplay -l =>
```
**** List of PLAYBACK Hardware Devices ****
card 0: sun50ia64audio [sun50i-a64-audio], device 0: 1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0 [1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sun9ihdmi [sun9i-hdmi], device 0: SUN9I-HDMI PCM i2s-hifi-0 [SUN9I-HDMI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

aconnect -l =>
```
client 0: 'System' [type=kernel]
    0 'Timer           '
    1 'Announce        '
```

aconnect -o gives nothing ! 

Then I tried to follow https://debian-facile.org/doc:materiel:cartes-son:depannage
  - Could not install alsa-base : Unable to locate package
  - Got only error from alsactl init  =>

```
alsa-lib parser.c:2783:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "simple-card" "" "" "" ""
Hardware is initialized using a generic method
alsa-lib parser.c:2783:(load_toplevel_config) Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'.
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "sun9i-hdmi" "" "" "" ""
Hardware is initialized using a generic method
```

Same SBC is still working with buster version, so I can compare both cases, but I do not know what to compare... I would like to use bookworms because buster gives problem with perl scripts (can not install libwww-perl) and to be frank, I had very hard time with buster before to get sound on jack, even do not know how it came ok !

I am lost ! What could I do ?

MichaIng, 2024-01-16:
You are aware that the guide you tried to use is from 2011? 😄

`alsa-utils` is installed already, otherwise the `aplay` command etc wouldn't be present. `alsactl init` runs automatically at boot, and when you enable audio via `dietpi-config`, so no need to run this manually. UCM is not installed, hence the errors. `alsactl -U init` skips the attempt to use UCM.

Do you know what `aconnect` is used for? For audio playback, `aplay` is the only relevant command to check.

However, we already know that it is an issue with the device tree. @PanderMusubi opened an issue at the Armbian repo, but it was closed by someone else who seem to have not understood that it is an issue with the device tree, and not one with "running" Armbian, even that Pander is the official maintainer for this SBC 🤔. Or @PanderMusubi were you able to fix it, and it is just some `alsamixer` tuning required?

BJDwv, 2024-01-17:
Thank to your reply and explanations, even if I am not smart enough to fully understand...

About aconnect, I already made wild tries like `aconnect 14:0 0:0` or `aconnect 0:0 14:0` which gave only errors, I did so because on buster version which gives sound on my SBC, aconnect -o gives something as :
```
    client 14: 'Midi through' [type:noyau]
    client 0 : 'Midi through'  Port 0
```
instead of nothing with bookworm,

other tests done today : `lsmod | grep snd_` =>
```
snd_soc_simple_card    24576  0
snd_soc_hdmi_codec     24576  1
snd_soc_simple_card_utils    24576  1 snd_soc_simple_card
```
I tried to blacklist hdmi sound card into alsa-base-blacklist.conf but it did not help

Then I compare with the working buster version where` lsmod | grep snd_` gives much more
```
snd_soc_hdmi_codec     20480  1
snd_soc_simple_card    24576  2
snd_soc_spdif_tx       16384  0
snd_soc_simple_card_utils    20480  1 snd_soc_simple_card
snd_soc_core          176128  7 sun4i_i2s,snd_soc_spdif_tx,snd_soc_hdmi_codec,sun50i_codec_analog,sun8i_codec,snd_soc_simple_card_utils,snd_soc_simple_card
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               106496  5 sun4i_i2s,snd_soc_hdmi_codec,sun8i_codec,snd_soc_core,snd_pcm_dmaengine
snd_timer              40960  1 snd_pcm
snd                    86016  8 snd_soc_hdmi_codec,snd_timer,snd_soc_core,snd_pcm
```
Well, it is true that I made a lot of tries with buster 2 years ago before to get sound

Other details : When I plug or unplug the jack, it is reported on the display, but in reverse way : display "plug removed" when I plug in ! So I tred to install with and without plug inserted : no difference.
Also using bookworm, command `init 0` does not switch off power as It does using Buster. Another command to use ?

Thank a lot for all your time helping people like me, just tell me if I should stop to post helpless stuff !

BJDwv, 2024-01-17:
I just tried : `poweroff` does exactly the same as `init 0` on my SBC, it terminate everything but does not switch off the 3.3V which I use to maintain a power-switch relay for main line.

Thank anyway for your reply.

PanderMusubi, 2024-01-18:
I am one of the two maintainers and mainly test. Device trees are above my pay grade and we need help from others. Perhaps you can find somebody on the forum of Armbian who has fixed it in the past and/or can help out.

BJDwv, 2026-10-12:
Thank for your help, I recently got more time to spend on this webradio : It's now working with dietpi Trixie. Sound volume is a bit low and I can't push control on alsamixer without getting too much distorsion, but it's quite usable.

MichaIng, 2026-10-16:
Okay, thanks for testing again. I assume it solved itself with Linux 6.18. Marking this as closed.

closed 2026-10-16 by MichaIng