MichaIng/DietPi #4806: Jellyfin "Error while starting server"

mede-erik, 2021-10-06:
The Jellyfin server fails to start with this error returning: 
```
System.InvalidOperationException: The server is expected to host the web client, but the provided content directory is either invalid or empty: / usr / lib / jellyfin / bin / jellyfin-web. If you do not want to host the web client with the server, you may set the '--nowebclient' command line flag, or set'hostwebclient = false 'in your config settings.
```
The software was installed by `dietpi-software` and nothing has changed since the installation, tried to uninstall and reinstall but the problem persists
I hope the information is sufficient

####  Information
- DietPi version : 7
- Distro version : bullseye 0
- Kernel version : Linux DietPi 5.10.63-v8+
- SBC model : RPI3
- Software title : jellyfin

Joulinar, 2021-10-06:
Hi,

could you share your configuration file as well as service definition.

```
cat /etc/default/jellyfin
systemctl cat jellyfin.service
```

Joulinar, 2021-10-09:
What update you executed? And yes, just running `jellyfin` command will not work because you need to specify environment, where to find config files and web directory.

mede-erik, 2021-10-09:
I used these two commands:
```
apt-get update
apt-get upgrade -y
```

Joulinar, 2021-10-09:
maybe configuration was corrected due to reboot or the apt package update. Ok I will close the issue for now. Feel free to reopen if needed.

closed 2021-10-09 by Joulinar