MichaIng/DietPi #4807: Adding persistent alias for /dev/ttyXY?

danergo, 2021-10-07:
Hi!

I've came across to another easy-to-solve problem, which I'm interested in: how is the best to solve this with DietPi?

Let's say I have a device in /dev with an unrecognizable name of "ttyS1".

I wanted to create a nice symlink to it: `ln -s /dev/ttyS1 /dev/ttyBTHCI`

But this gets lost on restart, as /dev is actually a devtmpfs.

Is this another topic that I can solve best by using systemd service? Or maybe I can prefer adding some udev rules for this?

Thank you!

danergo, 2021-10-07:
I have added my rename rules into `/etc/udev/rules.d/90-serial-renaming.rules`, and it works as intended.

I'll leave this open for a short time in case you have some addition to this.

Thank you!

MichaIng, 2021-10-10:
Yes, udev rules are the right thing for such. Maybe you want to share the content or yours? 🙂

A simple alternative would be a file in `/etc/tmpfiles.d` via `systemd-tmpfiles`, but since it does not depend on the actual device creation, it would apply as well when the serial device is not (yet) there. In practise however it applies late enough at boot to work reliably in very most cases:
```sh
echo 'L /dev/ttyBTHCI - - - - /dev/ttyS1' > /etc/tmpfiles.d/serial-renaming.conf
```

I'll mark this as closed.

closed 2021-10-10 by MichaIng

MichaIng, 2021-10-10:
Definitely the most elegant and reliable solution, thanks for sharing 👍.