MichaIng/DietPi #4808: Docker fails to start after update to buster or bullseye

stoinov, 2021-10-07:
#### Required Information
- DietPi version 
  ```
  G_DIETPI_VERSION_CORE=7
  G_DIETPI_VERSION_SUB=6
  G_DIETPI_VERSION_RC=2
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'
  G_LIVE_PATCH_STATUS[0]='not applicable'
  G_LIVE_PATCH_STATUS[1]='applied'
  G_LIVE_PATCH_STATUS[2]='applied'
  G_LIVE_PATCH_STATUS[3]='applied'
  G_LIVE_PATCH_STATUS[4]='not applicable'
  ```
- Distro version | `bullseye`
- Kernel version | `Linux home 3.16.85+ #1 SMP PREEMPT Tue Jun 30 19:02:35 CEST 2020 aarch64 GNU/Linux`
- SBC model | `Odroid C2 (aarch64)`
- Power supply used | `5V 1A`
- SDcard used | none

#### Additional Information (if applicable)
- Software title | `docker-ce`
- Software was `migrated` to bullseye from strech
- Can this issue be replicated on a fresh installation of DietPi? - haven't tried

#### Steps to reproduce

1. Migrate to buster
2. Migrate to bullseye

#### Expected behaviour
- Docker should start automatically

#### Actual behaviour
- docker fails with:
  ```
  Running iptables --wait -t nat -L -n failed with message: `iptables v1.8.7 (nf_tables): Could not fetch rule set generation id: Invalid argument`, error: exit status 4
```

Joulinar, 2021-10-07:
Do you know if Docker has been working fine on Buster still? Or did you not checked?

stoinov, 2021-10-07:
I haven't tested it as I just pass through it but based on the link it should have the same issue as the change was introduced then.

stoinov, 2021-10-10:
I am curious if I can get a newer version of the kernel for my C2?

MichaIng, 2021-10-10:
Not from Meveric's repository at least, not an official Hardkernel kernel. Using a different kernel package may cause issues due to the /boot FAT partition: #4788

I see the Debian kernel supports the C2 as well and as well they ship a U-Boot package which does: https://packages.debian.org/bullseye/arm64/u-boot-amlogic/filelist
Could be tested with a spare/backup SD card, as likely some tinkering is required to get it booting.

stoinov, 2021-10-10:
As I am just a casual user of linux I am unable to tinker beyond the occasional update issue like this one.
If I ever hit a roadblock I guess I will be thinking about changing distros.

MichaIng, 2022-02-11:
Solved with DietPi v8.1: iptables legacy vs nft is now configured automatically, based on whether nft is supported or not.

closed 2022-02-11 by MichaIng

belveder79, 2022-10-19:
just to warm that up again, tried installing docker-ce on DietPi v8.9.2 on a Quartz64 board from Pine64. 

`Linux DietPi 5.19.0-rc1 #1 SMP PREEMPT Fri Jul 15 09:39:18 UTC 2022 aarch64 GNU/Linux`

The install of docker-ce from the interface failed, and iptables was not set to the legacy version automatically - I had to do this manually, afterwards docker started as desired.

```
update-alternatives --set iptables /usr/sbin/iptables-legacy
update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy
```

MichaIng, 2022-10-20:
Ah good hint that Docker use iptables as well, so we can apply the function there as well to use legacy iptables where necessary.