MichaIng/DietPi #4820: Open Beta v7.7 | Please help testing and hardening the upcoming release

MichaIng, 2021-10-09:
RC version | v7.7.0
---------- | -------
v7.7.0 PR | #4819
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2021-10-16**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av7.7
____
### Beta v7.7.0
(2021-10-09)

#### Removed software
- CouchPotato | Sadly, the CouchPotato project is not maintained anymore and has been abandoned. In turn we removed it from DietPi. The instance installed on your system will remain, but it cannot be installed, reinstalled or uninstalled anymore. We recommend to migrate to an alternative project, like Radarr, which can be found in DietPi-Software was well. Manual CouchPotato uninstall instructions: https://github.com/MichaIng/DietPi/issues/4323#issuecomment-927128724

#### Changes
- DietPi-Globals | G_GET_WAN_IP: We use our own GEO IP service now to show the systems WAN IP and location in the DietPi banner and DietPi-VPN. When Pi-hole was used, with a previous update, "freegeoip.app" was added to Pi-hole's whitlist, which is now not required anymore. You may hence remove that entry from the whitelist.
- DietPi-Globals | A new global function G_GET_NET has been added to print network interface details. Most importantly it prints info for the main interface, by following the priorities of /boot/dietpi/func/obtain_network_details: default gateway => state UP => IP assigned, but allows to additionally filter by IP family, type, interface name or print the default gateway explicitly. It aims to be a replacement for /boot/dietpi/func/obtain_network_details with more flexibility and to allow deriving always up-to-date interface info instead of depending on the correctness of a cache file.
- DietPi-Globals | The global functions G_DEV_1 and G_DEV_BENCH have been removed, which did exist for testing and development only but are not used in our current workflows.
- DietPi-Login | The DietPi banner on login won't be shown anymore if ~/.hushlogin exists, which is a common method to prevent the shell from printing /etc/motd on login and should hence be respected for the DietPi banner as well. Many thanks to @dnknth for doing this suggestion: https://github.com/MichaIng/DietPi/issues/4786
- DietPi-Software | Deluge: Logging is not done to /var/log/deluged/ anymore but to journal instead, accessible via "journalctl -u deluged -u deluge-web". This change only affects new installs and reinstalls of Deluge.
- DietPi-Software | Deluge: On fresh installs, the web interface is now accessible as expected with the chosen global software password, stored hashed with a fresh random salt. Previously the password was hardcoded to "dietpi".

#### Fixes
- General | Since the Armbian repository router does not reliably preservers HTTPS on redirects yet, APT by times fails, when detecting a downgrade from HTTPS to HTTP. We hence change the armbian.list to use plain HTTP, until the issues with the router have been resolved.
- DietPi-Backup | Resolved an issue where clearing the PATH cache via "hash" command did not work as of a wrong command line argument: https://github.com/MichaIng/DietPi/issues/4800
- DietPi-LetsEncrypt | Resolved an issue where the script failed when ownCloud or Nextcloud are were installed. Many thanks to @billouetaudrey for reporting this issue: https://github.com/MichaIng/DietPi/issues/4752
- DietPi-Software | Resolved an issue where software services failed with a cryptic error message, when an expected directory was not present. This was especially reported with Sonarr and Radarr, if their log directory was missing for some reason. When directories are missing, which are explicitly listed to be read-writeable within the systemd service, systemd prints "Failed at step NAMESPACE spawning", while Sonarr and Radarr themselves would print a clearer error message about the missing log directory. Many thanks to `@stevewitz` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=9463
- DietPi-Software | Lighttpd: Resolved an issue where the upgrade from Buster to Bullseye, following our guide, fails if HTTPS was enabled via DietPi-LetsEncrypt before. Many thanks to `@fhals` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=9477
- DietPi-Software | Kodi: Resolved an issue on RPi ARMv8/64-bit systems where Kodi fails to start when it was installed without a desktop. Many thanks to `@Klola` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?p=38079#p38079
- DietPi-Software | Deluge: Resolved an issue on Bullseye where the web interface service did not start as a new command line flag "-d" is required to keep it in foregound. Many thanks to @quyentruong for reporting this issue: https://github.com/MichaIng/DietPi/issues/4785
- DietPi-Software | Home Assistant: The Python version compiled with Home Assistant has been bumped to v3.9.7, which resolves and issue with installs on 32-bit ARM systems. Many thanks to @Przemek for reporting this issue: https://github.com/MichaIng/DietPi/issues/4372#issuecomment-936656595
- DietPi-Software | Home Assistant: Resolved an issue where Home Assistant did not start on ARM systems due to newly required runtime libraries.

MichaIng, 2021-10-11:
`vm.swappiness=0` does not disable swap usage but reduces it's usage to a minimum. This is mostly reasonable with an SD card as system drive where you want to avoid any possible disk I/O, with an SSD not so much, depending a bit on usage in general:
- Generally of course the swap space is much slower than the RAM, so keeping things in RAM instead of writing them to disk generally, on first view, increases performance, while the letter should only be considered a worst case fallback, when there is simply no left space in RAM, to avoid OOM kills. `vm.swappiness=0` behaves moreless like that, by only swapping data from RAM when physical RAM is mostly full.
- But due to Linux filesystem caching sometimes it makes sense to swap rarely accessed RAM data earlier. E.g. you may run a Java application which usually reserves several hundred MiBs in RAM to store the full runtime and for heap data, while the actually only a little part of Java classes and heap are (regularly) used, and the reserved memory is mostly a waste of RAM space. Now when the RAM is mostly full, with `vm.swappiness=0` foreground process data is still kept preferably in physical RAM, so that there is few RAM available and used for the filesystem cache. So another process which e.g. reads often accessed files from disk will be slower, because that disk files are not cached to RAM while the useless Java data is reserving it.

_Sorry for Java bashing here, but for low memory devices this is a real problem 😄._

billouetaudrey, 2021-10-11:
By default, it's 1 right ?

MichaIng, 2021-10-11:
Ah lol, whoops, `vm.swappiness=0` DOES disable swap usage completely, `vm.swappiness=1` is minimum usage (and default on DietPi via `/etc/sysctl.d/dietpi.conf`). So that is what may be tested.

MichaIng, 2021-10-14:
@billouetaudrey 
Without a swap file your memory usage is close to max, so a swap file definitely makes sense. As you can see, which higher swappiness the used swap space is higher and the used RAM space lower, as expected. What is missing is the `buff/cache` info to know whether the additional available RAM space has been effectively used for filesystem caching or not. E.g. here on my server:
```
# free -m
               total        used        free      shared  buff/cache   available
Mem:             972         385         150          93         437         478
Swap:           4095           1        4094
```
Means from 972 MiB RAM, 385 MiB are actively used and another 437 MiB were just filled with regularly accessed files from the filesystem, to speed up further accesses. The remaining 150 MiB in my case of completely unused, and obviously this didn't change significantly as the swap file was used with only 1 MiB. Since the filesystem cache is not strictly required, it can be freed quickly to make additional RAM available, if required, hence the higher "available" RAM of 478 MiB. But when accessing those files then at a later time, they need to be read from disk, hence it will take longer. In my case however a higher swappiness would have zero effect as of the completely unused 150 MiB physical RAM space. I only have the swap file because I can and it doesn't hurt as its on a 3 TiB 15% used external drive.

MichaIng, 2021-10-16:
#### Beta v7.7.2 has been released: #4839

MichaIng, 2021-10-16:
#### DietPi v7.7 has just been released, many thanks to all testers ❤️!