MichaIng/DietPi #4821: I'm getting an error from the jellyfin I installed with dietpi-software.

alpgul, 2021-10-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2

- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.63-v7+ #/1457 SMP Tue Sep 28 11:25:31 BST 2021 armv7l GNU/Linux
- SBC model | RPi 3 Model B (armv7l)

- Power supply used | 5V 1A 
- SDcard used |  SanDisk ultra 2gb

#### Additional Information (if applicable)
- Software title | jellyfin
- Was the software title installed freshly or updated/migrated?Installed freshly.
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1.  I downloaded diet pi and made all its updates.
2.  I downloaded jellyfin with dietpi-software.
3.  I got[ libmmal_core.so](https://github.com/jellyfin/jellyfin-ffmpeg/issues/32) error.
4.  I installed ffmpeg with dietpi-software. and I defined ffmpeg folder(`/usr/bin/ffmpeg`) in jellyfin. I solved libmmal_core.so error.
5. Jellyfin was very slow. I activated the omx setting in jellyfin playback.
6. Now i am getting this error`[h264_omx @ 0x2f93200] libOMX_Core.so not found
[h264_omx @ 0x2f93200] libOmxCore.so not found`. Why am I constantly dealing with errors? Is the installation wrong from the beginning?

#### Expected behaviour

- Working without error

#### Actual behaviour

- Sequential errors

#### Extra details

- What should I do to use jellyfin in dietpi?

MichaIng, 2021-10-10:
Basically the Jellyfin FFmpeg package needs to be rebuild onto current RPi Bullseye, as there were quite some breaking changes recently, including the `libmmal_core.so` problem, and the pretty related `libOMX_Core.so` one. I reported it upstream already: https://github.com/jellyfin/jellyfin-ffmpeg/issues/75

Joulinar, 2021-10-11:
Basically source of the issue is a package from Raspbian repository changing location/name of some important libraries. Now, this would need to respect by Jellyfin FFmpeg package. Means going to a different version of Jellyfin FFmpeg package will not change anything I guess.

Could you share following pls.

`ls -la /usr/lib/arm-linux-gnueabihf |grep libOmxCore`

alpgul, 2021-10-11:
I ran `ls -la /usr/lib/arm-linux-gnueabihf |grep libOmxCore` . It resulted empty line. I installed ffmpeg with dietip-software.

MichaIng, 2021-10-16:
Could you try to really purge `jellyfin-ffmpeg`:
```sh
apt purge jellyfin-ffmpeg
```

barukh27, 2021-10-17:
Manage to solve it in my case. Thanks!

I've run:
apt purge jellyfin-ffmpeg
Then install ffmpeg

And in the jeelyfin options for ffmpeg select the route: /usr/share/ffmpeg

MichaIng, 2021-10-17:
Great, many thanks for the feedback 👍.

MichaIng, 2021-10-18:
You are right, the latest release v10.7.7 meta package now depends on `jellyfin-ffmpeg` as well 🤔. That wasn't the case earlier. However, it should also be possible to only install `ffmpeg` and switch the setting. Else, actually `jellyfin` is just a meta package, so what you could do:
```sh
apt-mark manual jellyfin-server jellyfin-web
apt purge jellyfin-ffmpeg
apt install ffmpeg
```
This way the server and web component stays installed, the meta package gets removed together with `jellyfin-ffmpeg`, but it doesn't matter then.

toast003, 2021-10-19:
Tried using ffmpeg with h264_v4l2m2m after enabling the camera, and now it says that it can't find a valid device, even if there there are more modules loaded and there are things on /dev/v4l

MichaIng, 2021-10-19:
And I guess that command fails the same way on the console, so one of the arguments is invalid in this combination.

toast003, 2021-10-19:
Yup, although I don't think one of the arguments is invalid since the command `ffmpeg -i "Weathering With You.mkv" -c:v h264_v4l2m2m test.mkv` fails in the exact same way

MichaIng, 2021-10-19:
I think this is something to better continue on the Raspberry Pi forum: https://forums.raspberrypi.com/viewforum.php?f=38
It's the `ffmpeg` and `libraspberrypi0` packages for Bullseye (so no OpenMAX support) from archive.raspberrypi.org, so generally it's supposed to work. But I'm also no V/A/FFmpeg specialist, so not sure what may be missing or which flags/options may be required.

MichaIng, 2021-11-29:
I'll mark this as closed. The original library errors has been worked around the the decoder issue is something that needs to be discussed at the RPi forum.

closed 2021-11-29 by MichaIng