MichaIng/DietPi #4822: Making DietPi more "diet"?

danergo, 2021-10-10:
Hi!

My SBC has very limited storage, so I'm thinking on saving some more thus making DietPi a little bit more "diet".

Particularly, my "/boot" partition is 253MB large and have 203MB available for free currently.
I know sometimes it needs space for upgrade, but what would be the smallest size sufficient for "/boot" and how can I reach that? Can I resize it and add the extra MBs to "/"?

Also for "/boot": I have many files in there which seems unneeded, like:

```
kernel8.img
kernel71.img
kernel7.img
kernel.img
```
Which one of these is needed? Each of them are 5+ MBs, so in case only one is enough I can save 15MB+ extra.

I'm also having issues with drive-manager: I have 600Mb free space and this is displayed...

[image]

...and if I try toggling (disabling the swap):

[image]

So it behaves inconsistent at least in my case.

MichaIng, 2021-10-10:
Indeed, since the RPi images, respectively the kernel package, is compatible with all RPi models, it comes with for kernels. Especially also there are modules for all four of them in in `/lib/modules` which is the much larger part.

I was already asking whether it wouldn't make sense to remove the 64-bit kernel from the 32-bit/armhf kernel package: https://github.com/RPi-Distro/repo/issues/246
But it would break the OOTB ability to switch to aarch64 via config.txt entry, whatever the reason for this might be when using a 32-bit OS/userspace.

However, there are ways to exclude DEB package files, e.g. on my RPi systems I use something like this:
```
2021-10-10 21:10:35 root@Amizero:~# cat /etc/dpkg/dpkg.cfg.d/01-micha
...
# bcm2709 cut-down firmware only
path-exclude /boot/start4*.elf
path-exclude /boot/start_x.elf
path-exclude /boot/start_db.elf
path-exclude /boot/fixup4*.dat
path-exclude /boot/fixup_x.dat
path-exclude /boot/fixup_db.dat
path-exclude /boot/bcm27*.dtb
path-include /boot/bcm2708-rpi-zero-w.dtb
path-exclude /boot/kernel*.img
path-include /boot/kernel.img
path-exclude /lib/modules/*-v*/*
path-exclude /lib/modules/*-v*
...
```
This excludes every kernel, dtb and bootloader file despite the ones required for RPi Zero W (and only regular and cut-down firmware start/fixup files).

Joulinar, 2021-10-10:
to disable SWAP, run following on CLI

`/boot/dietpi/func/dietpi-set_swapfile 0`