MichaIng/DietPi #4825: PiVPN installation issues

maartenlangeveld, 2021-10-11:
NanoPi NEO2, DietPi 7.5..2 (buster)

Have had installed PiVPN and worked fine. Now I have reinstalled and encounter some issues.

#### Details:
- Date           | Mon Oct 11 10:01:45 CEST 2021
- Bug report     | 3333c105-1442-43f2-8d9d-387462560043
- DietPi version | v7.6.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | NanoPi NEO2 (aarch64) (ID=65)
- Kernel version | `Linux M-DNS 5.10.60-sunxi64 #21.08.1 SMP Wed Aug 25 18:29:57 UTC 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5)
- Command        | `./install.bash`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
:::
::: You are root.
::: Hostname length OK
::: Verifying free disk space...
:::
::: Package Cache update is needed, running apt-get update -y ...
 done!
:::
::: Checking apt-get for upgraded packages.... done!
:::
::: Your system is up to date! Continuing with PiVPN installation...
:::    Checking for git... already installed!
:::    Checking for tar... already installed!
:::    Checking for wget... already installed!
:::    Checking for curl... already installed!
:::    Checking for grep... already installed!
:::    Checking for dnsutils... already installed!
:::    Checking for whiptail... already installed!
:::    Checking for net-tools... already installed!
:::    Checking for bsdmainutils... already installed!
:::    Checking for iptables-persistent... already installed!
::: Using User: dietpi
:::
::: Checking for existing base files...
:::    Checking /usr/local/src/pivpn is a repo... OK!
:::     Updating repo in /usr/local/src/pivpn... done!
::: Using VPN: OpenVPN
::: Installing OpenVPN from Debian package...
:::    Checking for openvpn... not installed!
:::    Checking for grepcidr... already installed!
:::    Checking for expect... already installed!
:::    Failed to install openvpn!
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  liblzo2-2 libpkcs11-helper1
Suggested packages:
  openvpn-systemd-resolved
Recommended packages:
  easy-rsa
The following NEW packages will be installed:
  liblzo2-2 libpkcs11-helper1 openvpn
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 558 kB of archives.
After this operation, 1453 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main arm64 liblzo2-2 arm64 2.10-0.1 [49.0 kB]
Get:2 https://deb.debian.org/debian buster/main arm64 libpkcs11-helper1 arm64 1.25.1-1 [45.2 kB]
Get:3 https://deb.debian.org/debian buster/main arm64 openvpn arm64 2.4.7-1+deb10u1 [464 kB]
Preconfiguring packages ...
Can't locate object method "new" via package "Text::Iconv" (perhaps you forgot to load "Text::Iconv"?) at /usr/share/perl5/Debconf/Encoding.pm line 65.
Fetched 558 kB in 0s (2133 kB/s)
Selecting previously unselected package liblzo2-2:arm64.
(Reading database ... 25828 files and directories currently installed.)
Preparing to unpack .../liblzo2-2_2.10-0.1_arm64.deb ...
Unpacking liblzo2-2:arm64 (2.10-0.1) ...
Selecting previously unselected package libpkcs11-helper1:arm64.
Preparing to unpack .../libpkcs11-helper1_1.25.1-1_arm64.deb ...
Unpacking libpkcs11-helper1:arm64 (1.25.1-1) ...
Selecting previously unselected package openvpn.
Preparing to unpack .../openvpn_2.4.7-1+deb10u1_arm64.deb ...
Unpacking openvpn (2.4.7-1+deb10u1) ...
Setting up liblzo2-2:arm64 (2.10-0.1) ...
Setting up libpkcs11-helper1:arm64 (1.25.1-1) ...
Setting up openvpn (2.4.7-1+deb10u1) ...
Can't locate object method "new" via package "Text::Iconv" (perhaps you forgot to load "Text::Iconv"?) at /usr/share/perl5/Debconf/Encoding.pm line 65.
dpkg: error processing package openvpn (--configure):
 installed openvpn package post-installation script subprocess returned error exit status 29
Processing triggers for systemd (241-7~deb10u8) ...
Processing triggers for libc-bin (2.28-10) ...
Errors were encountered while processing:
 openvpn
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------

Besides why does the DietPi install script for PiVPN makes new directories for Music, Pictures and Video ( mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin)? Do not understand, what's the logic behind this? I do not need this directories on my Pi-hole SBC and PiVPN/OpenVPN does not need them either! So will delete them again..

maartenlangeveld, 2021-10-11:
Ok, i will try. It's a pity the Dietpi Installation and/or PiVPN does not work flawless anymore as it did before. Had to reinstall so it would be nice if I could get it working again...

Joulinar, 2021-10-11:
don't wait on us to fix PiVPN scripts as they are out of our control. We could try to contribute but don't know if someone from PiVPN will take it.

In meantime, try to install OpenVPN server core manually before running PiVPN installer again.

```
apt update
apt install openvpn
```

Joulinar, 2021-10-11:
where did you get these message? On `apt install openvpn` or somewhere else?

Joulinar, 2021-10-11:
you could check if there is a VPN is listen on UDP port 1194

4s3ti, 2021-11-01:
> don't wait on us to fix PiVPN scripts as they are out of our control. We could try to contribute but don't know if someone from PiVPN will take it.
> 
> In meantime, try to install OpenVPN server core manually before running PiVPN installer again.
> 
> ```
> apt update
> apt install openvpn
> ```

@Joulinar We always take it forward and Pull Requests are always welcomed :) 

@MichaIng @Joulinar just to make sure I got it right, from this issue all is missing is to make supre PiVPN checks if directories exist and creates them if needed right?

MichaIng, 2021-11-01:
Ah thanks for chiming in. The original issue has been solved with: https://github.com/pivpn/pivpn/pull/1367
And also the PR to fix the (harmless) logging configuration error messages has been merged: https://github.com/pivpn/pivpn/pull/1362
Both very quickly, so thanks to @4s3ti it is highly productive to report issues and offer PRs on PiVPN, many thanks for this 👍🙂.

I'll mark this issue hence as solved/closed.

closed 2021-11-01 by MichaIng