MichaIng/DietPi #4832: Bluetooth (hci) problems on Bullseye

danergo, 2021-10-13:
Hi!

I know this might be overcomplicated for DietPi, but you might heard/seen similar issues, which might help me:

I have an ESP32 on serial port as a Bluetooth co-processor. It is connected via hciattach, and everything works fine.

However, after a not-too-long time (few hours) it stops working in a way:

```
# l2ping my_smartphone
Can't connect: Operation now in progress
```

But this is only true for some devices. Some other devices are pingable with *l2ping*. (And vice-versa: *my_smartphone* IS pingable with *l2ping* from a different board)

First I found an escape from this situation: reset the ESP32. That solves this.

However today I found, that I don't have to restart ESP32: in case I restart the hci interface, it also solves:

```
# hciconfig hci0 down;hciconfig hci0 up;l2ping my_smartphone
WORKS
```

So, I think now, it's not ESP32's fault, but more the Linux. Do you have any addition by any chance?

Thank you!

MichaIng, 2022-02-20:
Does the issue persist with latest kernel/`apt upgrade`?

danergo, 2022-02-20:
Hi, I had no chance yet to validate, I need some more days and surely let you know here!

MichaIng, 2022-04-03:
As of issue cleanup, I'll mark this as closed. But feel free to reopen when you have news.

closed 2022-04-03 by MichaIng