MichaIng/DietPi #4843: Ombi Not started automatically after upgrading through apt

Tooloco, 2021-10-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
G_LIVE_PATCH_STATUS[3]='applied'
G_LIVE_PATCH_STATUS[4]='applied'`
- Distro version | `bullseye 1`
- Kernel version | `Linux DietPi3 5.10.60-v7l+ #1449 SMP Wed Aug 25 15:00:44 BST 2021 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | [Labists 5.1V 3A](https://labists.com/products/raspberry-pi-4-power-supply)
- SDcard used | SanDisk ultra 32GB

#### Additional Information (if applicable)
- Software title | Ombi
- Was the software title installed freshly or updated/migrated? Updated through apt
- Can this issue be replicated on a fresh installation of DietPi? Not tested

- Bug report ID | 18539501-a227-4fd7-a0a6-1cbc2f302b4c

#### Steps to reproduce

1. Have the Ombi service controlled through dietpi-services
2. Update Ombi through: ´apt update && apt upgrade -y´

#### Expected behaviour

- Ombi updates succesfully and then the service gets started automatically

#### Actual behaviour

- Ombi is not being started automatically

#### Extra details

- Apt outputs this message after upgrading: ´ombi.service is a disabled or a static unit, not starting it.´

Joulinar, 2021-10-15:
Well this seems more a information message than an error message. What is the status of the service after a reboot?

MichaIng, 2021-10-15:
```sh
dietpi-services start
```
Does it show ombi?

MichaIng, 2021-10-16:
I mark this as closed then. Feel free to reopen if required.

closed 2021-10-16 by MichaIng