MichaIng/DietPi #4844: [FAILED] DietPi-Survey

billouetaudrey, 2021-10-16:
Hey,

is it normal, I always have this bug after do dietpi-update : 

[FAILED] DietPi-Survey | Purging survey data
 - Command: curl --key  -m 20 -sT 80017429-532b-437d-9b75-256a91d206ec.txt sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/survey/
Thanks

MichaIng, 2021-10-16:
Many thanks for your report. What is the related error log?
```sh
> "$G_HW_UUID.txt"
curl --key '' -m 20 -sT "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/survey/'
```

billouetaudrey, 2021-10-16:
```
root@raspberrypi:~# curl --key '' -m 20 -T "$G_HW_UUID.txt" 'sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com/survey/'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (60) SSL peer certificate or SSH remote key was not OK
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
root@raspberrypi:~#
```

billouetaudrey, 2021-10-16:
Certificate issue , your or mine? :D

MichaIng, 2021-10-16:
Is the host key present?
```sh
grep 'ssh.dietpi.com' /root/.ssh/known_hosts
```

billouetaudrey, 2021-10-16:
> ```shell
> ```shell
> grep 'ssh.dietpi.com' /root/.ssh/known_hosts
> ```
> 
> 
>     
>       
>     
> 
>       
>     
> 
>     
>   
>       
> ```

Nah no file

MichaIng, 2021-10-16:
```sh
echo 'ssh.dietpi.com ssh-rsa AAAAB3NzaC1yc2EAAAADAQABAAABAQDE6aw3r6aOEqendNu376iiCHr9tGBIWPgfrLkzjXjEsHGyVSUFNnZt6pftrDeK7UX+qX4FxOwQlugG4fymOHbimRCFiv6cf7VpYg1Ednquq9TLb7/cIIbX8a6AuRmX4fjdGuqwmBq3OG7ZksFcYEFKt5U4mAJIaL8hXiM2iXjgY02LqiQY/QWATsHI4ie9ZOnwrQE+Rr6mASN1BVFuIgyHIbwX54jsFSnZ/7CdBMkuAd9B8JkxppWVYpYIFHE9oWNfjh/epdK8yv9Oo6r0w5Rb+4qaAc5g+RAaknHeV6Gp75d2lxBdCm5XknKKbGma2+/DfoE8WZTSgzXrYcRlStYN' >> /root/.ssh/known_hosts
```
However that one got removed.

billouetaudrey, 2021-10-16:
Dunno how but fixed :) 

Thanks :D

MichaIng, 2021-10-16:
Doing it on every upload seems a bit too much. Also if that file got removed, admins should notice and may try to remember of investigate whether they removed it manually (probably a cleanup/security guide or such, like "if you do not use the ssh command, you won't need any host keys") of probably some dumb script did it.

I'll move an older patch we had forward to be applied again and print a meaningful error message. Or better a yes/no, as some might have removed it intentionally, accepting the error message. It doesn't hold the update or break anything, just an information that accessing our SSH/SFTP server wan't possible.

closed 2021-10-16 by billouetaudrey

MichaIng, 2021-10-16:
Here the re-applied patch, with the option to accept or deny adding the host key: https://github.com/MichaIng/DietPi/commit/d3a70e5800df085245df532871cb07a68194a793