MichaIng/DietPi #4847: Airsonic fails starting after fresh install

Andaloup, 2021-10-16:
### bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='applied'
G_LIVE_PATCH_STATUS[3]='applied'
G_LIVE_PATCH_STATUS[4]='applied'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye 0`

- Kernel version | `uname -a`
`Linux DietPi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux`

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (aarch64)`

- Power supply used
`5V 3A`

- SDcard used 
`SanDisk ultra 32GB`

#### Additional Information 
- Software title
`Airsonic`

- Was the software title installed freshly or updated/migrated?
`installed freshly`

- Can this issue be replicated on a fresh installation of DietPi?
`Yes`

#### Steps to reproduce
1. Install Airsonic with `dietpi-software`
2. after install no conection to `http://<my.ip>:8080/airsonic`

#### Expected behaviour
Connection to airsonic-server

#### Actual behaviour
no connection, similar issue #3318

#### Extra details

My running services
- lighttpd
- openssh
- php7.4
- pi-hole-ftl

`java -version`
```
openjdk version "17-ea" 2021-09-14
OpenJDK Runtime Environment (build 17-ea+19-Debian-1)
OpenJDK 64-Bit Server VM (build 17-ea+19-Debian-1, mixed mode, sharing)
```

`journalctl -u airsonic`
```
Oct 16 13:51:06 DietPi systemd[1]: Started Airsonic Media Server (DietPi).
Oct 16 13:51:14 DietPi Airsonic[89762]:            _                       _
[... 44 lines cut ...]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.internal.LoadingCache.get(LoadingCache.java:34) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.AbstractClassGenerator$ClassLoaderData.get(AbstractClassGenerator.java:116) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.AbstractClassGenerator.create(AbstractClassGenerator.java:291) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.KeyFactory$Generator.create(KeyFactory.java:221) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.KeyFactory.create(KeyFactory.java:174) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.KeyFactory.create(KeyFactory.java:153) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.proxy.Enhancer.<clinit>(Enhancer.java:73) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         ... 22 common frames omitted
Oct 16 13:51:23 DietPi Airsonic[89762]: Caused by: java.lang.reflect.InaccessibleObjectException: Unable to make protected final java.lang.Class java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain) throws java.lang.ClassFormatError accessible: module java.base does not "opens java.lang" to unnamed module @449b2d27
Oct 16 13:51:23 DietPi Airsonic[89762]:         at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:357) ~[na:na]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297) ~[na:na]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at java.base/java.lang.reflect.Method.checkCanSetAccessible(Method.java:199) ~[na:na]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at java.base/java.lang.reflect.Method.setAccessible(Method.java:193) ~[na:na]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.ReflectUtils$1.run(ReflectUtils.java:61) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at java.base/java.security.AccessController.doPrivileged(AccessController.java:554) ~[na:na]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.ReflectUtils.<clinit>(ReflectUtils.java:52) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.KeyFactory$Generator.generateClass(KeyFactory.java:243) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.DefaultGeneratorStrategy.generate(DefaultGeneratorStrategy.java:25) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         at org.springframework.cglib.core.AbstractClassGenerator.generate(AbstractClassGenerator.java:329) ~[spring-core-4.3.25.RELEASE.jar!/:4.3.25.RELEASE]
Oct 16 13:51:23 DietPi Airsonic[89762]:         ... 34 common frames omitted
Oct 16 13:51:23 DietPi systemd[1]: airsonic.service: Main process exited, code=exited, status=1/FAILURE
Oct 16 13:51:23 DietPi systemd[1]: airsonic.service: Failed with result 'exit-code'.
Oct 16 13:51:23 DietPi systemd[1]: airsonic.service: Consumed 52.311s CPU time.

```

MichaIng, 2021-10-16:
Many thanks for your report.

Airsonic has been achieved: https://github.com/airsonic/airsonic
It doesn't seem to support Java 17.

But there is a fork: https://github.com/airsonic-advanced/airsonic-advanced
I just wanted to test it and implement it as drop-in replacement. You can try it easily:
```sh
curl -sSfL 'https://github.com/airsonic-advanced/airsonic-advanced/releases/download/v10.6.0/airsonic.war' -o /mnt/dietpi_userdata/airsonic/airsonic.war
systemctl restart airsonic
```
That one is moreless of the same age as the latest original Airsonic, but it is worked on a new v11 version as well. To test that one:
```sh
curl -sSfL 'https://github.com/airsonic-advanced/airsonic-advanced/releases/download/11.0.0-SNAPSHOT.20211014191809/airsonic.war' -o /mnt/dietpi_userdata/airsonic/airsonic.war
systemctl restart airsonic
```

MichaIng, 2021-11-14:
PR is open to migrate to Airsonic-Advanced: #4948
I just face two issues:
1. On a fresh install on Bullseye with Java 17, playback and query handling does not work without first manually removing and re-adding the player: https://github.com/airsonic-advanced/airsonic-advanced/issues/646
2. Initial access to the web interface is extremely slow. When a page or button/action was used the first time, which has a several seconds delay, then it reacts faster, but on a fresh installation where you naturally navigate though several pages, the first impression is extremely bad, so I guess many users remove it directly and use another media streamer: https://github.com/airsonic-advanced/airsonic-advanced/issues/602
    But this is the same with the current/original Airsonic, and with all OpenJDK versions, it seems to be a bug in the Java internal Tomcat for WAR handling.

MichaIng, 2022-01-03:
Migration merged.

closed 2022-01-04 by MichaIng